LibreELEC/LibreELEC.tv #5950: REQ - linux: enable CONFIG_RTL8XXXU

heitbaum, 2021-12-05:
now that rtl8xxxu is more stable, can this be added back into mainline? see https://forum.libreelec.tv/thread/10925-use-mainline-rtl8xxxu-driver/

rtl8xxxu driver works well for my various USB WiFI adapters with buster/bullseye RPiOS.

_Originally posted by @reukiodo in https://github.com/LibreELEC/LibreELEC.tv/issues/287#issuecomment-984344175_

heitbaum, 2022-02-17:
@CGarces - is this an option? Viable?

chewitt, 2022-02-17:
It needs to be combined with `CONFIG_RTL8XXXU_UNTESTED=y` as the number of devices which are formally reported as working is rather small .. but it's probably the right direction and thing to do.

CGarces, 2022-02-17:
> @CGarces - is this an option? Viable?

Is something that we must do sooner of later.
RTL8XXXU should replace RTL8192EU and RTL8192CU

heitbaum, 2022-02-17:
Suggest we PR this change post 5.17 is merged into master then?

closed 2022-03-21 by CvH