LibreELEC/LibreELEC.tv #5998: [BUG] Hardware mpeg2video decoding not available on Gemini Lake Refresh in 10.0.1

bconway, 2021-12-19:
### Describe the bug
Hardware decoding for mpeg2video is not available on a Celeron N4020 (Gemini Lake Refresh) mini pc, nor any VAAPI-base deinterlacers. The same system correctly provides hardware decoding of H.264 and HEVC, including HEVC 10-bit. This was previously discussed in https://github.com/LibreELEC/LibreELEC.tv/issues/5879, but does not appear successful here or on an older 5th gen NUC (previous ticket closed).

### To Reproduce
Steps to reproduce the behavior:
1. Open 720p video source via tvheadend PVR client
2. See software decoding being used after hitting 'o'.
3. Same applies to mpeg2-based MKV files.

### Information
 - LE Version: LE 10.0.1
 - Hardware Platform: Intel generic, [Intel® Celeron® Processor N4020](https://ark.intel.com/content/www/us/en/ark/products/197310/intel-celeron-processor-n4020-4m-cache-up-to-2-80-ghz.html)

### Log file
http://ix.io/3IRY (720p content, no deinterlacers muddying the waters, no obvious VAAPI errors)

### Additional context
Previous ticket with the same issue under older hardware: https://github.com/LibreELEC/LibreELEC.tv/issues/5879 . Closed in favor of focusing on newer hardware. I have not yet had a chance to rebuild the system with a nightly.

bconway, 2021-12-20:
Hardware decoding for mpeg2video is also not working in the latest Generic nightly.

Log file using the same 720p channel via PVR: http://ix.io/3IYQ

@heitbaum Do you still have this working after the fixes in https://github.com/LibreELEC/LibreELEC.tv/issues/4732? If so, what hardware? I am unable to get hardware decoding working under any combination of Braswell, Gemini Lake Refresh, 10.0.1, or latest nightly.

Thanks.

heitbaum, 2021-12-21:
#4771 fixed it for me. I’m on holidays at the moment, so I don’t have all my test rig. Just a build environment and a player 🏕🏔🌠 Will check when home.

bconway, 2021-12-21:
Thanks for the feedback and taking a look. I don't think there's anything wrong with my setup, as it works flawlessly on 9.2.6, but I can't seem to get hardware decoding (of mpeg2) with any content, PVR or plan ol' MKV files. Thanks, I appreciate it! Have a good holiday.

mglae, 2021-12-21:
VAAPI MPEG2 acceleration is disabled by default.

Go to Settings->Player, activate "Expert" settings level and enable VAAPI MPEG2 acceleration.

bconway, 2021-12-21:
Ugh, can't believe I missed that! I normally run at the Advanced setting level. It's working beautifully now, especially with (American) football at 1080i. Thanks @mglae @heitbaum.

I will close this, obvious not a bug. Perhaps there's a way to make it more discoverable, or bring it back as the default, like in 9.2? Though I imagine this new default was chosen for a reason.

closed 2021-12-21 by bconway

mglae, 2021-12-21:
> Though I imagine this new default was chosen for a reason.

IIRR there have been issues playing DVDs or DVD images.