LibreELEC/LibreELEC.tv #6002: [BUG] Build error after PR #5944 was merged

pretoriano80, 2021-12-20:
### Describe the bug
ncurses building fails 

### Informations
 - LE Version: LE11
 - Hardware Platform: RPi4

### Log file
[32.log](https://github.com/LibreELEC/LibreELEC.tv/files/7746738/32.log)

mglae, 2021-12-20:
```
[1;31mCLEAN[0m      ncurses
    [1;31m*[0m      [0;31mRemoving /home/pretoriano/LibreELEC.tv/build.LibreELEC-RPi4.arm-11.0-devel/build/ncurses-6.1-20181215 ...[0m
    [1;31m*[0m      [0;31mRemoving /home/pretoriano/LibreELEC.tv/build.LibreELEC-RPi4.arm-11.0-devel/install_pkg/ncurses-6.1-20181215 ...[0m
```

Clean build is required. Incremental builds do not succeed.

pretoriano80, 2021-12-20:
Ok, thanks! Will try a clean build and report back

pretoriano80, 2021-12-20:
After a "distclean"  the error is gone, thanks!

closed 2021-12-20 by pretoriano80

shantigilbert, 2021-12-21:
FYI: I added `--without-tests` to `PKG_CONFIGURE_OPTS_HOST` and it compiled even without a `distclean`. 
But not sure if the tests are required for ncurses:host

mglae, 2021-12-21:
You did still update ncurses 6.1 without widechar to 6.3 with widechar?

shantigilbert, 2021-12-21:
@mglae I just pulled the latest changes, got the same error as the one posted here, added no-test and it compiled.