LibreELEC/LibreELEC.tv #6003: [BUG] Can't install vfs.libarchive addon

WallyCZ, 2021-12-21:
### Describe the bug
Can't install vfs.libarchive addon, it throws this error: libcrypto.so.1.1: cannot open shared object file: No such file or directory

### To Reproduce
Steps to reproduce the behavior:
1. Try to install vfs.libarchive on current nightly build version

### Informations
 - LE Version: current nightly build
 - Hardware Platform: RPi4

### Log file
```
2021-12-21 00:24:18.782 T:12827    INFO <general>: CAddonMgr::FindAddon: vfs.libarchive v20.0.0.1 installed
2021-12-21 00:24:18.790 T:12819   ERROR <general>: Unable to load /storage/.kodi/addons/vfs.libarchive/vfs.libarchive.so.20.0.0, reason: libcrypto.so.1.1: cannot open shared object file: No such file or directory
2021-12-21 00:24:21.406 T:12819   ERROR <general>: IAddonInstanceHandler::CreateInstance: vfs.libarchive returned bad status "Permanent failure" during instance creation
2021-12-21 00:24:21.406 T:12819   FATAL <general>: CVFSEntry - Couldn't create instance on add-on 'Archive support'
```

### Additional context

chewitt, 2021-12-21:
You will need to wait until new libssl 3.0 compatible add-ons are published.

WallyCZ, 2021-12-21:
Closing since it is known thing.

closed 2021-12-21 by WallyCZ