LibreELEC/LibreELEC.tv #6005: Build error on openSUSE Leap 15.3 using gcc 10 - undefined reference to `exit_curses'

coffeinflash, 2021-12-22:
Building the latest master on openSUSE Leap 15.3 using gcc 10 fails with "undefined reference to `exit_curses'":

`/home/Herbie/source/libreelec/build.LibreELEC-RPi2.arm-11.0-devel/toolchain/bin/host-gcc -march=native -O2 -Wall -pipe -Wno-format-security --param max-inline-insns-single=1200  -o color_content ../obj_s/color_content.o -Wl,-rpath,/home/Herbie/source/libreelec/build.LibreELEC-RPi2.arm-11.0-devel/toolchain/lib -L/home/Herbie/source/libreelec/build.LibreELEC-RPi2.arm-11.0-devel/toolchain/lib -I. -I/home/Herbie/source/libreelec/build.LibreELEC-RPi2.arm-11.0-devel/build/ncurses-6.3/test -I../test -DHAVE_CONFIG_H -DDATA_DIR=\"/home/Herbie/source/libreelec/build.LibreELEC-RPi2.arm-11.0-devel/toolchain/share\" -I. -I/home/Herbie/source/libreelec/build.LibreELEC-RPi2.arm-11.0-devel/build/ncurses-6.3/test -I../include -I/home/Herbie/source/libreelec/build.LibreELEC-RPi2.arm-11.0-devel/build/ncurses-6.3/test/../include -I/home/Herbie/source/libreelec/build.LibreELEC-RPi2.arm-11.0-devel/toolchain/include -D_DEFAULT_SOURCE -D_XOPEN_SOURCE=600 -DNDEBUG -march=native -O2 -Wall -pipe -Wno-format-security --param max-inline-insns-single=1200  -fPIC  `echo "-L../lib -lform -lmenu -lpanel -lncurses -ltinfo  " | sed -e 's/-lform.*-lpanel[^ ]*//'` -lutil
/usr/lib64/gcc/x86_64-suse-linux/11/../../../../x86_64-suse-linux/bin/ld: ../obj_s/cardfile.o: in function `add_content':
cardfile.c:(.text+0x3b3): undefined reference to `exit_curses'
/usr/lib64/gcc/x86_64-suse-linux/11/../../../../x86_64-suse-linux/bin/ld: ../obj_s/cardfile.o: in function `main':
cardfile.c:(.text.startup+0xe3): undefined reference to `exit_curses'
/usr/lib64/gcc/x86_64-suse-linux/11/../../../../x86_64-suse-linux/bin/ld: cardfile.c:(.text.startup+0x120): undefined reference to `exit_curses'
/usr/lib64/gcc/x86_64-suse-linux/11/../../../../x86_64-suse-linux/bin/ld: ../obj_s/background.o: in function `usage':
background.c:(.text+0x100): undefined reference to `exit_curses'
/usr/lib64/gcc/x86_64-suse-linux/11/../../../../x86_64-suse-linux/bin/ld: ../obj_s/background.o: in function `main':
background.c:(.text.startup+0x100): undefined reference to `exit_curses'`

The complete build log can be found [here](https://clbin.com/Sp9Ud)

mglae, 2021-12-22:
I expected a clean build required but @shantigilbert found a configure option for a [possible solution](https://github.com/LibreELEC/LibreELEC.tv/issues/6002#issuecomment-998965327)

coffeinflash, 2021-12-24:
Adding "--without-tests" to PKG_CONFIGURE_OPTS_HOST fixed the build issue of ncurses. Thx for the hint!

But now it is failing at another step. I will open a new issue.

mglae, 2021-12-24:
Was the error on a clean or incremental build?

coffeinflash, 2021-12-25:
Good morning,

a complete clean build from scratch worked fine. This issue can be closed.

closed 2021-12-25 by jernejsk