LibreELEC/LibreELEC.tv #6007: [BUG] Downloading LibreElec image for RaspberryPi4 from the website

anujdatar, 2021-12-22:
### Describe the bug

### To Reproduce
Steps to reproduce the behavior:
1. Go to 'https://libreelec.tv/downloads/raspberry/'
2. Click on image download link
3. Nothing happens on chrome unless you do 'right-click -> save link as', on firefox it happens right away. Error says `File not downloaded: Potential security risk.. File uses insecure connection`

### Informations
 - LE Version: 10.0
 - Hardware Platform: Windows

### Log file

### Additional context

RobLoach, 2021-12-24:
Looks like the cert for http://releases.libreelec.tv/ expired.

heitbaum, 2022-02-17:
Known issue with chrome and other modern browsers. Issue is that with mirrors - require to use http, without change to the download mirrors.

closed 2022-02-17 by heitbaum

CvH, 2022-02-17:
should be fixed next week, the new mirroring is ready to get deployed if time permits