LibreELEC/LibreELEC.tv #6023: [BUILD ERROR][RPi4] Build error on Ubuntu 20.04 aarch64

pretoriano80, 2021-12-26:
### Describe the bug
Build fails due to a meson related error

### Informations
 - LE Version: LE11
 - Hardware Platform: RPi4
 - OS/Architecture: Ubuntu 20.04/ARM
 - Clean build environment 

### Log file
[142.log](https://github.com/LibreELEC/LibreELEC.tv/files/7776782/142.log)

heitbaum, 2021-12-26:
Test clean build in my dev branch with meson 0.60.3 and glib 2.71.0 (x86) all ok.

Believe issue is `--fatal-meson-warnings` 
- https://github.com/GNOME/glib/blob/glib-2-70/NEWS

running tests on master now.

heitbaum, 2021-12-26:
@pretoriano80 - has this build ever worked for you? Building RPi4 on RPi4?

please remove the following code from `packages/devel/glib/package.mk` and test.
```
if [ "${MACHINE_HARDWARE_NAME}" = "aarch64" -a "${TARGET_ARCH}" = "arm" ]; then
  PKG_MESON_PROPERTIES_TARGET="needs_exe_wrapper = true"
fi
```

pretoriano80, 2021-12-26:
@heitbaum  i'm building for RPi4, but on an Oracle   VM (Ampere A1 CPU and Ubuntu 20.04 ARM)
It's the first time I'm trying to build on such a machine, so i have no idea if it ever worked.

heitbaum, 2021-12-26:
I’m pretty sure the fix will work for you - as this is legacy code that we included in all builds now, and with meson now failing on errors (duplicate lines) here is the reference. 

GOOD FIND

https://github.com/LibreELEC/LibreELEC.tv/blob/070386f0fc102ccf4f4ff9cb0eee62eafa1559b5/config/functions#L545

pretoriano80, 2021-12-26:
Thanks for the fix, will try it later.
It's strange that nobody reported this issue before.

The VM instance i'm using its part of an Oracle Always free offer so it's great value, especially for building projects like LE.

pretoriano80, 2021-12-26:
@heitbaum  just tested and PR #6024 fixed the issue, glib compiled just fine.

heitbaum, 2021-12-26:
Thanks for testing

closed 2021-12-27 by CvH

heitbaum, 2021-12-28:
Hi @pretoriano80

could you please share the `lscpu` output for the **Ampere A1** if that is ok?
Also the build statistics (for both of the following)? 
Interested to see how this cloud server performs? 

`make distclean; make image`
`make clean; make image`

e.g. 
```
Total Build Time: 00:01:38.553 (wall clock)
Accum Build Time: 00:13:08.333 (8 slots)

Breakdown by status (all slots):

  Status   Usage         ( Pct )  Count  State
  ACTIVE     :01:49.028  (13.8%)  235    busy 
  FAILED     :  :00.000  (00.0%)  0           
  GETPKG     :  :00.000  (00.0%)  0      busy 
  IDLE       :10:54.633  (83.0%)  293         
  LOCKED     :  :11.920  (01.5%)  571         
  MUTEX      :  :00.000  (00.0%)  0      busy 
  MUTEX/W    :  :00.000  (00.0%)  0      stall
  STALLED    :  :02.172  (00.3%)  5      stall
  UNLOCK     :  :10.581  (01.3%)  571         
  -------------------------------------
  TOTAL      :13:08.333  ( 100%)  1675 

Peak concurrency: 8 out of 8 slots

5 job slots were held in a "stall" state for 00:00:02.172

Slot usage (time in a "busy" state):     | Concurrency breakdown ("busy"):
                                         |
#Rank  Slot  Usage        ( Pct )        | # of Slots  Usage        ( Pct )
 #01    01     :01:05.196 (08.3%)        |     01        :01:18.898 (10.0%)
 #02    04     :  :11.939 (01.5%)        |     02        :  :07.646 (01.0%)
 #03    07     :  :06.263 (00.8%)        |     03        :  :06.650 (00.8%)
 #04    06     :  :06.102 (00.8%)        |     04        :  :05.745 (00.7%)
 #05    03     :  :05.798 (00.7%)        |     05        :  :04.349 (00.6%)
 #06    08     :  :05.268 (00.7%)        |     06        :  :02.351 (00.3%)
 #07    02     :  :05.113 (00.6%)        |     07        :  :01.956 (00.2%)
 #08    05     :  :03.348 (00.4%)        |     08        :  :01.433 (00.2%)
-----------------------------------------+---------------------------------
 TOTALS        :01:49.028 (13.8%)                        :01:49.028 (13.8%)
```

pretoriano80, 2021-12-28:
Hello @heitbaum , yeah, no problem i will do the required tests then post the results.
In the meantime, this is the lscpu output :  
```
Architecture:                    aarch64
CPU op-mode(s):                  32-bit, 64-bit
Byte Order:                      Little Endian
CPU(s):                          4
On-line CPU(s) list:             0-3
Thread(s) per core:              1
Core(s) per socket:              4
Socket(s):                       1
NUMA node(s):                    1
Vendor ID:                       ARM
Model:                           1
Model name:                      Neoverse-N1
Stepping:                        r3p1
BogoMIPS:                        50.00
NUMA node0 CPU(s):               0-3
Vulnerability Itlb multihit:     Not affected
Vulnerability L1tf:              Not affected
Vulnerability Mds:               Not affected
Vulnerability Meltdown:          Not affected
Vulnerability Spec store bypass: Mitigation; Speculative Store Bypass disable                                 d via prctl
Vulnerability Spectre v1:        Mitigation; __user pointer sanitization
Vulnerability Spectre v2:        Not affected
Vulnerability Srbds:             Not affected
Vulnerability Tsx async abort:   Not affected
Flags:                           fp asimd evtstrm aes pmull sha1 sha2 crc32 a                                 tomics fphp asimdhp cpuid asimdrdm lrcpc dcp                                 op asimddp ssbs
```
My instance configuration is : 

> CPU = 4 core Ampere A1
RAM= 24 GB
HDD= 200 GB (boot block)
VNIC= 4 Gbit

The above configuration should be an always free configuration, meaning that after the 30 days trial period, it should remain free of any charge.

pretoriano80, 2021-12-28:
@heitbaum unfortunally Generic build fails.

[279.log](https://github.com/LibreELEC/LibreELEC.tv/files/7785903/279.log)

pretoriano80, 2021-12-29:
@heitbaum here are the stats for a RPi4 build (distclean).

```
Total Build Time: 02:06:27.664 (wall clock)
Accum Build Time: 08:25:50.601 (4 slots)                                                                                                           
                                                                                                                                                   
Breakdown by status (all slots):                                                                                                                   
                                                                                                                                                   
  Status   Usage         ( Pct )  Count  State                                                                                                     
  ACTIVE   05:47:29.951  (68.7%)  770    busy                                                                                                      
  FAILED     :  :00.000  (00.0%)  0                                                                                                                
  GETPKG     :  :00.377  (00.0%)  1      busy                                                                                                      
  IDLE     02:36:31.804  (30.9%)  284                                                                                                              
  LOCKED     :01:02.382  (00.2%)  859                                                                                                              
  MUTEX      :  :09.680  (00.0%)  6      busy                                                                                                      
  MUTEX/W    :  :00.000  (00.0%)  0      stall                                                                                                     
  STALLED    :  :00.057  (00.0%)  1      stall                                                                                                     
  UNLOCK     :  :36.351  (00.1%)  859                                                                                                              
  -------------------------------------                                                                                                            
  TOTAL    08:25:50.601  ( 100%)  2780                                                                                                             
                                                                                                                                                   
Peak concurrency: 4 out of 4 slots                                                                                                                 
                                                                                                                                                   
1 job slots were held in a "stall" state for 00:00:00.057                                                                                          
                                                                                                                                                   
Slot usage (time in a "busy" state):     | Concurrency breakdown ("busy"):                                                                         
                                         |                                                                                                         
#Rank  Slot  Usage        ( Pct )        | # of Slots  Usage        ( Pct )                                                                        
 #01    04   01:50:08.733 (21.8%)        |     01      02:06:24.022 (25.0%)
 #02    03   01:21:53.345 (16.2%)        |     02      01:19:38.936 (15.7%)
 #03    01   01:20:36.563 (15.9%)        |     03      01:15:29.973 (14.9%)
 #04    02   01:15:01.367 (14.8%)        |     04      01:06:07.077 (13.1%)
-----------------------------------------+---------------------------------
 TOTALS      05:47:40.008 (68.7%)                      05:47:40.008 (68.7%)
```

pretoriano80, 2021-12-29:
make clean build

```
Total Build Time: 00:51:50.896 (wall clock)                                                                                              
Accum Build Time: 03:27:23.527 (4 slots)                                                                                                           
                                                                                                                                                   
Breakdown by status (all slots):                                                                                                                   
                                                                                                                                                   
  Status   Usage         ( Pct )  Count  State                                                                                                     
  ACTIVE   02:36:21.504  (75.4%)  770    busy                                                                                                      
  FAILED     :  :00.000  (00.0%)  0                                                                                                                
  GETPKG     :  :00.000  (00.0%)  0      busy                                                                                                      
  IDLE       :49:24.780  (23.8%)  284                                                                                                              
  LOCKED     :  :53.655  (00.4%)  859                                                                                                              
  MUTEX      :  :06.520  (00.1%)  6      busy                                                                                                      
  MUTEX/W    :  :03.349  (00.0%)  2      stall                                                                                                     
  STALLED    :  :00.719  (00.0%)  2      stall                                                                                                     
  UNLOCK     :  :33.000  (00.3%)  859                                                                                                              
  -------------------------------------                                                                                                            
  TOTAL    03:27:23.527  ( 100%)  2782                                                                                                             
                                                                                                                                                   
Peak concurrency: 4 out of 4 slots                                                                                                                 
                                                                                                                                                   
4 job slots were held in a "stall" state for 00:00:04.068                                                                                          
                                                                                                                                                   
Slot usage (time in a "busy" state):     | Concurrency breakdown ("busy"):                                                                         
                                         |                                                                                                         
#Rank  Slot  Usage        ( Pct )        | # of Slots  Usage        ( Pct )                                                                        
 #01    02     :43:49.715 (21.1%)        |     01        :51:47.818 (25.0%)                                                                        
 #02    03     :41:42.319 (20.1%)        |     02        :44:16.309 (21.3%)                                                                        
 #03    01     :36:38.231 (17.7%)        |     03        :32:32.288 (15.7%)                                                                        
 #04    04     :34:17.758 (16.5%)        |     04        :27:51.609 (13.4%)                                                                        
-----------------------------------------+---------------------------------                                                                        
 TOTALS      02:36:28.023 (75.4%)                      02:36:28.023 (75.4%) 
```

heitbaum, 2021-12-29:
interesting stats. Not real zippy but not awful.

pretoriano80, 2021-12-30:
Yeah, it's not an outstanding VM, but it's free (or at least it should be after the trial ends), so i cant complain.