LibreELEC/LibreELEC.tv #6029: [BUG] docker logs going to journalctl

heitbaum, 2021-12-27:
Review docker configuration - as console also outputting to journalctl.

https://docs.docker.com/config/containers/logging/

lrusak, 2021-12-27:
Why do you think this is a bug?

https://github.com/LibreELEC/LibreELEC.tv/blob/140ad28a258167e0e87daf1e474db37215b2caf3/packages/addons/service/docker/source/system.d/service.system.docker.service#L12

heitbaum, 2021-12-28:
Hi @lrusak - then its strange - it wasn’t doing that before. `docker ubuntu bash; make image` All that output is going to both journald and my ssh session.

lrusak, 2021-12-28:
> Hi @lrusak - then its strange - it wasn’t doing that before. `docker ubuntu bash; make image` All that output is going to both journald and my ssh session.

Oh wait, do you mean the LibreELEC build container?

heitbaum, 2021-12-28:
[image]

Here is the image - yup - even weirder - now that I am at home and can see it on the big screen at my desk. need to dig in to this further ... maybe `systemd 250`

heitbaum, 2022-01-04:
Confirmed this is a change (regression between systemd 249.7 and 250)

heitbaum, 2022-01-04:
First part of the reading seems to indicate for “interactive shell” based containers:
`--log-driver none` should be the setting. 
- Test and update the https://github.com/LibreELEC/LibreELEC.tv/blob/master/tools/docker/README.md 
- more reading at https://docs.docker.com/config/containers/logging/configure/

closed 2022-01-23 by CvH