LibreELEC/LibreELEC.tv #6033: [BUILD ERROR][Generic/x86_64] Cross compile error on Ubuntu 20.4 aarch64

pretoriano80, 2021-12-28:
### Describe the bug
GRUB cross-compile fails when building Generic/x86_64 on Ubuntu 20.04 ARM64

### Informations
 - LE Version: LE11 - HEAD 140ad28a258167e0e87daf1e474db37215b2caf3
 - Hardware Platform: Generic/x86_64
 - OS: Ubuntu 20.4
 - Architecture: aarch64
 - CPU: Ampere A1
 - Clean build

### Log file
[279.log](https://github.com/LibreELEC/LibreELEC.tv/files/7786142/279.log)

### Additional context
Probably and hopefully this is the only error when cross-compile from aarch64 -> x86_64

heitbaum, 2021-12-28:
Try #5091

pretoriano80, 2021-12-28:
Nope, unfortunately it still fails with the same error.

As a side note, i tried to build GRUB outside of LibreELEC build environment and it fails with the same error. :(

heitbaum, 2021-12-29:
Can you please check if grub:host builds and it is just grub:target that errors?
the log shows grub:target - which I know doesn’t work right on target!=host as it uses grub as an executable. 

This is the log from my build host - see **grub(host)**

```
s/build grub
UNPACK      grub
    APPLY PATCH (common)      packages/tools/grub/patches/grub-gnulib-acprereq-2-64.patch
patching file configure.ac
    APPLY PATCH (common)      packages/tools/grub/patches/grub-gnulib-regcomp.patch
patching file grub-core/lib/gnulib-patches/fix-uninit-structure.patch
patching file grub-core/lib/gnulib-patches/fix-regcomp-uninit-token.patch
patching file bootstrap.conf
BUILD      grub (host)
    TOOLCHAIN      configure
```

pretoriano80, 2021-12-29:
I'm pretty sure grub:host builds fine and only grub:target fails, but i will double check.

pretoriano80, 2021-12-29:
@heitbaum is this "packages/tools/grub/patches/grub-gnulib-regcomp.patch" a patch that you are using on your local branch only?

heitbaum, 2021-12-29:
Part of my bump to 2.06 and gnulib “compatibility” https://github.com/heitbaum/LibreELEC.tv/commit/cc60a15b70

pretoriano80, 2021-12-29:
Double checked and only grub:target is failing.

heitbaum, 2022-01-01:
#5091 will fix the execution of the binary (to test) - but :target triplet needs to be investigated.

heitbaum, 2022-01-02:
I have fired up the `Ampere A1` and can reproduce the error. cant see the issue from a quick look, have updated to latest gnulib and grub (for the test - that has code issues.) will continue to look into it - in the background.

mglae, 2022-01-02:
Did you read the "Cross-compiling the GRUB" section of INSTALL?

Maybe BULD_CC, HOST_CC and TARGET_CC are not detected correctly.

heitbaum, 2022-01-02:
> Did you read the "Cross-compiling the GRUB" section of INSTALL?
> 
> Maybe BULD_CC, HOST_CC and TARGET_CC are not detected correctly.

I only had a quick look. Been “Clearing down my backlog of commits” 😎

heitbaum, 2022-03-09:
I believe the fix is as below for both building on x86_64 and aarch64:

```
commit 89eebaea1a7beefd9e9ac64d13d68a4cc549ec3d
Author: heitbaum <rudi@heitbaum.com>
Date:   Wed Mar 9 15:32:52 2022 +0000

    grub: use TARGET_ for build and set --disable-werror to allow build on non x86_64

diff --git a/packages/tools/grub/package.mk b/packages/tools/grub/package.mk
index db5c31f0a0..16975f3cbd 100755
--- a/packages/tools/grub/package.mk
+++ b/packages/tools/grub/package.mk
@@ -21,6 +21,9 @@ pre_configure_host() {
   unset LDFLAGS
   unset CPP
 
+  # when the build host is not x86_64; -Werror needs to be disabled
+  PKG_CONFIGURE_OPTS_HOST="--disable-werror"
+
   cd ${PKG_BUILD}
     # keep grub synced with gnulib
     ./bootstrap --gnulib-srcdir=$(get_build_dir gnulib) --copy --no-git --no-bootstrap-sync --skip-po
@@ -46,6 +49,13 @@ pre_configure_target() {
 
   mkdir -p .${TARGET_NAME}
     cd .${TARGET_NAME}
+
+  # configure requires explicit TARGET_PREFIX binaries when cross compiling.
+  export TARGET_CC="${TARGET_PREFIX}gcc"
+  export TARGET_OBJCOPY="${TARGET_PREFIX}objcopy"
+  export TARGET_STRIP="${TARGET_PREFIX}strip"
+  export TARGET_NM="${TARGET_PREFIX}nm"
+  export TARGET_RANLIB="${TARGET_PREFIX}ranlib"
 }
 
 make_target() {
```

@pretoriano80 - I haven’t run tested it yet. But build test works and the config.log is definitely cleaner.

pretoriano80, 2022-03-09:
Great, will test asap and report back.

pretoriano80, 2022-03-10:
@heitbaum i tested your PR #6289 but grub:target fails with this error:
```
/home/ubuntu/LibreELEC.tv/packages/tools/grub/package.mk: line 49: /home/ubuntu/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/grub-635ef55ed1252f92fe3bf70caefd185dcc507c43/grub-mkimage: cannot execute binary file: Exec format error
[1;31mFAILURE: scripts/build grub:target during makeinstall_target (package.mk)[0m
*********** FAILED COMMAND ***********
${PKG_BUILD}/grub-mkimage -d . -o bootia32.efi -O i386-efi -p /EFI/BOOT boot chain configfile ext2 fat linux search efi_gop efi_uga part_gpt gzio gettext loadenv loadbios memrw
**************************************
*********** FAILED COMMAND ***********
${SCRIPTS}/build "${1}" "${PARENT_PKG}"
**************************************
[1;31mFAILURE: scripts/install grub:target has failed![0m
```
However,if i use the package.mk from your dev branch (commit 9372e9b) and apply PR #6289 to it,then grub builds just fine.
This is the package.mk that works for me :

```
# SPDX-License-Identifier: GPL-2.0
# Copyright (C) 2016-present Team LibreELEC (https://libreelec.tv)

PKG_NAME="grub"
PKG_VERSION="635ef55ed1252f92fe3bf70caefd185dcc507c43" # 2020-12-18
PKG_SHA256="e099d18bdeef5312765f20c6de3384ca9ff03f776f1c7632b1e4bc4d1715a961"
PKG_ARCH="x86_64"
PKG_LICENSE="GPLv3"
PKG_SITE="https://www.gnu.org/software/grub/index.html"
PKG_URL="http://git.savannah.gnu.org/cgit/grub.git/snapshot/${PKG_NAME}-${PKG_VERSION}.tar.gz"
PKG_DEPENDS_HOST="toolchain:host"
PKG_DEPENDS_TARGET="toolchain flex freetype:host gettext:host grub:host"
PKG_DEPENDS_UNPACK="gnulib"
PKG_LONGDESC="GRUB is a Multiboot boot loader."
PKG_TOOLCHAIN="configure"

pre_configure_host() {
  unset CFLAGS
  unset CPPFLAGS
  unset CXXFLAGS
  unset LDFLAGS
  unset CPP

  # when the build host is not x86_64; -Werror needs to be disabled
  PKG_CONFIGURE_OPTS_HOST="--disable-werror"

  cd ${PKG_BUILD}
    # keep grub synced with gnulib
    ./bootstrap --gnulib-srcdir=$(get_build_dir gnulib) --copy --no-git --no-bootstrap-sync --skip-po

  mkdir -p .${HOST_NAME}
    cd .${HOST_NAME}
}

pre_configure_target() {
  PKG_CONFIGURE_OPTS_TARGET="--target=i386-pc-linux \
                             --disable-nls \
                             --with-platform=efi"

  unset CFLAGS
  unset CPPFLAGS
  unset CXXFLAGS
  unset LDFLAGS
  unset CPP

  cd ${PKG_BUILD}
    # keep grub synced with gnulib
    ./bootstrap --gnulib-srcdir=$(get_build_dir gnulib) --copy --no-git --no-bootstrap-sync --skip-po

  mkdir -p .${TARGET_NAME}
    cd .${TARGET_NAME}

  # configure requires explicit TARGET_PREFIX binaries when cross compiling.
  export TARGET_CC="${TARGET_PREFIX}gcc"
  export TARGET_OBJCOPY="${TARGET_PREFIX}objcopy"
  export TARGET_STRIP="${TARGET_PREFIX}strip"
  export TARGET_NM="${TARGET_PREFIX}nm"
  export TARGET_RANLIB="${TARGET_PREFIX}ranlib"
}

make_target() {
  make CC=${CC} \
       AR=${AR} \
       RANLIB=${RANLIB} \
       CFLAGS="-I${SYSROOT_PREFIX}/usr/include -fomit-frame-pointer -D_FILE_OFFSET_BITS=64" \
       LDFLAGS="-L${SYSROOT_PREFIX}/usr/lib"
}

makeinstall_target() {
  ${PKG_BUILD}/.${HOST_NAME}/grub-mkimage -d grub-core -o bootia32.efi -O i386-efi -p /EFI/BOOT \
    boot chain configfile ext2 fat linux search efi_gop \
    efi_uga part_gpt gzio gettext loadenv loadbios memrw

  mkdir -p ${INSTALL}/usr/share/grub
     cp -P bootia32.efi ${INSTALL}/usr/share/grub

  mkdir -p ${TOOLCHAIN}/share/grub
     cp -P bootia32.efi ${TOOLCHAIN}/share/grub
}
```

SupervisedThinking, 2022-03-10:
could you test master + https://github.com/LibreELEC/LibreELEC.tv/pull/5091 + https://github.com/LibreELEC/LibreELEC.tv/pull/6289 ? I guess the host / target build split is missing

pretoriano80, 2022-03-10:
@SupervisedThinking that worked (as expected) and grub built just fine.

Note: PR #6289 won't apply cleanly on top of #5091, so i guess that one of them will require a rebase.

heitbaum, 2022-03-10:
> @SupervisedThinking that worked (as expected) and grub built just fine.
> 
> Note: PR #6289 won't apply cleanly on top of #5091, so i guess that one of them will require a rebase.

Correct. I purposely left them stand-alone. 6289 fixes+corrects the build of the target using the cross compiler. Whereas 5091 fixes both the incompatibilities of host versus target.

thanks for testing

closed 2022-03-20 by CvH