LibreELEC/LibreELEC.tv #6035: [BUG] Latest master - No Screen - Raspberry Pi 3

coffeinflash, 2021-12-29:
After building the latest master Libreelec fails to start correctly. I see the bootlogo and then nothing. The machine is accessible via ssh.
The system is a Raspberry Pi 3 Model B Rev 1.2:

`48.054733] [drm:drm_crtc_commit_wait] *ERROR* flip_done timed out
[   48.054774] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [CRTC:72:crtc-3] commit wait timed out
[   58.078073] [drm:drm_crtc_commit_wait] *ERROR* flip_done timed out
[   58.078118] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [CONNECTOR:32:HDMI-A-1] commit wait timed out
[   68.108059] [drm:drm_crtc_commit_wait] *ERROR* flip_done timed out
[   68.108111] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [PLANE:66:plane-3] commit wait timed out
[   78.131381] [drm:drm_crtc_commit_wait] *ERROR* flip_done timed out
[   78.131422] vc4-drm soc:gpu: [drm] *ERROR* Timed out waiting for commit
[   88.158076] [drm:drm_atomic_helper_wait_for_flip_done] *ERROR* [CRTC:72:crtc-3] flip_done timed out
[   98.611409] [drm:drm_crtc_commit_wait] *ERROR* flip_done timed out
[   98.611455] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [CRTC:72:crtc-3] commit wait timed out
[  108.638046] [drm:drm_crtc_commit_wait] *ERROR* flip_done timed out
[  108.638081] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [PLANE:73:plane-4] commit wait timed out
[  118.664747] [drm:drm_crtc_commit_wait] *ERROR* flip_done timed out
[  118.664803] vc4-drm soc:gpu: [drm] *ERROR* Timed out waiting for commit
[  128.691392] [drm:drm_atomic_helper_wait_for_flip_done] *ERROR* [CRTC:72:crtc-3] flip_done timed out
[  138.931394] [drm:drm_crtc_commit_wait] *ERROR* flip_done timed out
[  138.931448] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [CRTC:72:crtc-3] commit wait timed out
[  148.958056] [drm:drm_crtc_commit_wait] *ERROR* flip_done timed out
[  148.958111] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [PLANE:73:plane-4] commit wait timed out
[  158.984723] [drm:drm_crtc_commit_wait] *ERROR* flip_done timed out
[  158.984784] vc4-drm soc:gpu: [drm] *ERROR* Timed out waiting for commit
[  169.011388] [drm:drm_atomic_helper_wait_for_flip_done] *ERROR* [CRTC:72:crtc-3] flip_done timed out
[  179.251392] [drm:drm_crtc_commit_wait] *ERROR* flip_done timed out
[  179.251446] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [CRTC:72:crtc-3] commit wait timed out
[  189.278052] [drm:drm_crtc_commit_wait] *ERROR* flip_done timed out
[  189.278119] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [PLANE:73:plane-4] commit wait timed out
[  199.304727] [drm:drm_crtc_commit_wait] *ERROR* flip_done timed out
[  199.304789] vc4-drm soc:gpu: [drm] *ERROR* Timed out waiting for commit
[  209.331391] [drm:drm_atomic_helper_wait_for_flip_done] *ERROR* [CRTC:72:crtc-3] flip_done timed out
[  219.571379] [drm:drm_crtc_commit_wait] *ERROR* flip_done timed out
[  219.571431] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [CRTC:72:crtc-3] commit wait timed out
[  229.601354] [drm:drm_crtc_commit_wait] *ERROR* flip_done timed out
[  229.601401] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [PLANE:73:plane-4] commit wait timed out
[  239.624710] [drm:drm_crtc_commit_wait] *ERROR* flip_done timed out
[  239.624767] vc4-drm soc:gpu: [drm] *ERROR* Timed out waiting for commit
[  249.651386] [drm:drm_atomic_helper_wait_for_flip_done] *ERROR* [CRTC:72:crtc-3] flip_done timed out
[  259.894705] [drm:drm_crtc_commit_wait] *ERROR* flip_done timed out
[  259.894755] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [CRTC:72:crtc-3] commit wait timed out
[  269.918032] [drm:drm_crtc_commit_wait] *ERROR* flip_done timed out
[  269.918080] [drm:drm_atomic_helper_wait_for_dependencies] *ERROR* [PLANE:73:plane-4] commit wait timed out
[  279.944738] [drm:drm_crtc_commit_wait] *ERROR* flip_done timed out
[  279.944797] vc4-drm soc:gpu: [drm] *ERROR* Timed out waiting for commit`

The output of dmesg is [here](https://paste.opensuse.org/78066956)
The system log can be found [here](https://paste.opensuse.org/61990887)
The hardware log can be found [here](https://paste.opensuse.org/18813435)

antonlacon, 2021-12-29:
It's an issue in the upstream RPi kernel. They know about it, but holidays. You can revert to an earlier kernel (I'm using ~5.15.5) in the mean time.

coffeinflash, 2021-12-29:
Thx, downgrading to linux 5.15.5 solved the issue. ([commit](https://github.com/LibreELEC/LibreELEC.tv/commit/40bdb650b811276043cfacde2a3887d4506a7224))

jmeloranta, 2022-01-09:
The nightly builds still seem to have this issue? It would be nice to have bootable nightly builds for pi 3.

HiassofT, 2022-01-09:
RPi3B+ works fine here with latest nightlies since #6089 was merged 2 days ago.

Please test with the latest nightly and close the issue or post a full pastekodi log if it's still present

coffeinflash, 2022-01-10:
Good morning,

I can confirm it is running fine again with the latest master.

closed 2022-01-10 by coffeinflash