LibreELEC/LibreELEC.tv #6042: [BUG] audio delayed

r2ixuz, 2021-12-30:
### Describe the bug
The audio track while playing a video file in LibreELEC 10.0.0.1 is about 140ms delayed.

### To Reproduce
Watch any video file in FullHD or 4K with AAC or any Dolby audio format. For example gunshots are visible before audio.

### Informations
 - LE Version: 10.0.1
 - Hardware Platform: RPi4 4GB RAM

### Additional context
Audio was in sync while using LibreELEC 9.2.8 and older versions. Since the upgrade to LibreELEC 10.0.1 audio is always delayed.

r2ixuz, 2021-12-30:
kodi.log after boot:

```
2021-12-26 20:25:26.633 T:1156     INFO <general>: -----------------------------------------------------------------------
2021-12-26 20:25:26.633 T:1156     INFO <general>: Starting Kodi (19.3 (19.3.0) Git:19.3-Matrix). Platform: Linux ARM 32-bit
2021-12-26 20:25:26.633 T:1156     INFO <general>: Using Release Kodi x32
2021-12-26 20:25:26.633 T:1156     INFO <general>: Kodi compiled 2021-10-29 by GCC 10.2.0 for Linux ARM 32-bit version 5.10.76 (330316)
2021-12-26 20:25:26.633 T:1156     INFO <general>: Running on BCM2835 with LibreELEC (official): 10.0.1, kernel: Linux ARM 64-bit version 5.10.76
2021-12-26 20:25:26.633 T:1156     INFO <general>: FFmpeg version/source: 4.3.2-Kodi
2021-12-26 20:25:26.633 T:1156     INFO <general>: 4 CPU cores available
2021-12-26 20:25:26.633 T:1156     INFO <general>: ARM Features: Neon enabled
2021-12-26 20:25:26.633 T:1156     INFO <general>: special://xbmc/ is mapped to: /usr/share/kodi/
2021-12-26 20:25:26.633 T:1156     INFO <general>: special://xbmcbin/ is mapped to: /usr/lib/kodi
2021-12-26 20:25:26.633 T:1156     INFO <general>: special://xbmcbinaddons/ is mapped to: /usr/lib/kodi/addons
2021-12-26 20:25:26.633 T:1156     INFO <general>: special://masterprofile/ is mapped to: /storage/.kodi/userdata
2021-12-26 20:25:26.633 T:1156     INFO <general>: special://envhome/ is mapped to: /storage
2021-12-26 20:25:26.633 T:1156     INFO <general>: special://home/ is mapped to: /storage/.kodi
2021-12-26 20:25:26.633 T:1156     INFO <general>: special://temp/ is mapped to: /storage/.kodi/temp
2021-12-26 20:25:26.633 T:1156     INFO <general>: special://logpath/ is mapped to: /storage/.kodi/temp
2021-12-26 20:25:26.633 T:1156     INFO <general>: The executable running is: /usr/lib/kodi/kodi.bin
2021-12-26 20:25:26.633 T:1156     INFO <general>: Local hostname: LibreELEC
2021-12-26 20:25:26.633 T:1156     INFO <general>: Log File is located: /storage/.kodi/temp/kodi.log
2021-12-26 20:25:26.634 T:1156     INFO <general>: -----------------------------------------------------------------------
2021-12-26 20:25:26.634 T:1156     INFO <general>: loading settings
2021-12-26 20:25:26.636 T:1156     INFO <general>: special://profile/ is mapped to: special://masterprofile/
2021-12-26 20:25:26.658 T:1156     INFO <general>: Loaded settings file from special://xbmc/system/advancedsettings.xml
2021-12-26 20:25:26.659 T:1156     INFO <general>: Contents of special://xbmc/system/advancedsettings.xml are...
                                                   <?xml version="1.0" encoding="utf-8" ?>
                                                   <advancedsettings version="1.0">
                                                     <cputempcommand>/usr/bin/cputemp</cputempcommand>
                                                     <gputempcommand>/usr/bin/gputemp</gputempcommand>
                                                     <showexitbutton>false</showexitbutton>
                                                     <remotedelay>1</remotedelay>
                                                     <samba>
                                                       <clienttimeout>30</clienttimeout>
                                                     </samba>
                                                     <fanartres>720</fanartres>
                                                     <imageres>540</imageres>
                                                   </advancedsettings>

2021-12-26 20:25:26.662 T:1156     INFO <general>: Loaded settings file from special://profile/advancedsettings.xml
2021-12-26 20:25:26.663 T:1156     INFO <general>: Contents of special://profile/advancedsettings.xml are...
                                                   <advancedsettings>
                                                     <cache>
                                                       <buffermode>1</buffermode>
                                                       <memorysize>256000000</memorysize>
                                                       <readfactor>10</readfactor>
                                                     </cache>
                                                     <video>
                                                       <ignoresecondsatstart>10</ignoresecondsatstart>
                                                     </video>
                                                   </advancedsettings>

2021-12-26 20:25:26.663 T:1156  WARNING <CSettingsManager>: missing version attribute
2021-12-26 20:25:26.664 T:1156     INFO <general>: Default Video Player: VideoPlayer
2021-12-26 20:25:26.664 T:1156     INFO <general>: Default Audio Player: paplayer
2021-12-26 20:25:26.664 T:1156     INFO <general>: Disabled debug logging due to GUI setting. Level 0.
2021-12-26 20:25:26.664 T:1156     INFO <general>: Log level changed to "INFO"
2021-12-26 20:25:26.664 T:1156     INFO <general>: CMediaSourceSettings: loading media sources from special://masterprofile/sources.xml
2021-12-26 20:25:26.685 T:1156     INFO <general>: creating subdirectories
2021-12-26 20:25:26.685 T:1156     INFO <general>: userdata folder: special://masterprofile/
2021-12-26 20:25:26.685 T:1156     INFO <general>: recording folder:
2021-12-26 20:25:26.685 T:1156     INFO <general>: screenshots folder: /storage/screenshots/
2021-12-26 20:25:26.773 T:1156     INFO <general>: Running database version Addons33
```

Isayso, 2022-02-21:
Don't know if this is connected, I found audio/video out of sync with any 60 Hz video on RPI3 from LE10 and LE11 on.

dschr, 2022-03-03:
On a rpi4B 8GB and the libreelec-versions 10.0, 10.0.0.1, as well as on the nightly-builds up to 28.Feb.2022 I experienced a delay between video & audio as well. After some seconds, video stutters and audio comes in sync again. Since it was/is not a stable delay time, it is not feasible to just use the audio-delay-setting.

The workaround for me was to switch from "direct-to-plane" to "EGL". With that, everything is in sync again. I am curious why direct-to-plane doesn't work.

Isayso, 2022-03-13:
> On a rpi4B 8GB and the libreelec-versions 10.0, 10.0.0.1, as well as on the nightly-builds up to 28.Feb.2022 I experienced a delay between video & audio as well. After some seconds, video stutters [...]

I have similar issues on RPI3/3+ from LE10 up to latest nightly LE11, frequent small audio stutter, after some minutes a tiny video/audio jump and audio comes in sync again. I play often music videos and that is the most annoying problem since Kodi19. The "EGL" work around does not work for me, I tried all options for audio/video screen refresh rates and so on.

r2ixuz, 2022-03-29:
Please open a new bug report for audio stutter. My issue is only with audio delayed (fixed intervall per file). But the delay is more, if higher bitrate files are played. So your workaround with EGL did not fixed my problem.

Besides the problem persists with LE 10.0.2.

r2ixuz, 2022-03-29:
Btw. what does the label "RPi-V4L2" mean?

Any comment to my initial bug report by the devs would be really nice after 3 month later... Thanks.

heitbaum, 2022-03-29:
> Btw. what does the label "RPi-V4L2" mean?
RaspberryPi Video 4 Linux 2
- a distinction of using the v4l2 decoding. Now that all LE is v4l2 - probably not so relevant (so read as RPi.)

heitbaum, 2023-09-04:
Closing as potentially fixed. If not - please reopen using current LE and updated logs.