LibreELEC/LibreELEC.tv #6079: [BUG] Settings add-on crashs at start LE11 nightly

CvH, 2022-01-03:
At least with LibreELEC-Generic.x86_64-11.0-nightly-20220101-9a279e4.tar and current master.

```
2022-01-03 21:48:49.678 T:825     ERROR <general>: SETTINGS: updates.get_hardware_flags_x86_64 # ValueError('not enough values to unpack (expected 2, got 1)')
2022-01-03 21:48:49.857 T:825     ERROR <general>: SETTINGS: updates.get_hardware_flags_x86_64 
*********************************** Unhandled exception detected ***********************************
####################################################################################################
                                           Diagnostic info
----------------------------------------------------------------------------------------------------
Exception type  : ValueError
Exception value : not enough values to unpack (expected 2, got 1)
System info     : uname_result(system='Linux', node='LibreELEC', release='5.15.8', version='#1 SMP Sat Jan 1 11:31:50 CET 2022', machine='x86_64', processor='')
Python version  : 3.8.12 (default, Jan  1 2022, 11:35:16)  [GCC 10.3.0]
Kodi version    : 20.0-ALPHA1 (19.90.101) Git:5633f5940067d57bff1395aaf2dd3a8536d0e437
sys.argv        : ['']
----------------------------------------------------------------------------------------------------
sys.path:
['/usr/share/kodi/addons/service.libreelec.settings',
 '/usr/lib/python38.zip',
 '/usr/lib/python3.8',
 '/usr/lib/python3.8/lib-dynload',
 '/usr/lib/python3.8/site-packages',
 '/usr/share/kodi/addons/service.libreelec.settings/resources/lib',
 '/usr/share/kodi/addons/service.libreelec.settings/resources/lib/modules']

####################################################################################################
                                            Stack Trace
====================================================================================================
File:
/usr/share/kodi/addons/service.libreelec.settings/service.py:110
----------------------------------------------------------------------------------------------------
Code context:
  106:         dbus_utils.LOOP_THREAD.stop()
  107: 
  108: 
  109: if __name__ == '__main__':
  110:>    Monitor().run()

----------------------------------------------------------------------------------------------------
Local variables:
Monitor = <class '__main__.Monitor'>
Service_Thread = <class '__main__.Service_Thread'>
dbus_utils = <module 'dbus_utils' from '/usr/share/kodi/addons/service.libreelec.settings/resources/lib/dbus_utils.pyc'>
log = <module 'log' from '/usr/share/kodi/addons/service.libreelec.settings/resources/lib/log.pyc'>
oe = <module 'oe' from '/usr/share/kodi/addons/service.libreelec.settings/resources/lib/oe.pyc'>
os = <module 'os' from '/usr/lib/python3.8/os.pyc'>
socket = <module 'socket' from '/usr/lib/python3.8/socket.pyc'>
sys = <module 'sys' (built-in)>
syspath = <module 'syspath' from '/usr/share/kodi/addons/service.libreelec.settings/syspath.py'>
threading = <module 'threading' from '/usr/lib/python3.8/threading.pyc'>
xbmc = <module 'xbmc' (built-in)>
xbmcout = <class '__main__.xbmcout'>
====================================================================================================
File:
/usr/share/kodi/addons/service.libreelec.settings/resources/lib/log.py:69
----------------------------------------------------------------------------------------------------
Code context:

----------------------------------------------------------------------------------------------------
Local variables:
arg = <__main__.Monitor object at 0x7fe1b12f7a20>
args = (<__main__.Monitor object at 0x7fe1b12f7a20>,)
function = <function Monitor.run at 0x7fe1b1388b80>
header = 'SETTINGS: Monitor.run '
kwargs = {}
level = 0
====================================================================================================
File:
/usr/share/kodi/addons/service.libreelec.settings/service.py:81
----------------------------------------------------------------------------------------------------
[... 1514 lines cut ...]
                                                                         'CM1/CM3',
                                                          'releases': {   '0': {   'file': {   'name': 'LibreELEC-Slice3.arm-9.2.6.tar',
                                                                                               'sha256': '5044a14d5e5f85b1d04aadb5bf137cac03aea42d52f69fe7fdea9166015aa0b9',
                                                                                               'size': '146647040'},
                                                                                   'image': {   'name': 'LibreELEC-Slice3.arm-9.2.6.img.gz',
                                                                                                'sha256': '9a8a7d4766ff73271422de52247368f8dfe6f5fa0504c4469fc010432ff44916',
                                                                                                'size': '132127924'}},
                                                                          '1': {   'file': {   'name': 'LibreELEC-Slice3.arm-9.2.8.tar',
                                                                                               'sha256': 'a67d1697b0121ab96913600c249d40d474138aa4c436ebaa0cdb2cecbc599a70',
                                                                                               'size': '146667520'},
                                                                                   'image': {   'name': 'LibreELEC-Slice3.arm-9.2.8.img.gz',
                                                                                                'sha256': '07739db11bedc843584be035948eeda4fb5d6bc0afb3b3aa5daae327735447c8',
                                                                                                'size': '132139883'}}}},
                                        'TinkerBoard.arm': {   'displayName': 'Rockchip '
                                                                              'RK3288',
                                                               'releases': {   '0': {   'file': {   'name': 'LibreELEC-TinkerBoard.arm-9.2.6-rk3288.tar',
                                                                                                    'sha256': 'e6999085f4b27337b66bb820a35d64b5a26031aa9bd551c8d52b9cd9004f45cd',
                                                                                                    'size': '159006720'},
                                                                                        'image': {   'name': 'LibreELEC-TinkerBoard.arm-9.2.6-rk3288.img.gz',
                                                                                                     'sha256': '49a3ac610258fc576ce2d92747f1fc0dc9263fe16cd264ce6a8a078b08a230cd',
                                                                                                     'size': '142871682'}}}}},
                         'url': 'http://releases.libreelec.tv/'}}
update_thread = <updateThread(Thread-2, started 140606978905664)>
************************************** End of object info ******************************************
```

mglae, 2022-01-04:
```
drv_path = ("ERROR: ld.so: object '/usr/lib/libtcmalloc_minimal.so' from LD_PRELOAD "
 'cannot be preloaded (cannot open shared object file): '
 'ignored./devices/pci0000:00/0000:00:02.0')
```
That is really weird. Who is defining LD_PRELOAD=/usr/lib/libtcmalloc_minimal.so?

1. pastekodi
2. What if you test a clean install with the run option of the installation stick?

CvH, 2022-01-09:
I think I messed up somehow, added random stuff to the kodi.conf (included some other).
Removing it fixed it.

closed 2022-01-09 by CvH