LibreELEC/LibreELEC.tv #6084: [BUG]Boot from emmc of RADXA Zero does not work

Karl-opec, 2022-01-04:
### Describe the bug
an actual image of LibreELEC-AMLGX.arm-11.0-devel-20220104183257-817e4a3-radxa-zero.img.gz does not boot
from the onboard emmc of the device. booting from an SD-Card works fine.

After booting from sdcard the onboard emmc ist fine and can be accessed.

### To Reproduce
Steps to reproduce the behavior:
1. flash the above image to emmc
2. flash the above image to SD-card
3. boot either with Sd card it boot from SD-Card . Boot without SD-Card get the following log on serial

### Informations
 - LE Version: 11 git 817e4a3
 - Hardware Platform: RADXA zero

### Log file
`?G12A:BL:0253b8:61aa2d;FEAT:E0F83180:2000;POC:F;RCY:0;EMMC:0;READ:0;0.0
                                                                          bl2_stage_init 0x01
bl2_stage_init 0x81
hw id: 0x0001 - pwm id 0x00
bl2_stage_init 0xc0
bl2_stage_init 0x02

L0:00000000
L1:00000703
L2:00008067
L3:04000000
S1:00000000
B2:00002000
B1:e0f83180

TE: 110455

BL2 Built : 15:21:48, Aug 28 2019. g12a g1bf2b53 - luan.yuan@droid15-sz

Board ID = 2
Set cpu clk to 24M
Set clk81 to 24M
CPU clk: 1200 MHz
Set clk81 to 166.6M
eMMC boot @ 0
sw8 s
DDR driver_vesion: LPDDR4_PHY_V_0_1_18 build time: Aug 28 2019 15:21:45
board id: 2
Load FIP HDR from eMMC, src: 0x00010200, des: 0xfffd0000, size: 0x00004000, part: 0
fw parse done
Load ddrfw from eMMC, src: 0x00060200, des: 0xfffd0000, size: 0x0000c000, part: 0
Load ddrfw from eMMC, src: 0x00038200, des: 0xfffd0000, size: 0x00004000, part: 0
PIEI prepare done
fastboot data load
00000000
emmc switch 1 ok
ddr saved addr:00016000
Load ddr parameter from eMMC, src: 0x02c00000, des: 0xfffd0000, size: 0x00001000, part: 0
00000000
emmc switch 0 ok
fastboot data verify
verify result: 267
Cfg max: 2, cur: 1. Board id: 255. Force loop cfg
LPDDR4 probe
ddr clk to 792MHz
Load ddrfw from eMMC, src: 0x0003c200, des: 0xfffd0000, size: 0x0000c000, part: 0

dmc_version 0000
Check phy result
INFO : End of CA training
INFO : End of initialization
INFO : Training has run successfully!
Check phy result
INFO : End of initialization
INFO : End of read enable training
INFO : End of fine write leveling
INFO : End of Write leveling coarse delay
INFO : Training has run successfully!
Check phy result
INFO : End of initialization
INFO : End of read dq deskew training
INFO : End of MPR read delay center optimization
INFO : End of write delay center optimization
INFO : End of read delay center optimization
INFO : End of max read latency training
INFO : Training has run successfully!
1D training succeed
Load ddrfw from eMMC, src: 0x00048200, des: 0xfffd0000, size: 0x0000c000, part: 0
Check phy result
INFO : End of initialization
INFO : End of 2D read delay Voltage center optimization
INFO : End of 2D read delay Voltage center optimization
INFO : End of 2D write delay Voltage center optimization
INFO : End of 2D write delay Voltage center optimization
INFO : Training has run successfully!

channel==0
RxClkDly_Margin_A0==236 ps 12
TxDqDly_Margin_A0==256 ps 13
RxClkDly_Margin_A1==256 ps 13
TxDqDly_Margin_A1==256 ps 13
TrainedVREFDQ_A0==74
TrainedVREFDQ_A1==74
VrefDac_Margin_A0==26
DeviceVref_Margin_A0==40
VrefDac_Margin_A1==25
DeviceVref_Margin_A1==40

channel==1
RxClkDly_Margin_A0==236 ps 12
TxDqDly_Margin_A0==256 ps 13
RxClkDly_Margin_A1==236 ps 12
TxDqDly_Margin_A1==256 ps 13
TrainedVREFDQ_A0==74
TrainedVREFDQ_A1==74
VrefDac_Margin_A0==26
DeviceVref_Margin_A0==40
VrefDac_Margin_A1==25
DeviceVref_Margin_A1==40

 dwc_ddrphy_apb_wr((0<<20)|(2<<16)|(0<<12)|(0xb0):0004 

soc_vref_reg_value 0x 00000014 00000016 00000015 00000016 00000014 00000016 00000014 00000014 00000015 00000012 00000016 00000014 00000015 00000015 00000015 00000015 00000016 00000016 00000014 00000016 00000015 00000017 00000015 00000015 00000017 00000014 00000015 00000016 00000016 00000015 00000015 00000016 dram_vref_reg_value 0x 0000005b
2D training succeed
aml_ddr_fw_vesion: LPDDR4_PHY_V_0_1_18 build time: Aug 28 2019 13:53:31
auto size-- 65535DDR cs0 size: 1024MB
DDR cs1 size: 1024MB
DMC_DDR_CTRL: 00e0001bDDR size: 2048MB
cs0 DataBus test pass
cs1 DataBus test pass
cs0 AddrBus test pass
cs1 AddrBus test pass

100bdlr_step_size ps== 471
result report
boot times 0Enable ddr reg access
00000000
emmc switch 3 ok
Authentication key not yet programmed
get rpmb counter error 0x00000007
00000000
emmc switch 0 ok
Load FIP HDR from eMMC, src: 0x00010200, des: 0x01700000, size: 0x00004000, part: 0
Load BL3X from eMMC, src: 0x0006c200, des: 0x0175c000, size: 0x000f0000, part: 0
bl2z: ptr: 05129330, size: 00001e40
0.0;M3 CHK:0;cm4_sp_mode 0
MVN_1=0x00000000
MVN_2=0x00000000
[Image: g12a_v1.1.3390-6ac5299 2019-09-26 14:09:46 luan.yuan@droid15-sz]
OPS=0x30
ring efuse init
28 0b 30 00 01 1f 0c 00 00 18 35 39 53 42 56 50 
[0.017150 Inits done]
secure task start!
high task start!
low task start!
run into bl31
NOTICE:  BL31: v1.3(release):4fc40b1
NOTICE:  BL31: Built : 15:57:33, May 22 2019
NOTICE:  BL31: G12A normal boot!
ERROR:   Error initializing runtime service opteed_fast

U-Boot 2021.10 (Jan 04 2022 - 19:04:43 +0100) radxa-zero

Model: Radxa Zero
SoC:   Amlogic Meson G12A (S905Y2) Revision 28:b (30:2)
DRAM:  2 GiB
MMC:   sd@ffe03000: 0, sd@ffe05000: 1, mmc@ffe07000: 2
Loading Environment from nowhere... OK
In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.
Hit any key to stop autoboot:  0 
Card did not respond to voltage select! : -110
Card did not respond to voltage select! : -110
unable to select a mode : -5
starting USB...
Bus usb@ff500000: Register 3000140 NbrPorts 3
Starting the controller
USB XHCI 1.10
scanning bus usb@ff500000 for devices... 1 USB Device(s) found
       scanning usb for storage devices... 0 Storage Device(s) found

Device 0: unknown device
No ethernet found.
missing environment variable: pxeuuid
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/00000000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0000000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/000000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/00000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/000
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/00
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/0
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default-arm-meson-u200
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default-arm-meson
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default-arm
No ethernet found.
missing environment variable: bootfile
Retrieving file: pxelinux.cfg/default
No ethernet found.
Config file not found
No ethernet found.
No ethernet found.
`
### Additional context

chewitt, 2022-01-05:
Please explain how you flashed the image to eMMC?

Karl-opec, 2022-01-05:
I installed it according the description here [Linux Install Radxa Wiki](https://wiki.radxa.com/Zero/install/eMMC) 
I worked as described.

Actually i use a work-around to run on libreelec on emmc. I renamed STORAGE on SD-card to STORAGE1.
After boot df gives following:

Filesystem           1K-blocks      Used Available Use% Mounted on
devtmpfs                549396         4    549392   0% /dev
tmpfs                   404504     12984    391520   3% /run
/dev/mmcblk1p1          524008    128960    395048  25% /flash
/dev/loop0              117760    117760         0 100% /
/dev/mmcblk1p2         6655471    169529   6481846   3% /storage
tmpfs                  1011252         0   1011252   0% /dev/shm
tmpfs                     4096         0      4096   0% /sys/fs/cgroup
tmpfs                  1011252         8   1011244   0% /var
tmpfs                  1011252         0   1011252   0% /tmp
/dev/mmcblk0p1          524008    128960    395048  25% /var/media/LIBREELEC
/dev/mmcblk0p2        28795415       392  28790927   0% /var/media/STORAGE1

So the emmc is working. It seams to me, that the Kernel 5.16-rc8 does initialize emmc correctly but u-boot
doesn't. 

The log lines:

Hit any key to stop autoboot: 0
Card did not respond to voltage select! : -110
Card did not respond to voltage select! : -110

remember  me to a similar issue with other Amlogic soc in the past.

chewitt, 2022-01-05:
Boot https://chewitt.libreelec.tv/testing/LibreELEC-AMLGX.arm-10.85.0-radxa-zero.img.gz from an SD card, then access the console (SSH or UART) and download/copy the same image to /storage .. then install the image to eMMC by running this command: `emmctool w /storage/LibreELEC-AMLGX.arm-10.85.0-radxa-zero.img.gz` .. then reboot and remove the SD card to boot the emmc installed image. Does that resolve anything?

Karl-opec, 2022-01-05:
I tried your proposal:
emmctool gave attached output:
[emmctool-che.txt](https://github.com/LibreELEC/LibreELEC.tv/files/7817516/emmctool-che.txt)

Result is the same. It does not boot from emmc.

Karl-opec, 2022-02-21:
In the meantime i figured out that [branch radxa-zero-2021.01](https://github.com/radxa/u-boot/tree/radxa-zero-v2021.01) from u-boot tree radxa, does boot from emmc alone (without SD-Card inserted). The Branch 2021.07 from Radxa and the mainline u-boot in LE don't boot. 

To make 2021.01 from Radxa boot, I had to enable LZO in u-boot config.

chewitt, 2022-02-21:
LE forces LZO at build time see https://github.com/LibreELEC/LibreELEC.tv/blob/master/projects/Amlogic/bootloader/config so that is not an issue. See https://forum.libreelec.tv/thread/25286-radxa-zero-mmc-issues/?postID=166818#post166818 which found a problem which is now corrected in my 5.17.y tree (which is now being used for nightly images).

Karl-opec, 2022-02-21:
I know that lzo is not the issue. You address an issue which is fixed in the kernel. This Issue here is regarding u-boot. 
so the actual nightly test images don't address the problem.

The boot log of the first post shows a non working boot on emmc with the u-boot 2021.10 from your test image at that time.
The compiled u-boot.bin 2021.01 (see attached log ) boots Libreelec from there. So there must be a bug in the radxa zero u-boot
from 2021.07 and newer.

[radxa-emmc-boot.log](https://github.com/LibreELEC/LibreELEC.tv/files/8109222/radxa-emmc-boot.log)

Karl-opec, 2022-02-21:
sorry, perhaps my English is not so clear.

I have flashed the latest test Image from your  [Repo](https://chewitt.libreelec.tv/testing/LibreELEC-AMLGX.arm-10.85.0-radxa-zero.img.gz) and flashed it to the emmc. It does not boot.

Then i flash my self compiled u-boot as described onto that  image from your repo and it then does boot.
The command to flash i used gives the same result with u-boot.bin.sd.bin vs. u-boot-bin.

Why this old Radxa u-boot boots and any newer either Radxa 2021.07 or any mainline u-boot does not boot, i don't know.
looking in the the git the commit fedce3ee1537efc9dbd49322660ece09a8da2591 from Jack Ma <jack@radxa.com> for radxa zero in 2021.01 and commit ee5ab114ee5acda8b484e1f4363b34039c38c87e  in 2021.07 from you are quite different. All additions in later u-boot commits (mainline) seems to be based on ee5ab114ee5acda8b484e1f4363b34039c38c87e.

heitbaum, 2024-03-16:
Closing as inactive. Happy to open again - but will need to confirm that this is using the upstream u-boot - info in https://github.com/LibreELEC/LibreELEC.tv/issues/6084#issuecomment-1046915975.

closed 2024-03-16 by heitbaum