LibreELEC/LibreELEC.tv #6093: [BUG] WIFI driver not loaded

majki09, 2022-01-07:
### Describe the bug
WIFI driver is not loaded automatically. _dmesg_ says that _Direct firmware load for brcm/brcmfmac43455-sdio.txt failed with error -2_. However, after copying this driver: _/lib/firmware/brcm/brcmfmac43455-sdio.bin_ to _/storage/.config/firmware/brcm/_ it's loading with success and after reboot WIFI is OK.

### To Reproduce
Steps to reproduce the behavior:
1. Boot libreelec (no WIFI).
2. Copy _/lib/firmware/brcm/brcmfmac43455-sdio.bin_ to _/storage/.config/firmware/brcm/_.
3. Reboot (then WIFI working OK).

### Informations
 - LE Version: [LibreELEC-Generic.x86_64-10.0.1](https://releases.libreelec.tv/LibreELEC-Generic.x86_64-10.0.1.img.gz)
 - Hardware Platform: [MINIX Z83-4 mini PC, Intel]

### Log file
http://ix.io/3Loo

### Additional context
none

chewitt, 2022-01-09:
This needs the `brcmfmc_sdio-firmware` package adding to `Generic` and I'd suggest we use a firmware .dat file to cherry-pick that single specific firmware and nvmram file to the image. See https://github.com/LibreELEC/LibreELEC.tv/blob/master/projects/Amlogic/config/kernel-firmware.dat as an example of how that's done (filename matches firmware package name).

heitbaum, 2024-03-11:
Closing as completed. Please advise if still an issue with latest LE12 20240311-86dba4b images

closed 2024-03-11 by heitbaum