LibreELEC/LibreELEC.tv #6099: [BUG] usb3 port not working on ROC-3328-CC with LibreELEC 10.0

yegarti, 2022-01-08:
### Describe the bug
The USB3.0 port on my Firefly ROC-RK3328-CC is not showing up on the machine.
I connected my 2.5 external HDD and I can hear it turning on but no light and it's not detected in the system.
Moving it to the USB2 port makes it work.

Cannot reproduce with LibreELEC 9.2.

### To Reproduce
Steps to reproduce the behavior:
1. Burn [RK3328-ROC-CC 10.0.1](https://releases.libreelec.tv/LibreELEC-RK3328.arm-10.0.1-roc-cc.img.gz) Image using LibreELEC USB SD creator.
2. Connect drive to usb3 port
3. nothing would be detected on system.

### Informations
 - LE Version: 10.0.1
 - Hardware Platform: FireFly-ROC-3328-CC 4GB
### Log file
LibreELEC 10.0:
None as I already burned my SD Card with previous version, if you can't reproduce I'll burn 10.0 again to get more logs.

LibreELEC 9.2 (no reproduction):
`http://ix.io/3Lvk`

### Additional context

This is the output when using LibreELEC 10.0:
Output from lsusb:
```
# lsusb
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
# dmesg | grep usb
[    0.162678] usbcore: registered new interface driver usbfs
[    0.162770] usbcore: registered new interface driver hub
[    0.162861] usbcore: registered new device driver usb
[    0.528057] dwc2 ff580000.usb: supply vusb_d not found, using dummy regulator
[    0.528412] dwc2 ff580000.usb: supply vusb_a not found, using dummy regulator
[    0.541573] dwc2 ff580000.usb: DWC OTG Controller
[    0.541636] dwc2 ff580000.usb: new USB bus registered, assigned bus number 1
[    0.541714] dwc2 ff580000.usb: irq 46, io mem 0xff580000
[    0.542072] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 5.10
[    0.542084] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.542102] usb usb1: Product: DWC OTG Controller
[    0.542122] usb usb1: Manufacturer: Linux 5.10.76 dwc2_hsotg
[    0.542132] usb usb1: SerialNumber: ff580000.usb
[    0.548571] ehci-platform ff5c0000.usb: EHCI Host Controller
[    0.548621] ehci-platform ff5c0000.usb: new USB bus registered, assigned bus number 2
[    0.548856] ehci-platform ff5c0000.usb: irq 47, io mem 0xff5c0000
[    0.561379] ehci-platform ff5c0000.usb: USB 2.0 started, EHCI 1.00
[    0.561838] usb usb2: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 5.10
[    0.561856] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.561870] usb usb2: Product: EHCI Host Controller
[    0.561882] usb usb2: Manufacturer: Linux 5.10.76 ehci_hcd
[    0.561893] usb usb2: SerialNumber: ff5c0000.usb
[    0.565242] ohci-platform ff5d0000.usb: Generic Platform OHCI controller
[    0.565304] ohci-platform ff5d0000.usb: new USB bus registered, assigned bus number 3
[    0.565536] ohci-platform ff5d0000.usb: irq 48, io mem 0xff5d0000
[    0.625698] usb usb3: New USB device found, idVendor=1d6b, idProduct=0001, bcdDevice= 5.10
[    0.625722] usb usb3: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.625746] usb usb3: Product: Generic Platform OHCI controller
[    0.625755] usb usb3: Manufacturer: Linux 5.10.76 ohci_hcd
[    0.625765] usb usb3: SerialNumber: ff5d0000.usb
[    0.629115] usbcore: registered new interface driver cdc_acm
[    0.629530] usbcore: registered new interface driver uas
[    0.629732] usbcore: registered new interface driver usb-storage
[    0.699695] usbcore: registered new interface driver usbhid
[... 116 lines cut ...]
[    1.598757] usbcore: registered new interface driver usbtouchscreen
[    1.652905] usbcore: registered new interface driver bfusb
[    1.653017] usbcore: registered new interface driver btusb
[    1.720945] usbcore: registered new interface driver usbhid
[    1.720954] usbhid: USB HID core driver
[    2.057572] usb 3-1: new full-speed USB device number 2 using ohci-platform
[    2.253002] usb 3-1: New USB device found, idVendor=045e, idProduct=07a5
[    2.253028] usb 3-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    2.253042] usb 3-1: Product: Microsoft® 2.4GHz Transceiver v9.0
[    2.253056] usb 3-1: Manufacturer: Microsoft
[    2.261141] input: Microsoft Microsoft® 2.4GHz Transceiver v9.0 as /devices/platform/ff5d0000.usb/usb3/3-1/3-1:1.0/0003:045E:07A5.0001/input/input1
[    2.315380] hid-generic 0003:045E:07A5.0001: input,hidraw0: USB HID v1.11 Keyboard [Microsoft Microsoft® 2.4GHz Transceiver v9.0] on usb-ff5d0000.usb-1/input0
[    2.323757] input: Microsoft Microsoft® 2.4GHz Transceiver v9.0 as /devices/platform/ff5d0000.usb/usb3/3-1/3-1:1.1/0003:045E:07A5.0002/input/input2
[    2.374441] hid-generic 0003:045E:07A5.0002: input,hidraw1: USB HID v1.11 Mouse [Microsoft Microsoft® 2.4GHz Transceiver v9.0] on usb-ff5d0000.usb-1/input1
[    2.398543] input: Microsoft Microsoft® 2.4GHz Transceiver v9.0 as /devices/platform/ff5d0000.usb/usb3/3-1/3-1:1.2/0003:045E:07A5.0003/input/input3
[    2.451303] hid-generic 0003:045E:07A5.0003: input,hidraw2: USB HID v1.11 Device [Microsoft Microsoft® 2.4GHz Transceiver v9.0] on usb-ff5d0000.usb-1/input2
[   47.084302] usb 5-1: new SuperSpeed USB device number 2 using xhci-hcd
[   47.100258] usb 5-1: New USB device found, idVendor=0bc2, idProduct=2312
[   47.100271] usb 5-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   47.100279] usb 5-1: Product: Expansion
[   47.100285] usb 5-1: Manufacturer: Seagate 
[   47.100292] usb 5-1: SerialNumber: NA4AXB5S
[   81.621035] usb 5-1: reset SuperSpeed USB device number 2 using xhci-hcd
[  184.271951] usb 3-1: USB disconnect, device number 2
```

chewitt, 2022-01-09:
I'd suggest you test an LE11 nightly before we spend too much time investigating LE10 problems. Any difference there?

yegarti, 2022-01-09:
> I'd suggest you test an LE11 nightly before we spend too much time investigating LE10 problems. Any difference there?

Issue doesn't reproduce with "LibreELEC-RK3328.arm-11.0-nightly-20220109-78fb7ee-roc-cc.img.gz".
I guess I can close this issue.

Thanks!

closed 2022-01-09 by yegarti