LibreELEC/LibreELEC.tv #6108: BUG: tools/mkpkg/update_binary-addons master updated from Nexus to Matrix

heitbaum, 2022-01-10:
### Checked the logs:
`tools/mkpkg/update_binary-addons master` did the wrong thing.

```
::::::::::::::
yy-output
::::::::::::::
[mkpkg] Checking out https://github.com/xbmc/repo-binary-addons.git ...
Resolved version for audiodecoder.2sf: Nexus => 20.2.0-Nexus
Resolved version for audiodecoder.asap: Nexus => 20.2.0-Nexus
Resolved version for audiodecoder.dumb: Nexus => 20.2.0-Nexus
Resolved version for audiodecoder.gme: Nexus => 20.2.0-Nexus
Resolved version for audiodecoder.gsf: Nexus => 20.2.0-Nexus
[mkpkg] Skipped audiodecoder.hvl
Resolved version for audiodecoder.modplug: Nexus => 20.2.0-Nexus
Resolved version for audiodecoder.ncsf: Nexus => 20.2.0-Nexus
Resolved version for audiodecoder.nosefart: Nexus => 20.2.0-Nexus
Resolved version for audiodecoder.openmpt: Nexus => 20.2.0-Nexus
Resolved version for audiodecoder.organya: Nexus => 20.2.0-Nexus
Resolved version for audiodecoder.qsf: Nexus => 20.2.0-Nexus
Resolved version for audiodecoder.sacd: Nexus => 20.2.0-Nexus
Resolved version for audiodecoder.sidplay: Nexus => 20.2.0-Nexus
Resolved version for audiodecoder.snesapu: Nexus => 20.2.0-Nexus
Resolved version for audiodecoder.ssf: Nexus => 20.2.0-Nexus
Resolved version for audiodecoder.stsound: Nexus => 20.2.0-Nexus
Resolved version for audiodecoder.timidity: Nexus => 20.2.0-Nexus
Resolved version for audiodecoder.upse: Nexus => 20.2.0-Nexus
Resolved version for audiodecoder.usf: Nexus => 20.2.0-Nexus
Resolved version for audiodecoder.vgmstream: Nexus => 20.2.0-Nexus
Resolved version for audiodecoder.wsr: Nexus => 20.2.0-Nexus
Resolved version for audioencoder.flac: Nexus => 20.2.0-Nexus
Resolved version for audioencoder.lame: Nexus => 20.3.0-Nexus
Resolved version for audioencoder.vorbis: Nexus => 20.2.0-Nexus
Resolved version for audioencoder.wav: Nexus => 20.2.0-Nexus
Resolved version for imagedecoder.heif: Nexus => 20.1.0-Nexus
Resolved version for imagedecoder.mpo: Nexus => 20.1.0-Nexus
Resolved version for imagedecoder.raw: Nexus => 20.1.0-Nexus
Resolved version for inputstream.adaptive: Nexus => 20.1.0-Nexus
Resolved version for inputstream.ffmpegdirect: Nexus => 20.1.0-Nexus
Resolved version for inputstream.rtmp: Nexus => 20.1.0-Nexus
Resolved version for peripheral.joystick: Nexus => 20.1.0-Nexus
Resolved version for peripheral.xarcade: Nexus => 20.1.0-Nexus
Resolved version for pvr.argustv: Nexus => 20.4.0-Nexus
UPDATED pvr.argustv from 20.3.0-Nexus to 20.4.0-Nexus
RESET pvr.argustv PKG_REV from 1 to 1
Resolved version for pvr.demo: Nexus => 20.4.2-Nexus
Resolved version for pvr.dvblink: Nexus => 20.2.0-Nexus
Resolved version for pvr.dvbviewer: Nexus => 20.2.0-Nexus
Resolved version for pvr.filmon: Nexus => 20.2.1-Nexus
Resolved version for pvr.freebox: Nexus => 20.2.0-Nexus
Resolved version for pvr.hdhomerun: Nexus => 20.2.1-Nexus
Resolved version for pvr.hts: Nexus => 20.2.0-Nexus
Resolved version for pvr.iptvsimple: Nexus => 20.2.1-Nexus
Resolved version for pvr.mediaportal.tvserver: Nexus => 20.2.0-Nexus
Resolved version for pvr.mythtv: Nexus => 20.2.0-Nexus
Resolved version for pvr.nextpvr: Nexus => 20.2.1-Nexus
Resolved version for pvr.njoy: Nexus => 20.2.0-Nexus
Resolved version for pvr.octonet: Nexus => 20.2.0-Nexus
Resolved version for pvr.pctv: Nexus => 20.2.0-Nexus
Resolved version for pvr.plutotv: Nexus => 20.1.0-Nexus
Resolved version for pvr.sledovanitv.cz: Nexus => 20.2.0-Nexus
Resolved version for pvr.stalker: Nexus => 20.2.0-Nexus
Resolved version for pvr.teleboy: Nexus => 20.2.0-Nexus
Resolved version for pvr.vbox: Nexus => 20.2.0-Nexus
Resolved version for pvr.vdr.vnsi: Nexus => 20.2.1-Nexus
Resolved version for pvr.vuplus: Nexus => 20.2.0-Nexus
Resolved version for pvr.waipu: Nexus => 20.3.0-Nexus
Resolved version for pvr.wmc: Nexus => 20.2.1-Nexus
Resolved version for pvr.zattoo: Nexus => 20.2.1-Nexus
Resolved version for screensaver.asteroids: Nexus => 20.1.0-Nexus
[mkpkg] Skipped screensaver.cpblobs
Resolved version for screensaver.shadertoy: Nexus => 20.1.0-Nexus
[mkpkg] Skipped screensavers.rsxs
Resolved version for vfs.libarchive: Nexus => 20.1.0-Nexus
Resolved version for vfs.rar: Nexus => 20.1.0-Nexus
Resolved version for vfs.sftp: Nexus => 20.1.0-Nexus
Resolved version for visualization.fishbmc: Nexus => 20.1.0-Nexus
Resolved version for visualization.goom: Nexus => 20.1.0-Nexus
Resolved version for visualization.matrix: Nexus => 20.1.0-Nexus
Resolved version for visualization.shadertoy: Nexus => 20.2.0-Nexus
Resolved version for visualization.spectrum: Matrix => 19.0.1-Matrix
UPDATED visualization.spectrum from 20.1.0-Nexus to 19.0.1-Matrix
RESET visualization.spectrum PKG_REV from 1 to 1
Resolved version for visualization.starburst: Matrix => 19.0.1-Matrix
UPDATED visualization.starburst from 20.1.0-Nexus to 19.0.1-Matrix
RESET visualization.starburst PKG_REV from 1 to 1
Resolved version for visualization.waveform: Matrix => 19.0.2-Matrix
UPDATED visualization.waveform from 20.1.0-Nexus to 19.0.2-Matrix
RESET visualization.waveform PKG_REV from 1 to 1

The following addons were skipped, please add packages for these addons:
[ audiodecoder.hvl screensaver.cpblobs screensavers.rsxs]

Resolved hash for audiodecoder.fluidsynth: master => 6c70bad814040007a180788d64d278736e9400fa
UPDATED audiodecoder.fluidsynth from 20.0.1-Nexus to 6c70bad814040007a180788d64d278736e9400fa
BUMPED audiodecoder.fluidsynth PKG_REV from 1 to 2
SKIPPING excluded package vfs.sacd
Resolved hash for visualization.pictureit: master => 4060888ba482e99e9c08753a385ecf164a9ac821
UPDATED visualization.pictureit from 20.0.0-Nexus to 4060888ba482e99e9c08753a385ecf164a9ac821
BUMPED visualization.pictureit PKG_REV from 1 to 2
::::::::::::::
yy-output.updated
::::::::::::::
kodi-binary-addons: update to latest versions

- pvr.argustv: update 20.3.0-Nexus to 20.4.0-Nexus
- visualization.spectrum: update 20.1.0-Nexus to 19.0.1-Matrix
- visualization.starburst: update 20.1.0-Nexus to 19.0.1-Matrix
- visualization.waveform: update 20.1.0-Nexus to 19.0.2-Matrix
- audiodecoder.fluidsynth: update 20.0.1-Nexus to 6c70bad814040007a180788d64d278736e9400fa
- visualization.pictureit: update 20.0.0-Nexus to 4060888ba482e99e9c08753a385ecf164a9ac821
```
### cat zz-push-binaries
```
#ls -1 ../LibreELEC.binaries/ | grep git$ | while read x
#do
#  ln -s ../LibreELEC.binaries/$x .
#done

tools/mkpkg/update_binary-addons master | tee yy-output
echo "kodi-binary-addons: update to latest versions" > yy-output.updated
echo >> yy-output.updated
cat yy-output | grep "^UPDATED " | sed 's/^UPDATED /- /' | sed 's/ from /: update /' >> yy-output.updated
git co -- packages/mediacenter/kodi-binary-addons/audiodecoder.fluidsynth/package.mk
git co -- packages/mediacenter/kodi-binary-addons/visualization.pictureit/package.mk
git commit -eF yy-output.updated .
```

_Originally posted by @heitbaum in https://github.com/LibreELEC/LibreELEC.tv/issues/6106#issuecomment-1008477582_

HiassofT, 2022-01-10:
This works fine here, both with github API token and without on plain LE master (githash 5037356cd8d)

with API token:
```
hias@delle:~/private/libreelec/libreelec-git$ git reset --hard HEAD
HEAD is now at 5037356cd8d Merge pull request #6111 from HiassofT/le11-retro-addons5
hias@delle:~/private/libreelec/libreelec-git$ GITHUB_API_TOKEN="<redacted>" ./tools/mkpkg/update_binary-addons master Nexus
[mkpkg] Checking out https://github.com/xbmc/repo-binary-addons.git ...
Resolved version for audiodecoder.2sf: Nexus => 20.2.0-Nexus
...
Resolved version for visualization.fishbmc: Nexus => 20.1.0-Nexus
Resolved version for visualization.goom: Nexus => 20.1.0-Nexus
Resolved version for visualization.matrix: Nexus => 20.1.0-Nexus
Resolved version for visualization.shadertoy: Nexus => 20.2.0-Nexus
Resolved version for visualization.spectrum: Nexus => 20.1.0-Nexus
Resolved version for visualization.starburst: Nexus => 20.1.0-Nexus
Resolved version for visualization.waveform: Nexus => 20.1.0-Nexus
...
```
"cassic" version without API token:
```
hias@delle:~/private/libreelec/le11-addon-update-tmp$ ../libreelec-git/tools/mkpkg/update_binary-addons master Nexus
[mkpkg] Checking out https://github.com/xbmc/xbmc ...
[mkpkg] Checking out https://github.com/xbmc/repo-binary-addons.git ...
[mkpkg] Checking out https://github.com/xbmc/audiodecoder.2sf ...
Resolved version for audiodecoder.2sf: Nexus => 20.2.0-Nexus
...
[mkpkg] Checking out https://github.com/xbmc/visualization.fishbmc ...
Resolved version for visualization.fishbmc: Nexus => 20.1.0-Nexus
[mkpkg] Checking out https://github.com/xbmc/visualization.goom ...
Resolved version for visualization.goom: Nexus => 20.1.0-Nexus
[mkpkg] Checking out https://github.com/xbmc/visualization.matrix ...
Resolved version for visualization.matrix: Nexus => 20.1.0-Nexus
[mkpkg] Checking out https://github.com/xbmc/visualization.shadertoy ...
Resolved version for visualization.shadertoy: Nexus => 20.2.0-Nexus
[mkpkg] Checking out https://github.com/xbmc/visualization.spectrum ...
Resolved version for visualization.spectrum: Nexus => 20.1.0-Nexus
[mkpkg] Checking out https://github.com/xbmc/visualization.starburst ...
Resolved version for visualization.starburst: Nexus => 20.1.0-Nexus
[mkpkg] Checking out https://github.com/xbmc/visualization.waveform ...
Resolved version for visualization.waveform: Nexus => 20.1.0-Nexus
...
```

heitbaum, 2022-02-17:
Closing - not reproducible.

closed 2022-02-17 by heitbaum