LibreELEC/LibreELEC.tv #6109: [BUG] Logitech K400 keyboard types by itself (looks like a driver issue?)

motosba, 2022-01-10:
### Describe the bug
Logitech K400 keyboard types by itself

### To Reproduce
Steps to reproduce the behavior:
1. Plug in a wireless Logitech K400+ keyboard combo
2. Start playback
3. Pause playback with space
4. Playback will resume by itself after a while

### Informations
 - LE Version: 10.0.1
 - Hardware Platform: RPI4b

### Log file
N/A

### Additional context
Looks like a driver not issue. This only happens with this keyboard. I don't have issues with the keyboard while using it for other purposes.

eldersixpack13, 2022-01-10:
how long is "a while" ?

tried to reproduce it on nightly and an Intel nuc8 with an 2 month old k400 plus: movies keeps stopped

heitbaum, 2022-02-17:
@motosba - please reopen if this is still an issue - and provide an update.

closed 2022-02-17 by heitbaum