LibreELEC/LibreELEC.tv #6112: [BUG] Libreelec 10.0.1 Raspberry PI 4: Raspberry PI touchscreen drivers missing

andycockerill-cpi, 2022-01-11:
### Describe the bug
Using the latest 10.0.1 with a raspberry PI 4 + raspberry pi touchscreen connected using a ribbon cable, the display is white.
### To Reproduce
Steps to reproduce the behavior:
1. Flash 10.0.1 for a raspberry PI 4
2. Connect a raspberry PI touchscreen to the PI 4
3. Attach power cable
4. The Raspberry PI flash screen is displayed, followed by a white screen (no display).

### Informations
 - LE Version: 10.0.1
 - Hardware Platform: RPI4 + RPI touchscreen

### Log file
I can't connect using SSH, so no log file.

### Additional context
The drivers are present in 9.2.8

HiassofT, 2022-01-12:
This should already be fixed in the current LE 10 branch and will be included in 10.0.2.

Please test with this build https://www.horus.com/~hias/tmp/libreelec/LibreELEC-RPi4.arm-10.0-devel-20211218214422-480810e.tar 

Also have a look at this thread on our forums https://forum.libreelec.tv/thread/24349-dtoverlay-setting-in-config-txt-preventing-libreelec-10-rc1-from-starting/?postID=162930#post162930

heitbaum, 2022-02-17:
Closing as actioned. @andycockerill-cpi Please reopen if this is still an issue.

HiassofT, 2022-05-03:
Please provide more info:

Have you added the necessary settings to config.txt as in the linked forum post?

Add "ssh" to the end of cmdline.txt (on the same line after "quiet"), then boot the RPi and ssh in and run "pastekodi" and "modetest" and post the URLs you got

andycockerill-cpi, 2022-05-03:
Using 10.0.0.2 with no changes results in no touchscreen display.

Adding this to config.txt (after "include distroconfig.txt")
dtoverlay=vc4-kms-dsi-7inch
dtparam=audio=on

Results in a working display.

So this is not fixed in 10.0.0.2 unless manual changes are made to config.txt, then a workaround is possible.

HiassofT, 2022-05-03:
Thanks for reporting back!

The dtoverlay line is required on all kernels running kms, same as on RPiOS bullseye, and it's expected behaviour to get no output on the DSI screen without it.

closed 2022-05-03 by HiassofT

popcornmix, 2022-05-20:
Actually PRiOS bullseye includes `display_auto_detect=1`
That results in `dtoverlay=vc4-kms-dsi-7inch` being loaded automatically when detected.

Might be worth adding `display_auto_detect=1` to distroconfig.txt

HiassofT, 2022-05-20:
Thanks for the heads up, didn't know about that option before.

@andycockerill-cpi can you please test if adding display_auto_detect=1 to config.txt (instead of dtoverlay=vc4-kms-dsi-7inch) works on LE? I don't have a 7" display so can't test myself. If you can confirm it works fine then I'll include it in future LE versions.

andycockerill-cpi, 2022-05-23:
OK, so my config.txt now looks like:

[all]
include distroconfig.txt
#dtoverlay=vc4-kms-dsi-7inch
display_auto_detect=1
dtparam=audio=on
# uncomment to enable infrared remote recevier connected to GPIO 18
#dtoverlay=gpio-ir,gpio_pin=18

I can confirm that the display still works.

HiassofT, 2022-05-23:
Thanks a lot for testing and reporting back!

I've created PRs for master #6517 and LE10 #6518