LibreELEC/LibreELEC.tv #6164: [BUG] ffmpeg build fails on master

mjhammel, 2022-01-25:
### Describe the bug
I'm using ffmpeg's package.mk + patches to manually build ffmpeg for rk3399.  I download the 4.4-N-Alpha1 tarball, apply the libreelec, v4l2-request and v4l2-drmprime patches in that order.  I use the same configure options for default configs (not rpi, not amlogic) sans debug, VDPAU, and RPI.  The build fails with multiple error messages like these, all of which are in libavcoded/v4l2_request_mpeg2.c.

 /libavcodec/v4l2_request_mpeg2.c:40:26: note: (near initialization for '(anonymous)')
src/libavcodec/v4l2_request_mpeg2.c:41:10: error: 'struct v4l2_ctrl_mpeg2_sequence' has no member named 'vbv_buffer_size'
   41 |         .vbv_buffer_size = req->output.size,

src/libavcodec/v4l2_request_mpeg2.c:49:37: error: 'V4L2_MPEG2_SEQ_FLAG_PROGRESSIVE' undeclared (first use in this function); did you mean 'V4L2_DV_BT_CAP_PROGRESSIVE'?
   49 |         controls->sequence.flags |= V4L2_MPEG2_SEQ_FLAG_PROGRESSIVE;

The 5.16 kernel has definitions for V4L2_MPEG2_SEQ_FLAG_PROGRESSIVE and others that the build fails on in v4l2-controls.h, but the same file in ffmpeg's tree does not have these.  I suspect there is something missing in the v4l2-request patch.

### To Reproduce

- Download tarball as listed in master's ffmpeg package.mk
- Apply specified patches
- Compile with specific config options.

### Informations
 - LE Version: master
 - Hardware Platform: rk3399

### Log file
https://www.graphics-muse.org/archives/source/ffmpeg/ffmpeg-libreelec.txt

### Additional context
I'm attempting to reproduce master's build of ffmpeg in order to update rk3399 H.264 decode support from a 5.10.x version I currently have working to 5.15.14.  Once that decode is working I intend to backport a 5.18 patch for MJPEG encoding for the hantro and see if ffmpeg will "just work" with it.

mjhammel, 2022-01-25:
Okay, this may be a build problem on my end - there is no v4l2-controls.h in ffmpeg so it must be picking up the kernel-headers.  I must be pointed at the wrong kernel source.  At least I assume that's what's wrong here.

jernejsk, 2022-01-25:
This has nothing to do with LE, nightly images are built successfully. BTW, MPEG2 requires at least kernel 5.14.

closed 2022-01-25 by jernejsk