LibreELEC/LibreELEC.tv #6180: [BUG] Collated x86 GBM issues

heitbaum, 2022-02-01:
https://forum.libreelec.tv/thread/25212-kodi-flickering-tv-menu/?postID=166247

Who:
- heitbaum - NUC11PAHi7 (TGL)
- JoeAverage - NUC8i3BEH
- vhgroup - NUC5i3RYK
- CvH

What is erroring:
- Jitter menu on side - during playback or PVR
  - NUC5i3RYK

CvH, 2022-02-01:
audio dropouts at the menu while playing something
```
2022-02-01 10:02:03.310 T:7326  WARNING <general>: OutputPicture - timeout waiting for buffer
2022-02-01 10:02:10.272 T:7324     INFO <general>: Skipped 133 duplicate messages..
2022-02-01 10:02:10.272 T:7324     INFO <general>: CVideoPlayerAudio::Process - stream stalled
2022-02-01 10:02:10.284 T:7326  WARNING <general>: OutputPicture - timeout waiting for buffer
2022-02-01 10:02:10.324 T:11373    INFO <general>: COutput::OnStartup: Output Thread created
2022-02-01 10:02:10.393 T:776      INFO <general>: VAAPI::Close
2022-02-01 10:02:10.393 T:7327     INFO <general>: COutput::OnExit: Output Thread terminated
2022-02-01 10:02:10.915 T:7326  WARNING <general>: OutputPicture - timeout waiting for buffer
2022-02-01 10:02:19.564 T:7324     INFO <general>: Skipped 133 duplicate messages..
2022-02-01 10:02:19.564 T:7324     INFO <general>: CVideoPlayerAudio::Process - stream stalled
2022-02-01 10:02:19.596 T:7326  WARNING <general>: OutputPicture - timeout waiting for buffer
2022-02-01 10:02:19.599 T:11408    INFO <general>: COutput::OnStartup: Output Thread created
2022-02-01 10:02:19.693 T:776      INFO <general>: VAAPI::Close
2022-02-01 10:02:19.693 T:11373    INFO <general>: COutput::OnExit: Output Thread terminated
2022-02-01 10:02:20.212 T:7326  WARNING <general>: OutputPicture - timeout waiting for buffer
2022-02-01 10:02:28.671 T:7324     INFO <general>: Skipped 135 duplicate messages..
2022-02-01 10:02:28.671 T:7324     INFO <general>: CVideoPlayerAudio::Process - stream stalled
2022-02-01 10:02:28.695 T:7326  WARNING <general>: OutputPicture - timeout waiting for buffer
2022-02-01 10:02:28.725 T:11441    INFO <general>: COutput::OnStartup: Output Thread created
2022-02-01 10:02:28.773 T:776      INFO <general>: VAAPI::Close
2022-02-01 10:02:28.773 T:11408    INFO <general>: COutput::OnExit: Output Thread terminated
```

CvH, 2022-02-01:
video flickering while switching from the menu to a channel
```
022-02-01 10:03:57.731 T:776      INFO <general>: VAAPI::Close - closing decoder context
2022-02-01 10:03:57.732 T:1543     INFO <general>: Deleting audio codec
2022-02-01 10:03:57.732 T:1543     INFO <general>: Closing stream player 3
2022-02-01 10:03:57.732 T:1543     INFO <general>: Closing stream player 4
2022-02-01 10:03:57.732 T:1543     INFO <general>: waiting for teletext data thread to exit
2022-02-01 10:03:57.732 T:7325    ERROR <general>: Got MSGQ_ABORT or MSGO_IS_ERROR return true (-1)
2022-02-01 10:03:57.732 T:7325     INFO <general>: thread end: data_thread
2022-02-01 10:03:57.738 T:776      INFO <general>: GLES: Selecting YUV 2 RGB shader
2022-02-01 10:03:57.769 T:800      INFO <general>: CActiveAESink::OpenSink - initialize sink
2022-02-01 10:03:57.986 T:800      INFO <general>: CAESinkALSA::Initialize - Attempting to open device "@:CARD=U5,DEV=0"
2022-02-01 10:03:57.991 T:800      INFO <general>: CAESinkALSA::Initialize - Opened device "surround51:CARD=U5,DEV=0"
2022-02-01 10:03:57.991 T:800      INFO <general>: CAESinkALSA::InitializeHW - Your hardware does not support AE_FMT_FLOAT, trying other formats
2022-02-01 10:03:57.991 T:800      INFO <general>: CAESinkALSA::InitializeHW - Using data format AE_FMT_S24NE3
2022-02-01 10:03:58.449 T:1543     INFO <general>: Opening stream: 1001 source: 256
2022-02-01 10:03:58.449 T:1543     INFO <general>: [WHITELIST] Searching the whitelist for: width: 1280, height: 720, fps: 50.000, 3D: false
2022-02-01 10:03:58.450 T:1543     INFO <general>: Display resolution ADJUST : 1920x1080 @ 50.000000 Hz (31) (weight: 194720.875)
2022-02-01 10:03:58.470 T:1543  WARNING <general>: OpenStream - Unsupported stream 1001. Stream disabled.
2022-02-01 10:03:58.470 T:1543     INFO <general>: Opening stream: 1004 source: 256
2022-02-01 10:03:58.470 T:1543     INFO <general>: Finding audio codec for: 86019
2022-02-01 10:03:58.471 T:1543     INFO <general>: CDVDAudioCodecFFmpeg::Open() Successful opened audio decoder ac3
2022-02-01 10:03:58.471 T:1543     INFO <general>: Creating audio thread
2022-02-01 10:03:58.471 T:11828    INFO <general>: running thread: CVideoPlayerAudio::Process()
2022-02-01 10:03:58.471 T:1543     INFO <general>: Opening stream: 1007 source: 256
2022-02-01 10:03:58.471 T:1543     INFO <general>: Opening stream: 1006 source: 256
2022-02-01 10:03:58.471 T:1543     INFO <general>: waiting for teletext data thread to exit
2022-02-01 10:03:58.472 T:1543     INFO <general>: Creating teletext data thread
2022-02-01 10:03:58.472 T:11829    INFO <general>: running thread: CDVDTeletextData
2022-02-01 10:03:58.474 T:776      INFO <general>: [WHITELIST] Searching the whitelist for: width: 1280, height: 720, fps: 50.000, 3D: false
2022-02-01 10:03:58.475 T:776      INFO <general>: Display resolution ADJUST : 1920x1080 @ 50.000000 Hz (31) (weight: 0.000)
2022-02-01 10:03:58.496 T:1543     INFO <general>: Opening stream: 1001 source: 256
2022-02-01 10:03:58.496 T:1543     INFO <general>: [WHITELIST] Searching the whitelist for: width: 1280, height: 720, fps: 50.000, 3D: false
2022-02-01 10:03:58.496 T:1543     INFO <general>: Display resolution ADJUST : 1920x1080 @ 50.000000 Hz (31) (weight: 194720.875)
2022-02-01 10:03:58.510 T:1543     INFO <general>: Creating video codec with codec id: 27
2022-02-01 10:03:58.510 T:1543     INFO <general>: CDVDVideoCodecFFmpeg::Open() Using codec: H.264 / AVC / MPEG-4 AVC / MPEG-4 part 10
2022-02-01 10:03:58.510 T:1543     INFO <general>: Creating video thread
2022-02-01 10:03:58.510 T:11830    INFO <general>: running thread: video_thread
2022-02-01 10:03:58.510 T:1543     INFO <general>: Opening stream: 1004 source: 256
2022-02-01 10:03:58.510 T:1543     INFO <general>: Closing stream player 3
2022-02-01 10:03:58.511 T:1543     INFO <general>: Opening stream: 1007 source: 256
2022-02-01 10:03:58.511 T:1543     INFO <general>: Opening stream: 1006 source: 256
2022-02-01 10:03:58.511 T:1543     INFO <general>: Opening stream: 1001 source: 256
2022-02-01 10:03:58.511 T:1543     INFO <general>: [WHITELIST] Searching the whitelist for: width: 1280, height: 720, fps: 50.000, 3D: false
2022-02-01 10:03:58.512 T:11828    INFO <general>: CDVDAudioCodecFFmpeg::Open() Successful opened audio decoder ac3
2022-02-01 10:03:58.512 T:11828    INFO <general>: Creating audio stream (codec id: 86019, channels: 2, sample rate: 48000, no pass-through)
2022-02-01 10:03:58.512 T:1543     INFO <general>: Display resolution ADJUST : 1920x1080 @ 50.000000 Hz (31) (weight: 194720.875)
2022-02-01 10:03:58.515 T:776      INFO <general>: [WHITELIST] Searching the whitelist for: width: 1280, height: 720, fps: 50.000, 3D: false
2022-02-01 10:03:58.515 T:11831    INFO <general>: COutput::OnStartup: Output Thread created
2022-02-01 10:03:58.515 T:776      INFO <general>: Display resolution ADJUST : 1920x1080 @ 50.000000 Hz (31) (weight: 0.000)
2022-02-01 10:03:58.530 T:1543     INFO <general>: Opening stream: 1004 source: 256
2022-02-01 10:03:58.530 T:1543     INFO <general>: Finding audio codec for: 86019
2022-02-01 10:03:58.530 T:1543     INFO <general>: CDVDAudioCodecFFmpeg::Open() Successful opened audio decoder ac3
2022-02-01 10:03:58.530 T:1543     INFO <general>: Closing stream player 3
2022-02-01 10:03:58.530 T:1543     INFO <general>: Opening stream: 1007 source: 256
2022-02-01 10:03:58.530 T:1543     INFO <general>: Opening stream: 1006 source: 256
[... cut ...]
2022-02-01 10:03:58.753 T:800      INFO <general>: CAESinkALSA::InitializeHW - Your hardware does not support AE_FMT_FLOAT, trying other formats
2022-02-01 10:03:58.753 T:800      INFO <general>: CAESinkALSA::InitializeHW - Using data format AE_FMT_S24NE3
2022-02-01 10:03:58.913 T:11830 WARNING <general>: OutputPicture - timeout waiting for buffer
2022-02-01 10:04:00.194 T:11830    INFO <general>: Skipped 2 duplicate messages..
2022-02-01 10:04:00.194 T:11830    INFO <general>: CDVDVideoCodecFFmpeg::CDropControl: calculated diff time: 21818
```

eldersixpack13, 2022-02-01:
me is JoeAverage !

**audio sync error**
happened suddently to me today and usually do not suffering from it.
was watching TV for some minutes and suddenly the audio sync error occured.
stop and restart the channel again, all was fine again (means: no obvious audio sync errors on the same channel or others)

debug was off (will try to get a log with debug on)

[audiosync_error.txt](https://github.com/LibreELEC/LibreELEC.tv/files/7982567/audiosync_error.txt)

eldersixpack13, 2022-02-01:
**flickering during channel switching**

remote/keyboard is this:
https://www.orbsmart.de/produkte/orbsmart-am-1/

channel switched via up/down-key at 00:37:04.791 without flickering
channel switched via number key pad at 00:38:08.719 with flickering

debug was on
[kodi.log](https://github.com/LibreELEC/LibreELEC.tv/files/7982634/kodi.log)

smp79, 2022-02-02:
Flickering is fixed after reverting [this commit](https://github.com/xbmc/xbmc/commit/6f3a6fc2f737671f830f06748962083e9648cfcc).

https://github.com/xbmc/xbmc/pull/18812#issuecomment-734903719

eldersixpack13, 2022-02-03:
ping @heitbaum 
wouldn't it be better to move the audio sync bug to another thread ?
as said it seem the audio sync bug is a tricky one ${when "moon phases and weather" met}
at least here

@CvH comments ?

smp79, 2022-02-03:
"ActiveAE - large audio sync error" usually means there was some sort of issue with the transport stream. This is not something specific to Generic/GBM.

eldersixpack13, 2022-02-03:
@smp79 
quiet clear to me.
that's why I suggested to split "audio sync error" out to another bug report, even when I assume that the flickering bug will fixed soon.

@CvH is to blame ;-)
*he* misled me to post the audio sync error here, too

eldersixpack13, 2022-02-03:
it seems "moon phase and weather" met...

opened a new report:
https://github.com/LibreELEC/LibreELEC.tv/issues/6187

alas can't upload the kodi.log (1.1 MB)

idea's ?

CvH, 2022-02-04:
> that's why I suggested to split "audio sync error" out to another bug report

not sure thats something different, my bug happens after I switched from any video back to the menu, ~5secs later the audio drops out - wasn't happening with x11 build (didn't tested a recent one)

eldersixpack13, 2022-02-04:
@CvH which "menu" is meant, channel menu ?

eldersixpack13, 2022-04-15:
audio sync Error and Flickering seems to be fixed (never seen again since some longer time now)
therefore I closed my separat opened audio sync bug report (#6187) on request of @heitbaum today

@CvH, what's up with your issues here ?

CvH, 2022-05-07:
@eldersixpack13 just ran the x11 builds without any problems for ~2 months now, switched to yesterdays gbm build and the issues appear right away again :( 
It worked for ~30min perfect, then the issues happen every 2-3min, even after reboot. I am not sure if maybe a thermic aspect results in that problem ? Could be pure coincidence, need to run it a bit more.

smp79, 2022-05-08:
I tried the current master on a Tiger Lake based Acer EN2580 nettop and came across an annoying issue.
Sometimes when trying to switch from 1080p to a 4K video mode it fails. The top left corner of the image will become zoomed to a full screen and it stays like that until I restart Kodi. 
This shows up in the log when the failure occur:
`CDRMAtomic::DrmAtomicCommit - test commit failed: (Invalid argument)`

Never seen anything like that on my Gemini Lake box.

@heitbaum have you experienced a similar problem on your TGL-based NUC?

heitbaum, 2022-05-08:
> I tried the current master on a Tiger Lake based Acer EN2580 nettop and came across an annoying issue. Sometimes when trying to switch from 1080p to a 4K video mode it fails. The top left corner of [...]

I have seen it. Not for while though. I’ll put a note to check the log next time it happens.

heitbaum, 2022-05-08:
Note: End user loosing audio issue 
- https://forum.libreelec.tv/thread/25590-intel-nuc-too-slow-looking-for-an-other-device/?postID=169466#post169466
- Wonder if it is the same issue..

smp79, 2022-06-02:
> I have seen it.

I figured out how to reproduce. Opened [this kodi issue](https://github.com/xbmc/xbmc/issues/21504). This seem to be affecting only Tiger Lake.

CvH, 2022-09-11:
just a heads up, the same issue with gbm with todays build - just that it took longer to actually happen (no idea how that works out)
x11 still works like a charm

eldersixpack13, 2022-09-11:
I'm running nightly-20220910-f329064 with GBM (and GBM since month)
I sporadic see *some* of this error's "ActiveAE - large audio sync error: -89441.302337", when I *explicit* look in my log files, but I don't notice them when running SAT-TV, means: No hickup or such.

even on my worst SAT-TV channel I have here, which is according to a manual search on my DigiBit R1:
ntv, vox (TP:77, Freq: 12188) with Signalqualität: only 33 and Signalstärke: only 24. (for comparison the best values I could get here: 81, 27)
anyway, with the both worst values I do not have "Continuity Errors " or such (WEB-IF TvH: Status).

eldersixpack13, 2022-09-17:
ping @CvH 

you're aware of a newer bios (in case you haven't the last)
release note doesn't mention anythings what could fix your issue, but who knows ...

https://www.intel.de/content/www/de/de/download/18082/bios-update-rybdwi35.html

I usually do: bios install, **reset to defaults**, configure my special bios settings

P.S.
if unknown: I ~fixed~ my issue with complete fresh install in April

heitbaum, 2022-09-18:
I moved from the TGL to ADL yesterday. Just moved the nvme across - and all continued to work (I am running mesa 22.2rc and kernel 6.0rc) the only issue I have had with GBM is sometimes loosing Audio (pass through) and having to reboot to fix.

CvH, 2023-01-29:
AMD Athlon 5350 Kabini, so likewise not the intel driver
https://forum.libreelec.tv/thread/26560-libreelec-nexus-11-beta1/?postID=176262#post176262

eldersixpack13, 2023-01-29:
> AMD Athlon 5350 Kabini, so likewise not the intel driver https://forum.libreelec.tv/thread/26560-libreelec-nexus-11-beta1/?postID=176262#post176262

as I was CC'ed (though I was reading this thread and was waiting for the  fix to land in nightly)

my box is an intel NUC8i3.
with today's nightly (20230129-c2ff0c7) the flickering bug is still there, though I couldn't figure if it's already in nightly...

somewhere I read something like "couldn't reproduce":

with the keyboard/remote from comment "Feb 2, 2022" I could switch channels in 3 diff. ways
- up/down button where the "OK" is (frontside)
- with cursor keys (backside,  buttom right)
-  with directly keying in the number (top row, backside)
and with the last I (mostly) see flickering for 2-3 seconds until the new channel has ~settled~ : 
a) the displayed number I keyed in jumps out of the middle and
b) the top and bottom bar 

one could also easily reproduce it by doing the following:
- run a local movie
- pause (not stop) it
- go to EPG
- start a TV channe
l see the display is flickering for ~3 seconds 

maybe i need to mention:
I have some options ON in config menue "Player":
- adjust display refresh rate: on start/stop
- sync playback to display

and in config menue "System" :
- a whitelist what my monitor can do
- allow 3:2 pulldown refresh rates
- allow double refrsh rate

eldersixpack13, 2023-03-12:
thanks @smp79 !

CvH, 2023-03-12:
https://github.com/xbmc/xbmc/issues/22921 is not backportable to Kodi 20 because its basically a new feature

@sarbes did a fix that is worth a try if it works and should be easily backportable
[kodi-200.01-video_clear.patch](https://github.com/LibreELEC/LibreELEC.tv/files/10951684/kodi-200.01-video_clear.patch)

smp79, 2023-03-12:
> @sarbes did a fix that is worth a try if it works and should be easily backportable

It works

eldersixpack13, 2023-03-12:
"it works" and I would image it could fix another observation:
when I start the box (default starts to EPG) and directly view a TV channel the temperature of the box (sometimes) stays at ~33°.

when I temporary leave the TV channel view and play with LE config menu (e.g. to check the CPU/GPU temperature) the temperature goes up to ~40° and usually stays there until I stop the TV channel.
- don't know... -

P.S.
I'm one LE 12 (someone need to be the alpha tester ...)

P.P.S
to me it's sufficing that the bugs/shortcomings of LE are getting ironed out (sometimes); *exactly when* isn't my minor interest (as long as the bugs aren't hardcore regressions or such)

Thanks everyone !

closed 2023-09-04 by heitbaum