LibreELEC/LibreELEC.tv #6187: [BUG] ActiveAE - large audio sync error

eldersixpack13, 2022-02-03:
### Describe the bug

was watching SAT-TV for some minutes (< 5) after a reboot.
suddenly the audio got out of sync: mimik and speach doesn't match, stuttering audio
stopping the tv channel and starting it: all was fine again

this bug is *not* reliable reproduceable for me, it comes from time to time. 
noticed it first time: ~15 nightly ago (daily update here) 

### To Reproduce
Steps to reproduce the behavior:
1. watch TV
2. wait
3. 

### Informations
 - LE Version:  nightly-20220202-5347c5e
 - Hardware Platform:  Generic.x86_64

### Log file

I have a kodi.log, but I can't upload it cause of the size (1.1MB)

_**need instruction how do upload it somewhere !!!**_

any_way will try to paste it in the following comment_

eldersixpack13, 2022-02-04:
@heitbaum 
is it fine by you that I open another bug report ? 
(I'm no native english speaker, so I sometimes get think wrong...)

heitbaum, 2022-02-04:
> @heitbaum
> is it fine by you that I open another bug report ?
> (I'm no native english speaker, so I sometimes get think wrong...)

No problems.

eldersixpack13, 2022-02-04:
- no code reader here -
that said, but could it be that a fix for this bug is in the pipe:
https://github.com/xbmc/xbmc/pull/20938

a1rwulf, 2022-02-04:
@eldersixpack13 do you have refreshrate switching enabled?

eldersixpack13, 2022-02-04:
huhhh, don't know (exactly)
settings regarding refresh rate are:

**Player**
- Video => "Adjust display refresh rates" is OFF

**System**
- Display
   => "Allow 3:2 pulldown refresh rates" is ON
   => "Allow double refresh rates" is ON 

was my answer a (at least partly) useful answer ?

eldersixpack13, 2022-02-04:
today I had the audio sync error while watching a movie from german mediathek.
alas debug was off.
http://ix.io/3OGB

=> time: 2022-02-04 23:52:05.322 T:626

a1rwulf, 2022-02-05:
As you don't have refreshrate switch enabled, I don't think https://github.com/xbmc/xbmc/pull/20938 will fix your issue.
Your log looks a bit like too less data is incoming to keep everything going, but I'm not sure.

eldersixpack13, 2022-02-06:
well, all is via LAN
and more important:
- mostly audio sync error occur when I first time start a movie out of the mediathek or from Sat-TV (within the first ~10 minutes)
- when this error occured and I stop and restart the movie again all is fine (for a long or all the time)

weird !

heitbaum, 2022-04-15:
@eldersixpack13  - the upstream patch in in the nightly kernels now. Are you still experiencing these issues?

eldersixpack13, 2022-04-15:
@heitbaum, 
no since a  long time already (one/two nightlies following 5347c5e).

btw:
~2 days ago I thought to close this, too
telepathy ?
;-)

eldersixpack13, 2022-04-15:
this bug seems to be fixed.

closed 2022-04-15 by eldersixpack13