LibreELEC/LibreELEC.tv #6213: [BUG] build is failing because repo mirror is outdated

nokludge, 2022-02-09:
### Describe the bug
These 2 are not existing on the sources.libreelec.tv/mirror, at least in the current master (72b6d0d):
- rtmpdump
- edid-decode

because of that the build process wont work.
I saw the changes you guys made there are not a mistake and i dont want to open a PR with reverting these changes.
Maybe the sources.libreelec.tv/mirror is outdated?

Idk. how to update the mirror.

### To Reproduce
Steps to reproduce the behavior:
`
docker run -v $(pwd):/build -w /build -it -e PROJECT=RPi -e DEVICE=RPi4 -e ARCH=arm libreelec make image
`

### Informations
Linux Mint 20.3 "Una" + Docker version 20.10.7, build 20.10.7-0ubuntu5~20.04.2

### Additional context
Its possible to Workoaround these bugs:

1) use the old Version (Year 2016) of this file: **packages/multimedia/rtmpdump/package.mk**
Means use these settings:
`
PKG_VERSION="fa8646d"
PKG_SHA256="dba4d4d2e1c7de6884b01d98194b83cab6784669089fa3c919152087a3a38fd2"
PKG_URL="https://sources.libreelec.tv/mirror/rtmpdump/${PKG_VERSION}.tar.gz"
`
2) use the old version of this file: **packages/debug/edid-decode/package.mk**
Means use these settings:
`
PKG_VERSION="15df4aebf06da579241c58949493b866139d0e2b"
PKG_SHA256="58743c8ba768134ef1421e1ce9f4edf0eafdae29377fe5b8f4cb285f16dc142e"
PKG_URL="https://git.linuxtv.org/edid-decode.git/snapshot/${PKG_VERSION}.tar.xz"
`
and
`
make_target() {
  echo "${CC} ${CFLAGS} -Wall ${LDFLAGS} -o edid-decode edid-decode.c -lm"
  ${CC} ${CFLAGS} -Wall ${LDFLAGS} -o edid-decode edid-decode.c -lm
}
`

heitbaum, 2022-02-11:
Hi @nokludge - I cant reproduce your error (from the primary sources):
```
docker@2ab3a54165bd:~/LibreELEC.master$ s/get rtmpdump
GET      rtmpdump (archive)
--2022-02-11 08:23:59--  http://repo.or.cz/rtmpdump.git/snapshot/c5f04a58fc2aeea6296ca7c44ee4734c18401aa3.tar.gz
Resolving repo.or.cz (repo.or.cz)... 195.113.20.142, 2001:718:1e03:801::8e
Connecting to repo.or.cz (repo.or.cz)|195.113.20.142|:80... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://repo.or.cz/rtmpdump.git/snapshot/c5f04a58fc2aeea6296ca7c44ee4734c18401aa3.tar.gz [following]
--2022-02-11 08:24:02--  https://repo.or.cz/rtmpdump.git/snapshot/c5f04a58fc2aeea6296ca7c44ee4734c18401aa3.tar.gz
Connecting to repo.or.cz (repo.or.cz)|195.113.20.142|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/x-gzip]
Saving to: '/home/docker/LibreELEC.master/sources/rtmpdump/rtmpdump-c5f04a58fc2aeea6296ca7c44ee4734c18401aa3.tar.gz'

/home/docker/LibreELEC.master     [    <=>                                           ] 139.68K   140KB/s    in 1.0s

2022-02-11 08:24:13 (140 KB/s) - '/home/docker/LibreELEC.master/sources/rtmpdump/rtmpdump-c5f04a58fc2aeea6296ca7c44ee4734c18401aa3.tar.gz' saved [143037]

    INFO      Calculated checksum: fd8c21263d93fbde8bee8aa6c5f6a657789674bb0f9e74f050651504d5f43b46
```
and
```
docker@2ab3a54165bd:~/LibreELEC.master$ s/get edid-decode
GET      edid-decode (archive)
--2022-02-11 08:25:20--  https://repo.or.cz/edid-decode.git/snapshot/770cfb947ea9d9eb5cda57a87dc66d13c60cfefc.tar.gz
Resolving repo.or.cz (repo.or.cz)... 195.113.20.142, 2001:718:1e03:801::8e
Connecting to repo.or.cz (repo.or.cz)|195.113.20.142|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/x-gzip]
Saving to: '/home/docker/LibreELEC.master/sources/edid-decode/edid-decode-770cfb947ea9d9eb5cda57a87dc66d13c60cfefc.tar.gz'

/home/docker/LibreELEC.master     [    <=>                                           ] 109.47K   111KB/s    in 1.0s

2022-02-11 08:25:41 (111 KB/s) - '/home/docker/LibreELEC.master/sources/edid-decode/edid-decode-770cfb947ea9d9eb5cda57a87dc66d13c60cfefc.tar.gz' saved [112095]

    INFO      Calculated checksum: eb6a766b89ef5f7d06d6649fc442ee57377ee7add801cde0a5ba6b636da78075
```

heitbaum, 2022-02-14:
Closing - Pelase reopen if still an issue.

closed 2022-02-14 by heitbaum