LibreELEC/LibreELEC.tv #6232: [BUG] gtk3 build failure in jenkins

heitbaum, 2022-02-14:
To check issue:

```
<<< gtk3:target seq 485 <<<
[1;36mUNPACK[0m      gtk3
    [1;32mAPPLY PATCH [1;37m(common)[0m[0m      packages/addons/addon-depends/chrome-depends/gtk3/patches/disable-to-pixdata-build.patch
patching file gtk/gen-gtk-gresources-xml.py
    [1;33mFIXCONFIG[0m      /var/lib/jenkins/LE/build4/workspace/Addons/All_Addons-Generic/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/gtk3-3.24.31/build-aux/
[1;33mBUILD[0m      gtk3 [1;37m(target)[0m
    [1;35mTOOLCHAIN[0m      meson
    [1;35mPKG_CONFIG_PATH[0m      /var/lib/jenkins/LE/build4/workspace/Addons/All_Addons-Generic/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/install_pkg/libXft-2.3.4/usr/lib/pkgconfig
    [1;35mPKG_CONFIG_PATH[0m      /var/lib/jenkins/LE/build4/workspace/Addons/All_Addons-Generic/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/install_pkg/pango-1.50.3/usr/lib/pkgconfig
    [1;35mPKG_CONFIG_PATH[0m      /var/lib/jenkins/LE/build4/workspace/Addons/All_Addons-Generic/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/install_pkg/gdk-pixbuf-2.42.6/usr/lib/pkgconfig
    [1;35mPKG_CONFIG_PATH[0m      /var/lib/jenkins/LE/build4/workspace/Addons/All_Addons-Generic/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/install_pkg/shared-mime-info-2.1/usr/share/pkgconfig
Executing (target): meson --prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --sysconfdir=/etc --libdir=/usr/lib --libexecdir=/usr/lib --localstatedir=/var --buildtype=plain -Dstrip=true --cross-file=/var/lib/jenkins/LE/build4/workspace/Addons/All_Addons-Generic/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/gtk3-3.24.31/.x86_64-libreelec-linux-gnu/meson.conf -Dbroadway_backend=false -Dcloudproviders=false -Dcolord=no -Ddemos=false -Dexamples=false -Dgtk_doc=false -Dintrospection=false -Dman=false -Dprint_backends=file,lpr -Dquartz_backend=false -Dtests=false -Dwayland_backend=false -Dwin32_backend=false -Dxinerama=no -Dbuiltin_immodules=yes -Dx11_backend=true /var/lib/jenkins/LE/build4/workspace/Addons/All_Addons-Generic/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/gtk3-3.24.31
The Meson build system
Version: 0.61.1
Source dir: /var/lib/jenkins/LE/build4/workspace/Addons/All_Addons-Generic/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/gtk3-3.24.31
Build dir: /var/lib/jenkins/LE/build4/workspace/Addons/All_Addons-Generic/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/gtk3-3.24.31/.x86_64-libreelec-linux-gnu
Build type: cross build
Project name: gtk+-3.0
Project version: 3.24.31
C compiler for the host machine: /var/lib/jenkins/LE/build4/workspace/Addons/All_Addons-Generic/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/bin/x86_64-libreelec-linux-gnu-gcc (gcc 10.3.0 "x86_64-libreelec-linux-gnu-gcc-10.3.0 (GCC) 10.3.0")
C linker for the host machine: /var/lib/jenkins/LE/build4/workspace/Addons/All_Addons-Generic/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/bin/x86_64-libreelec-linux-gnu-gcc ld.bfd 2.35.1
C compiler for the build machine: /var/lib/jenkins/LE/build4/workspace/Addons/All_Addons-Generic/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/bin/host-gcc (gcc 8.3.0 "gcc (Debian 8.3.0-6) 8.3.0")
C linker for the build machine: /var/lib/jenkins/LE/build4/workspace/Addons/All_Addons-Generic/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/toolchain/bin/host-gcc ld.bfd 2.31.1
Build machine cpu family: x86_64
Build machine cpu: x86_64
Host machine cpu family: x86_64
Host machine cpu: x86_64
Target machine cpu family: x86_64
Target machine cpu: x86_64
Has header "crt/externs.h" : NO 
Has header "dlfcn.h" : YES 
Has header "ftw.h" : YES 
Has header "inttypes.h" : YES 
Has header "linux/memfd.h" : YES 
Has header "linux/input.h" : YES 
Has header "dev/evdev/input.h" : NO 
Has header "locale.h" : YES 
Has header "memory.h" : YES 
Has header "stdint.h" : YES 
Has header "stdlib.h" : YES 
Has header "string.h" : YES 
Has header "strings.h" : YES 
Has header "sys/mman.h" : YES 
Has header "sys/param.h" : YES 
Has header "sys/stat.h" : YES 
Has header "sys/sysinfo.h" : YES 
Has header "sys/systeminfo.h" : NO 
Has header "sys/time.h" : YES 
Has header "sys/types.h" : YES 
Has header "unistd.h" : YES 
Library m found: YES
Checking for function "dcgettext" with dependency -lm: YES 
Checking for function "getpagesize" with dependency -lm: YES 
Checking for function "getresuid" with dependency -lm: YES 
Checking for function "lstat" with dependency -lm: YES 
Checking for function "mmap" with dependency -lm: YES 
Checking for function "nearbyint" with dependency -lm: YES 
Checking for function "posix_fallocate" with dependency -lm: YES 
Checking for function "_lock_file" with dependency -lm: NO 
Checking for function "flockfile" with dependency -lm: YES 
Checking for function "mkstemp" with dependency -lm: YES 
Checking for function "mallinfo" with dependency -lm: YES 
Checking for function "round" with dependency -lm: YES 
Checking for function "rint" with dependency -lm: YES 
Checking for function "log2" with dependency -lm: YES 
Checking for function "exp2" with dependency -lm: YES 
Checking for function "sincos" with dependency -lm: YES 
[... 44 lines cut ...]
Run-time dependency cairo found: YES 1.17.4
Run-time dependency cairo-gobject found: YES 1.17.4
Run-time dependency pango found: YES 1.50.3
Run-time dependency fribidi found: YES 1.0.11
Run-time dependency pangoft2 found: YES 1.50.3
Run-time dependency freetype2 found: YES 24.1.18
Run-time dependency pangocairo found: YES 1.50.3
Run-time dependency gdk-pixbuf-2.0 found: YES 2.42.6
Run-time dependency epoxy found: YES 1.5.5
Run-time dependency atk found: YES 2.36.0
Run-time dependency harfbuzz found: YES 3.2.0
Run-time dependency xkbcommon found: YES 1.3.1
Found CMake: -DCMAKE_INSTALL_PREFIX=/usr (3.22.2)
Run-time dependency iso-codes found: NO (tried pkgconfig and cmake)
Run-time dependency cairo-xlib found: NO (tried pkgconfig and cmake)

../meson.build:521:97: ERROR: Dependency "cairo-xlib" not found, tried pkgconfig and cmake

A full log can be found at /var/lib/jenkins/LE/build4/workspace/Addons/All_Addons-Generic/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/gtk3-3.24.31/.x86_64-libreelec-linux-gnu/meson-logs/meson-log.txt
[1;31mFAILURE: scripts/build gtk3:target during configure_target (default)[0m
*********** FAILED COMMAND ***********
CC="${HOST_CC}" CXX="${HOST_CXX}" meson ${TARGET_MESON_OPTS} --cross-file=${MESON_CONF} ${PKG_MESON_OPTS_TARGET} ${PKG_MESON_SCRIPT%/*}
**************************************
[1;31mFAILURE: scripts/build gtk3:target has failed![0m
```

SupervisedThinking, 2022-02-14:
I guess `cairo-xlib` is missing because you build `Generic` which means a GBM build. If you build with a displayserver `x11` `cairo-xlib` should be present: https://github.com/LibreELEC/LibreELEC.tv/blob/master/packages/graphics/cairo/package.mk#L76 and for Wayland GTK3 needs adjustment like this: https://github.com/SupervisedThinking/LibreELEC-RR/commit/e511281fe5e6e51698b66422cf968ae78d82249b

No clue if one could build GTK without any displayserver https://wiki.archlinux.org/title/GTK#GDK_backends

So all in all it's IMHO not really a bug, the underlaying problem was the switch from X11 to GBM which can't build all packages anymore.

heitbaum, 2022-02-17:
@CvH should we remove chrome from the LE11 addons build - because it can’t work?

SupervisedThinking, 2022-02-17:
Well it works pretty much perfectly fine with X11 builds. I would rather offer multiple builds like an "GBM" & X11 build and let the user choose what suits his needs.

The advantages of x11 are IMHO rather few unless you've no HDR setup there is basically no missing feature at all if you compare GBM builds to X11.

heitbaum, 2022-07-19:
Closing as we are building it separately on the new build-host. (it is broken due to dependencies - but that is a separate issue.)

closed 2022-07-19 by heitbaum