LibreELEC/LibreELEC.tv #6233: [BUG] connman: DNS broken

3nprob, 2022-02-14:
### Describe the bug
It seems to me that DNS support for Wireguard VPN is not working.

### To Reproduce

Connect to Wireguard VPN when already connected to DHCP-enabled wifi network:

```
$ cat /storage/.config/wireguard/myvpn.config
[provider_wireguard]
Type = WireGuard
Name = WireGuard VPN Tunnel
Host = 1.2.3.4
Wireguard.Address = 10.1.100.121/24
Wireguard.ListenPort = 1234
Wireguard.PrivateKey = foo
Wireguard.PublicKey = bar
Wireguard.PresharedKey = baz
Wireguard.DNS = 10.1.100.2
Wireguard.AllowedIPs = 10.1.100.0/24
Wireguard.EndpointPort = 2345
Wireguard.PersistentKeepalive = 30

$ connmanctl  connect vpn_1_2_3_4
Connected vpn_1_2_3_4

$ wg
## Connected, data transfering in both directions.

$ nslookup foo.mydomain.com 
(does not work, looks up using ISP DNS acquired via DHCP)

$ nslookup foo.mydomain.com 10.1.100.2
(resolves correctly)

$ cat /etc/resolv.conf
# Generated by Connection Manager
nameserver $ISP_DHCP_DNS
```
So it seems connman ignores the desired DNS from wireguard.

I suspect the `-r` here is the reason:

https://github.com/LibreELEC/LibreELEC.tv/blob/acc9f0567a6af0b9313af13ac39d227e0b35a1b0/packages/network/connman/system.d/connman.service#L15

If the `-r` is removed, connman  would instead run a local DNS proxy and keep track of DNS config properly.

It would be great if at the very least this could be made overridable by env vars (or just re-enable connman's built-in DNS functionality?)

### Informations
 - LE Version: 10.0.1
 - Hardware Platform: Rpi400

3nprob, 2022-02-14:
Overriding `connman.service` with a version sans `-r` and DNS over VPN works as expected

heitbaum, 2026-02-13:
closing as self-fixed

closed 2026-02-13 by heitbaum