LibreELEC/LibreELEC.tv #6237: [BUG]LE10/LE11 HBR Audio PT Audio issue with 4k@23.976 with J4125

the-dreamer, 2022-02-15:
### Describe the bug
As a starting point i need more guidance and knowledge of some experts. 

After serveral bug fixes of truehd pt in kodi last time my hope was to have finally no audio drop with hbr pt audio. But not for me. After many tests and tries i could figured out that i have audio drops and sometimes cracks on every HBR Audio PT like Truehd/Atmos/DTS HD Master Audio and only with 4K @ 23.976. When i disable PT and use MultiCH PCM over HDMI the same situation. 

When i switch to 4K@24p the audio drops are gone. When i switch to 2k@23.976 also drops are gone. 

I have tried already to get some help on LE Forum and on Kodi Forum without any success:  

https://forum.kodi.tv/showthread.php?tid=347674&pid=3086730#pid3086730
https://forum.libreelec.tv/thread/25245-le10-11-truehd-atmos-audio-drops/

Debug Logs are showing for me completly no hint for that specific issue. 

I have the htpc connected to an Onkyo RZ830 and then to LG C1

My current assumption is that the 4k@23.976 timing for HDMI, ALSA or KODI is not okay and the AVR is loosing sync with audio. 
With MultiCh PCM and DTS-HD Master Audio situation is much worse. 

With DTS-HD Master Audio it could happend that audio is lost completly. Changing Audio Mode on AVR restores Audio. 
The AVR is showing no indication that is something wrong. 

I could not found any indication on internet that the AVR has an issue. 

Picture is always stable and during audio drops there are no frame drops. 

When system is switched to 4k@24p a full movie has no audio issue. 

As i can build images by my self, i can do some tests when i am guided. Any help is welcome. If necessary i can provide more information

### To Reproduce
Steps to reproduce the behavior:
1. Go to '...'
2. Play '....'
3. See error

Playing the Zootopia movie sample which was shared in kodi forum and listen to the audio drop. 

### Informations
 - LE Version: LE master from early Feb. 
 - Hardware Platform: Intel J4125, 8GB, SSD
 - AVR Onkyo RZ830
 - LG C1
 - 
 Happens on LE10 and LE11 but i have figured out the real issue with build from early feb 

### Log file
http://ix.io/3PI3
### Additional context

the-dreamer, 2022-02-21:
I have tested that issue with the my RPI4 also. 

Beside of the choopy Video playback (seems that drm prime hw accel not working at all with 4k), 4k@23,976 has no audio drop with truehd and my AVR. So AVR seems okay for me. 

Additionally to the BUG on the Intel System 4K@29.97 has also Audio drops. Other Refreshrate seems okay as described.

isamudaison, 2023-05-06:
I am experiencing this exact behavior even on Intel's new AL 'n-series' platform (N100 specifically) with LE 11.0.1. I suspect there may be a poorly-implemented 23.976 Hz mode on Intel's IGD (still; they have quite a history of munging that refresh rate up). I have a Vero4k+ that is able to pass through all audio formats without issue (thus verifying the rest of the a/v chain).
Some references : 
https://www.intel.com/content/www/us/en/support/articles/000060142/intel-nuc.html
https://www.intel.com/content/www/us/en/support/articles/000087953/intel-nuc.html
Another possibility would be sub-par BIOSes for these more inexpensive platforms.

legoman666, 2023-05-10:
Also experiencing on an n100 system. Only seems to happen with HDR 4K 23.976hz files. 4k YouTube videos work fine. 1080p videos work fine although I dont seem have any HDR 1080p videos to test with. Happens on both DTS & ATMOS. Very intermittent.

heitbaum, 2023-09-06:
I’m interested to know if this issue occurs with LE12 nightlies on Intel. There have been changes and drivers to support the newer N series devices in kernels 6.4/6.5.

legoman666, 2023-09-06:
HDR was not working on LE12 previously. Is it working now?

the-dreamer, 2023-09-21:
> I’m interested to know if this issue occurs with LE12 nightlies on Intel. There have been changes and drivers to support the newer N series devices in kernels 6.4/6.5.

I will try

the-dreamer, 2023-09-29:
Some updates on this, i build an image from master from 27th Sept. 

I have no clear picture right now. Some situations have been improved i little bit. Zoomania clip has no audo drop so far. 

MKV with DTS-MA and True-HD has still drops. DTS-MA got no change at all. I have big drop outs like a link lost to AVR 1 or 2 drop outs every 1 to 3 minutes. It's varying a lot. 
 
All of my mkv tests are HDR10 movies. I am not sure if that triggers the problem. 

Overall, the problem still exists.

heitbaum, 2024-10-29:
Is this still an issue with LE13 nightlies?

legoman666, 2024-10-29:
> Is this still an issue with LE13 nightlies?

Yes