LibreELEC/LibreELEC.tv #6239: [BUG]build failed：cannot find /usr/lib/libc.so.6 inside /**/sysroot

hptsf, 2022-02-16:
### Describe the bug
build failed when 
[082/276] [FAIL] build   gcc:host
and the log as below

switch to sysroot dir, and ls -l; there many broken links file
lrwxrwxrwx 1 ubuntu ubuntu      159 Feb 16 11:23 libutil.so -> '/te_data/01.project/32.libreelec/LibreELEC.tv/build.LibreELEC-RK3399.arm-11.0-devel/toolchain/armv8a-libreelec-linux-gnueabihf/sysroot\/usr/lib/libutil-2.32.so'
lrwxrwxrwx 1 ubuntu ubuntu      159 Feb 16 11:23 libutil.so.1 -> '/te_data/01.project/32.libreelec/LibreELEC.tv/build.LibreELEC-RK3399.arm-11.0-devel/toolchain/armv8a-libreelec-linux-gnueabihf/sysroot\/usr/lib/libutil-2.32.so'

so the required library cannot be found.

### To Reproduce
Steps to reproduce the behavior:
git clone
git check master
cd LibreELEC.tv
PROJECT=Rockchip DEVICE=RK3399 ARCH=arm UBOOT_SYSTEM=nanopc-t4 make image -j${JOBS}

### Informations
 - LE Version: [master]
 - Hardware Platform: [RK3399]

### Log file
/te_data/01.project/32.libreelec/LibreELEC.tv/build.LibreELEC-RK3399.arm-11.0-devel/toolchain/armv8a-libreelec-linux-gnueabihf/bin/ld: cannot find /usr/lib/libc.so.6 inside /te_data/01.project/32.libreelec/LibreELEC.tv/build.LibreELEC-RK3399.arm-11.0-devel/toolchain/armv8a-libreelec-linux-gnueabihf/sysroot
/te_data/01.project/32.libreelec/LibreELEC.tv/build.LibreELEC-RK3399.arm-11.0-devel/toolchain/armv8a-libreelec-linux-gnueabihf/bin/ld: cannot find /usr/lib/ld-linux-armhf.so.3 inside /te_data/01.project/32.libreelec/LibreELEC.tv/build.LibreELEC-RK3399.arm-11.0-devel/toolchain/armv8a-libreelec-linux-gnueabihf/sysroot

heitbaum, 2022-02-16:
was this a clean build? I just double checked `PROJECT=Rockchip DEVICE=RK3399 ARCH=arm UBOOT_SYSTEM=nanopc-t4 make image` is building. What is your host build environment?

hptsf, 2022-02-16:
Yes , it's clean build
My host is ubuntu1804， and change shell from dash to bash.

heitbaum, 2022-02-16:
Given you are using ubuntu. If you follow this: https://github.com/LibreELEC/LibreELEC.tv/blob/master/tools/docker/README.md pretty sure it will sort you out. It is how I do my builds.

heitbaum, 2022-02-16:
Just install docker

heitbaum, 2022-02-16:
Another suggestion is not to use -j

hptsf, 2022-02-16:
ubuntu@VM-71-95-ubuntu LibreELEC.tv_docker:$ sudo docker run --log-driver none -v `pwd`:/build -w /build -it -e PROJECT=Rockchip -e DEVICE=RK3399 -e ARCH=arm libreelec make image
./scripts/image mkimage
mkdir: cannot create directory '/build/build.LibreELEC-RK3399.arm-11.0-devel': Permission denied
*********** FAILED COMMAND ***********
. config/options ""
**************************************
Makefile:12: recipe for target 'image' failed
make: *** [image] Error 1
ubuntu@VM-71-95-ubuntu

hptsf, 2022-02-16:
ubuntu@VM-71-95-ubuntu LibreELEC.tv_docker:$ sudo docker run --log-driver none -v /te_data/01.project/32.libreelec/LibreELEC.tv_docker:/build -w /build -it -e PROJECT=Rockchip -e DEVICE=RK3399 -e ARCH=arm libreelec make image
./scripts/image mkimage
mkdir: cannot create directory '/build/build.LibreELEC-RK3399.arm-11.0-devel': Permission denied
*********** FAILED COMMAND ***********
. config/options ""
**************************************
Makefile:12: recipe for target 'image' failed
make: *** [image] Error 1

heitbaum, 2022-02-16:
what are the permissions of your files (in your directory) 
ls -la .

heitbaum, 2022-02-18:
Hope your build is all working now. Please let us know / reopen if you are still having issues.

closed 2022-02-18 by heitbaum