LibreELEC/LibreELEC.tv #6240: [GENERAL] Building LibreELEC for Raspberry Pi4 Failed

jitendrazala, 2022-02-16:
### Building LibreELEC for RPi4
I have been trying to build LibreELEC for Raspberry Pi4 in using ubuntu 20.04 [WSL]. I got errors while building and I rectified those errors. But somehow I have been stuck in end.

### To Reproduce
Steps to reproduce the behavior:
1. Setting for WSL as per [](https://wiki.libreelec.tv/development-1/building-windows-wsl)
2. Install dependancies
**_sudo apt update
sudo apt upgrade
sudo apt install gcc make git unzip wget xz-utils bc gperf zip unzip makeinfo g++ mkfontscale mkfontdir bdftopcf xsltproc java_**

3. Clone LibreELEC 
**_cd ~
[](git clone https://github.com/LibreELEC/LibreELEC.tv.git)
cd LibreELEC.tv_**

4. Download tools
**_PROJECT=Generic ARCH=x86_64 tools/download-tool_**

5. Build
 PROJECT=RPi DEVICE=RPi4 ARCH=arm make image

### Informations
 - LE Version: 10.0 [master]
 - Hardware Platform: RPi4

### Log file
[](https://pastebin.com/rtWktPye)

Can anyone help me for the same ?

Thanks.
Jitendra

heitbaum, 2022-02-16:
Hi @jitendrazala 
- what version of WSL are you using?
  - WSL2? 
- Are you building master of libreelec-10.0?

What OS? e.g. `Welcome to Ubuntu 20.04.3 LTS (GNU/Linux 5.10.93.2-microsoft-standard-WSL2 x86_64)`

```
$ cat /etc/os-release
NAME="Ubuntu"
VERSION="20.04.3 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.3 LTS"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal
```

jitendrazala, 2022-02-16:
Thanks for your prompt reply.

1. I am using WSL2.
2. Yes. I am using master of libreelec-10.0
3. OS :
`NAME="Ubuntu"
VERSION="20.04.3 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04.3 LTS"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal`

CvH, 2022-02-16:
can you post the output of `uname -a` ?

btw I give it a go when at home, I build RPi before so it worked at some point

jitendrazala, 2022-02-16:
```
$ uname -a
Linux Jitendra 5.10.60.1-microsoft-standard-WSL2 #1 SMP Wed Aug 25 23:20:18 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux
```
It would be really helpful if you can give it a shot.

kszaq, 2022-02-16:
@jitendrazala Maybe unrelated but removing Windows path on WSL solved a similar build issue for me. I used Option 1 from https://stackoverflow.com/a/63195953 - remember to restart WSL.

jitendrazala, 2022-02-17:
@kszaq , I already did that. It was mentioned in LibreELEC wiki page for WSL.
Bdw thanks.

CvH, 2022-02-17:
@jitendrazala I build Rpi4 (LE11 Matser) successfully yesterday with everything from https://wiki.libreelec.tv/development-1/building-windows-wsl applied 

I run a LE10 today and have a look if something won't work anymore.

SupervisedThinking, 2022-02-18:
Well maybe I've not seen the specs of your build machine but you should consider an out of resources problem. I'm usually hacking stuff together on an old ThinkPad and a custom package in my build failed with pretty much exact the same error.

I resolved it by freeing up RAM 🤔

heitbaum, 2022-02-18:
> i have repaire the broken link manual

thanks for the update. Closing the build issue. please reopen / let us know if you are still facing issues.

closed 2022-02-18 by heitbaum