LibreELEC/LibreELEC.tv #6244: [BUG]

hptsf, 2022-02-18:
### Describe the bug
I am using libreelec img on rockpi4 Platform.
I using hw_decode sample program of ffmpeg to test h264 decoder,
which command is: ./doc/examples/hw_decode drm ./Jasmine_1080p.h264 out.yuv
and using ffplay to play both:
>ffplay -video_size 1920x1080 -i out.yuv
>ffplay -video_size 1920x1080 -i out.yuv
I found that the decoded picture lost its true color， as below log file(pic1: h264 and pic2: yuv decoderd)
May I ask what is the reason?

### To Reproduce
Steps to reproduce the behavior:
1. build kernel
2. build ffmpeg
3. flash sd and start rockpi4

### Informations
 - LE Version: [master 11.0]
 - Hardware Platform: [Rockhip RK3399]

### Log file

the org pic：
[image]

hw_decoder‘s pic：
[image]

ffmpeg‘s config options：

PKG_FFMPEG_DEBUG="--enable-debug --disable-stripping"
PKG_FFMPEG_V4L2="--enable-v4l2_m2m --enable-libdrm --enable-libudev --enable-v4l2-request"
PKG_FFMPEG_VAAPI="--disable-vaapi"
PKG_FFMPEG_VDPAU="--disable-vdpau"
PKG_FFMPEG_HWACCEL="--enable-hwaccels --enable-hwaccel=h264_v4l2request"
PKG_FFMPEG_AV1="--enable-libdav1d"
PKG_FFMPEG_FPU="--enable-neon"

./configure --prefix=`pwd`/${INSTALL_DIR} --enable-shared               \
        --enable-gpl --disable-version3 --enable-logging --disable-doc ${PKG_FFMPEG_DEBUG} --enable-pic \
        --enable-optimizations --disable-extra-warnings --enable-avdevice --enable-avcodec              \
        --enable-avformat --enable-swscale --enable-postproc --enable-avfilter --disable-devices        \
        --enable-pthreads --enable-network --enable-gnutls --disable-openssl --disable-gray             \
        --enable-swscale-alpha --disable-small --enable-dct --enable-fft --enable-mdct --enable-rdft    \
        --disable-crystalhd ${PKG_FFMPEG_V4L2} ${PKG_FFMPEG_VAAPI} ${PKG_FFMPEG_VDPAU}                  \
        --enable-runtime-cpudetect --disable-hardcoded-tables ${PKG_FFMPEG_HWACCEL} --enable-parsers    \
        --enable-bsfs --enable-protocol=http --enable-filters                                           \
        --disable-avisynth --enable-bzlib --disable-lzma --disable-alsa --disable-frei0r                \
        --disable-libopencore-amrnb --disable-libopencore-amrwb --disable-libopencv --disable-libdc1394 \
        --disable-libfreetype --disable-libgsm --disable-libmp3lame --disable-libopenjpeg               \
        --disable-librtmp ${PKG_FFMPEG_AV1} --enable-libspeex --disable-libtheora                       \
        --disable-libvo-amrwbenc --disable-libvorbis --disable-libvpx --disable-libx264                 \
        --disable-libxavs --disable-libxvid --enable-zlib --enable-asm --disable-altivec                \
        ${PKG_FFMPEG_FPU} --disable-symver --enable-demuxers

knaerzche, 2022-02-18:
I'm not sure I fully understand what you trying to achieve: Neither direct usage of ffmpeg nor ffplay are supported by LE. 
Just a hint: Output of RK video decoder(s) is NV12 not YU12 which is the default in case of ffplay is used with a yuv file - so you might want to add it to it's options:

`ffplay -f rawvideo -pixel_format nv12 -video_size 1920x1080 out.yuv`

should do it.

closed 2022-02-18 by knaerzche

hptsf, 2022-02-20:
Yes， Your sample command is corrent.

thx very much.