LibreELEC/LibreELEC.tv #6250: [RFE] Improve Odroid XU4

heitbaum, 2022-02-18:
See commentary in these PRs
- #5932

MastaG, 2022-06-23:
@heitbaum @chewitt 
If any of you guys are willing to take a look at the MFC decoding issue on 5.10 and beyond, I can send over a spare XU3 (basically the same thing as XU4, but with added headphone jack and 100mbit Ethernet instead of 1gbit).
It's the only issue that holds back v4l2-m2m video decoding.
For the GBM buffer issue (causing Kodi to freeze up after a few seconds) a workaround has already been posted a while ago by AreaScout.
Just send me an email I can dispatch it this week.

chewitt, 2022-06-24:
@MastaG Neither of us write driver code so detailed fixing of MFC problems isn't likely to happen, but if there are patches that resolve or improve things (that apply-to and work on modern kernels) we can pick them. If you can point me to source patches in a repo that would be helpful. I've looked at @AreaScout GH repos but armbian/linux/xbmc ones do not show any updated branches for several years and I don't have time to read hundreds of disorganised posts on the HK forums.

AreaScout, 2022-06-24:
Have you guys missed this ? https://forum.odroid.com/viewtopic.php?p=350810#p350810 MFC patches, AND he has KODI running with Panfrost on XU4, patches to get the driver working are also within this post:

I will build a new NAS Image with KODI passed on that patch set + KODI will be installable from the HK minimal Ubuntu XU4 images, but it will take a week or two, still have two bugs to fix for KODI on ODROID-M1

RG,
Daniel

MastaG, 2022-06-24:
Hi AreaScout!
I totally missed it!
That's great news!

chewitt, 2022-06-24:
I'll spin an LE image based on 5.19-rc3 over the weekend and see what works.

chewitt, 2022-06-24:
@AreaScout @MastaG https://chewitt.libreelec.tv/testing/LibreELEC-Exynos.arm-10.85.0-odroid-xu4.img.gz <= no hardware decoding and prob. some more things to port over, but I am looking at the Kodi home screen via panfrost :)

AreaScout, 2022-06-24:
@chewitt That's so cool, didn't believe anymore in seeing Panfrost on XU4 and then bam, out of nowhere it was there :)
thank you for the link will give it a go tomorrow

chewitt, 2022-06-25:
I've updated the image I posted with "the usual patches" for kernel/kodi that have been doing the rounds for a while though those should be re-evaluated as some of them probably aren't needed. I also figured out kernel defconfig changes needed to include the s5p-* drivers; there's been a large rearrangement of drivers/media in Linux 5.19.y so those were missing a few CONFIG dependencies. The image is usable with software decoding and some 1080p test files. IIRC hardware decoding will need a rework of @mihailescu2m ffmpeg hack or the adaptation of https://patchwork.kernel.org/project/linux-samsung-soc/cover/20190301122055.7135-1-a.hajda@samsung.com/ ..which is technically better, although with the age of these boards now the ffmpeg hack probably has a greater chance of finding a willing volunteer to work on it.

NB: If anyone wants to look at ffmpeg, we should use the same V4L2 codebase that LE is currently using for Amlogic/RPi as this is theoretically the codebase that will eventually be upstreamed (by RPi Foundation devs) to ffmpeg.

AreaScout, 2022-06-28:
@chewitt didn't had time yet :( I still don't believe in drmPRIME on XU4, I saw it not working correctly with GSCALER inside FFmpeg but I love to get corrected here, this are the patches we used to run KODI on XU4 

https://forum.odroid.com/viewtopic.php?p=318877#p318877

mihailescu2m, 2022-06-28:
What @chewitt said is correct, I did get drmprime working as good as it was working on other boards back then.

AreaScout, 2022-06-28:
Hi memeka :) It's good to see you !

Well as far as I remember only 1080p files did work, no scaling to other resolutions or what so ever, only color conversation and the movie playback was flickering, I still have a video lying around showing it, if you did had that working then I don't get it why we didn't know about it

Regards,
Daniel

chewitt, 2022-06-28:
Here's a Kodi debug log (with ffmpeg debug) http://ix.io/412S - hardware decoding currently fails for me using the same ffmpeg sources LE uses with RPi/Amlogic. The RPi developer who's been working on their v4l2_m2m vdec commented: "Coded data goes in but the expected initial 'resolution changed' event never comes out of the decoder. Also, the decoder doesn't seem to support DECODER_START/STOP which is not technically fatal but doesn't inspire confidence" .. I need to experiment with another patch @MastaG pointed me to, but I think v4l2_m2m evolved a little and some rework is needed.

chewitt, 2022-06-30:
I've done some experimenting over the last few days. I'm working from an LE develpment branch which is currently using Kodi 20a1, ffmpeg 4.4, and Linux 5.19-rc4. Changes are here: https://github.com/chewitt/LibreELEC.tv/commits/samsung

Kernel: https://github.com/chewitt/linux/commits/samsung-5.19.y
Kodi: https://github.com/chewitt/LibreELEC.tv/commit/2bb2a65b523ddeed7f542191aef2f60e6be820f0

Using the same ffmpeg 4.4 with changes for improved v4l2_m2m on Amlogic/RPi, I have no decode and playback doesn't start (see comments from the RPi ffmpeg developer above). Using default ffmpeg 4.4 I can see hardware decode start, but it's using s5p-mjpeg not s5p-mfc so the frame-rate is horrific, see http://ix.io/41bC. I'm wondering if this is because s5p-mjpeg advertises YUV support? so is a better match for output? - RPi dev commented "it certainly is using mjpeg but that might be subtitles or something over the main video? I don't think the old unpatched v4l2m2m code is as chatty as mine when working so its hard to tell. I'd certainly believe that your decoder is happier if it has size setup before startup which is what the older decoder did."

The upstream kernel does not have a stable UAPI for stateful (v4l2_m2m) decode yet but there was some effort towards establishing conformance tests about 18-months ago, and the Amlogic/RPi drivers (and the ffmpeg we use) are aligned to those changes. I strongly suspect the upstream ffmpeg and s5p-mfc driver are still in a state that pre-dates those tests being available, so the driver needs some work. If that's done, it should play nicely with the Amlogic/RPi ffmpeg sources (which will go upstream at some point soon) and should reduce the need for Kodi hacks (zero would be the goal). The challenge is always finding people with the talent and motivation to do low-level work on the drivers, because it's a rather niche skillset.

AreaScout, 2022-06-30:
@chewitt 

I don't know the Amlogic/RPi changes, but I remember that Lucas had all that v4l2_m2m drmPrime in work, it is hosted here

https://github.com/lrusak/FFmpeg/tree/v4l2-drmprime-v5

If that changeset is applied https://patchwork.kernel.org/project/linux-samsung-soc/cover/20190301122055.7135-1-a.hajda@samsung.com/ I would have test it with the one from Lukas

RG,
Daniel

heitbaum, 2022-08-22:
closing now that #6827 has been commited

closed 2022-08-22 by heitbaum