LibreELEC/LibreELEC.tv #6251: [BUG] KODI 19 on Raspberry Pi 4 does not remember WiFi password

jbrepogmailcom, 2022-02-18:
### Describe the bug
After upgrade to Matrix I have this problem. When KODI disconnects from WiFi sometimes, it usually does not connect back and when I go to Connection settings, the WiFi password is not remembered for my SSID and I have to input it again. This never happened with KODI 18. Any solution please? More people have this problem: https://forum.libreelec.tv/thread/25244-kodi-19-on-raspberry-pi-4-does-not-remember-wifi-password/

### To Reproduce
Steps to reproduce the behavior:
1. Connect to wifi with password
2. Turn off the wifi and then turn it off
3. KODI does not connect to WiFi and when Connection is launched manually, then KODI asks for password it should remember

### Informations
 - LE Version: 10.0.1
 - Hardware Platform: RPi 4

### Log file
nothing relevant in kodi.log except for few messages like this:
2022-02-18 14:16:17.846 T:31716    INFO <general>: JSONRPC Server: Disconnection detected
2022-02-18 14:16:22.972 T:31716    INFO <general>: JSONRPC Server: New connection added
2022-02-18 14:21:18.141 T:31716    INFO <general>: WebSocket: connection closed by client

### Additional context

TheNamelessWonderer, 2022-02-20:
Same issue here on a Pi 4 running Librelec 10.0.1. It constantly disconnects and asks for the password again making it nearly unusable on Wi-Fi

jbrepogmailcom, 2022-04-09:
Still exists in 10.0.2, It is very annoying that it worked well in version 9. Beside migration to python3, which is totally transparent to the user, the upgrade broke funcional feature.

beck-llr, 2022-04-11:
I confirm the issue is still the same on 10.0.2.

TheNamelessWonderer, 2022-07-30:
Any plans to fix this issue? LibreELEC has been unusable for me for months. Is there a workaround or are we stuck on LibreELEC 9 for the foreseeable future?

volkanpaksoy, 2022-08-21:
I may have accidentally bumped into a workaround:

I set up 2 Raspberry Pi 4 instances the same way. The only difference was for one of them, during the setup, I used a wired connection. After the LibreELEC setup finished, I switched to WiFi and that setting was remembered after reboots.

In the second installation, I entered my WiFi passphrase during the setup, but it was not remembered even after a few tries.

What I noticed is that in the first installation, "Connect Automatically" was off by default. In the second one, however, it was on. I turned it off, rebooted, entered the passphrase again and turned on Connect Automatically, and it seemed to work.

Both of these are just anecdotal. I hope someone else can give it a go and verify whether they work or not.

Hope it helps someone.

jbrepogmailcom, 2022-10-31:
I also had other problem with v10 on RasPi 4, when browsing through list of movies it was freezing very often. So I decided to downgrade to v9 Leila and I had to find all add-ons that were done in Python 2. It was still worth the effort, KODI now keeps WiFi connection and does not freeze - it just works as expected. Sorry, for me v10 is not good (said politely).

Sebazzz, 2023-03-23:
Related: #7166

closed 2023-09-04 by heitbaum

6a6ak, 2024-07-04:
I wrote this repo and solved my problem if you do so, you can solve yours 
https://github.com/6a6ak/Auto_Wifi_RapsberryPi

heitbaum, 2024-07-05:
solution was wpa_supplicant which is not used by LE.

6a6ak, 2024-07-06:
It is surprising that @LibreELEC deleted my message, especially when many messages here do not help solve the problem. If you thought I wrote this message to advertise my repository, I am truly sorry. In the open-source world, such issues are not important at all. The problem for most people facing these because can't change /etc/wpa_supplicant/wpa_supplicant.conf
 the issue is that their memory is write-protected, and they are unaware of it. Not only does this prevent file editing, but it also makes the entire system unchangeable. If my message had a technical issue, it would have been better to inform me so I could improve it rather than deleting it without my knowledge, leaving many people in confusion as before.

chewitt, 2024-07-07:
@6a6ak The "technical issue" is; your script depends on files that do not exist in our distro, and files that exist but cannot be changed due to intentional (and well-known) packaging of the distro. The comment thus proves your script has not been tested on our distro (as it could not be) and does not solve the problem and is not relevant to the problem described. If you want to improve your GitHub profile stats and standing in the open-source community, test contributions first, and don't add to user confusion by post irrelevant comments with incorrect technical advice.