LibreELEC/LibreELEC.tv #6266: [BUG] Kodi crashing under certain circumstances

pretoriano80, 2022-02-25:
### Describe the bug
Kodi is crashing when operating an addo-on, in this case it's either Netflix plugin or Tubed. 
I only tested with these plug-ins because it's easy to reproduce (100% trigger rate) 

### To Reproduce
Steps to reproduce the behavior:
1. Open Netflix (or Tubed) addon 
2. Navigate to any category (no need to play any video) 
3. Go back and open Tubed (or Netflix if you used Tubed in the first step) addon 
4.  Navigate to a category and open it, this should trigger a crash and Kodi will restart 

Note: Sometimes it may crash already at step 3.

### Informations
 - LE Version: LE11 master ae5c471
 - Hardware Platform: RPi4 8GB  - SSD as main drive
 - Clean install, only Netflix and Tubed addons were installed 

### Log file
http://ix.io/3QHR

### Additional context
An important thing i noticed is that i can't reproduce the bug if only one of the addons is installed, meaning that if only Netflix is installed, then it won't cause any crash, same thing if only Tubed addon is installed. 

I have no clue if this is a bug caused by LibreELEC or some other Kodi related component. 
I did some research and couldn't find any similar bug reports for Netflix addon, Tubed, Inputstream Adaptive or Input Stream Helper.

heitbaum, 2022-02-26:
Please have a look at; _Originally posted by @Maven85 in https://github.com/xbmc/xbmc/issues/20853#issuecomment-1025579982_ and https://forum.libreelec.tv/core/search-result/129687/?highlight=Python

going to close these 2 issues as known issues - upstream addons. If there is a specific LE bug that needs to be fixed. Please reopen. But will need to know the detail and addon. Feedback via the forum would be great too.

pretoriano80, 2022-02-26:
Ok, i did found those comments you linked, but it's the Netflix addon and none of the addons mentioned there are installed on my system.

I pinged CastagnaIT on Kodi forums about it, maybe he's aware of this issue.
It does seem that other addons are affected, not only the ones mentioned in the your links.

In the meantime on the production drive i switched back to an older nigthly that i had  around  and will keep another testing drive for further development on the matter.

HiassofT, 2022-02-26:
I suspect it might be caused by this python bug which doesn't seem to be fully fixed (python issue is still open):
https://bugs.python.org/issue46070

heitbaum, 2022-02-26:
@pretoriano80 thanks for the additional info - reopened

pretoriano80, 2022-02-27:
@heitbaum  and @HiassofT  i went further with the testing and i can reproduce the same crashing pattern with the latest Kodi nightly on my Android device, so definitely not a LibreELEC bug and not limited to LibreELEC.

However, i think that you guys have a much better way to communicate with the Kodi team directly and i'm certain that it will be solved at some point. 

P. S. Maybe it's also a good thing to keep this bug report open, so if other LibreELEC users encounter this problem, will not open duplicate bug reports.

HiassofT, 2022-02-27:
@pretoriano80 please create a kodi issue about the Android crash

pretoriano80, 2022-02-27:
Done, here it is - > https://github.com/xbmc/xbmc/issues/21045

pretoriano80, 2022-03-11:
> I suspect it might be caused by this python bug which doesn't seem to be fully fixed (python issue is still open): https://bugs.python.org/issue46070

I applied the patch provided in that bug report but it seems that the issue is still there.
I only applied it to LibreELEC so maybe patching is also required on Kodi's side, but i have to admit that i didn't looked further into it because this python thing is a bit overwhelming for me.  :)
Probably there's more than only the patch i tested, but python's bug tracking is a bit confusing for me (to say at least). 

However, over the weekend i will have some more time for testing, so if you guys have more hints about this issue i will gladly test them.

heitbaum, 2022-03-11:
Is this the patch you tried?

https://github.com/python/cpython/commit/52937c26adc35350ca0402070160cf6dc838f359

pretoriano80, 2022-03-11:
> Is this the patch you tried?
> 
> [python/cpython@52937c2](https://github.com/python/cpython/commit/52937c26adc35350ca0402070160cf6dc838f359)

Nope,i've tested the patch from here --> https://bugs.python.org/issue46070
It seems that the patch from your link it's already merged in 3.9.10.

WallyCZ, 2022-03-22:
I have built python asi debug and got something more from stacktrace:
```
Thread 1 (Thread 0xd5899280 (LWP 3040)):
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:49
#1  0xf4fb0798 in __GI_abort () at abort.c:79
#2  0xf4fbd9cc in __assert_fail_base (fmt=0xf50af7d0 "%s%s%s:%u: %s%sAssertion `%s' failed.\n%n", assertion=0xeabdfb68 "PyStruct_Check(self)", assertion@entry=0xeabdf6d0 "/libreelec/build.LibreELEC-RPi4.arm-11.0-devel/build/Python3-3.9.10/Modules/_struct.c", file=0xeabdf6d0 "/libreelec/build.LibreELEC-RPi4.arm-11.0-devel/build/Python3-3.9.10/Modules/_struct.c", file@entry=0xf50af7d0 "%s%s%s:%u: %s%sAssertion `%s' failed.\n%n", line=line@entry=1870, function=0xeabe0338 "s_pack", function@entry=0xd5899280 "\001") at assert.c:92
#3  0xf4fbda7c in __GI___assert_fail (assertion=0xeabdf6d0 "/libreelec/build.LibreELEC-RPi4.arm-11.0-devel/build/Python3-3.9.10/Modules/_struct.c", file=0xf50af7d0 "%s%s%s:%u: %s%sAssertion `%s' failed.\n%n", line=1870, function=0xd5899280 "\001") at assert.c:101
#4  0xeabdd078 in ?? () from /usr/lib/python3.9/lib-dynload/_struct.so
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

    soself = (PyStructObject *)self;
    assert(PyStruct_Check(self));
    assert(soself->s_codes != NULL);
    if (nargs != soself->s_len)
Line where it crashed was on `assert(PyStruct_Check(self))`:
```c
static PyObject *
s_pack(PyObject *self, PyObject *const *args, Py_ssize_t nargs)
{
    char *buf;
    PyStructObject *soself;

    /* Validate arguments. */
    soself = (PyStructObject *)self;
    assert(PyStruct_Check(self));
    assert(soself->s_codes != NULL);
    if (nargs != soself->s_len)
    {
        PyErr_Format(_structmodulestate_global->StructError,
            "pack expected %zd items for packing (got %zd)", soself->s_len, nargs);
        return NULL;
    }
   ...
```

prahal, 2022-03-22:
I checked and all the patches in https://bugs.python.org/issue46070, including https://github.com/python/cpython/commit/52937c26adc35350ca0402070160cf6dc838f359, are included in python 3.9.10, which is now in LibreElec nightly builds. Which still crashes to me.

Also in this bug issue 46070, there are multiple issues:
- asyncio intilialisation : bug.py crashes on python 3.8 while our bug seems python 3.9 and up, and it is fixed in python 3.9.10 by [bpo-46070](https://bugs.python.org/issue46070): Fix asyncio initialisation guard ([GH-30423](https://github.com/python/cpython/pull/30423))
- GC race triggered by _sre module exception (only with python 3.9, not python 3.10) : [win_py399_crash_reproducer.py](https://bugs.python.org/file50494/win_py399_crash_reproducer.py) seems to be windows only. I tried to run this test sixty times under odroid-c2 LE 11 chewitt (python 3.9.11) no crash. Also fixed in python 3.9.10 [bpo-46070](https://bugs.python.org/issue46070)[: _PyGC_Fini() untracks objects (](https://github.com/python/cpython/commit/52937c26adc35350ca0402070160cf6dc838f359)https://github.com/python/cpython/pull/30577[)

Any patches to debug by printing the interpreter of the obj are pending but no more fix is left in this 46070 report. Though I admit our issue might be related to the objects shared between interpreters, we will have to add to the issue for it to go further.

My own issue is [Kodi crashes when trying to play a second video, using Jellyfin add-on](https://forum.libreelec.tv/thread/25376-kodi-crashes-when-trying-to-play-a-second-video-using-jellyfin-add-on/?postID=167574#post167965) which could be close to this one.
I also can reproduce with the latest kodi nightly on android.
The backtrace in the log ends up in libpython3.9.so.1.0 with a corrupted stack (previous frame identical to this frame).

pretoriano80, 2022-03-22:
@WallyCZ the log is from a crash caused by the issue i presented heee, when switching from an add-on to another?

@prahal apart from the issue you have with jellyfin, did you also tried to reproduce the issue from this bug report?
Yes, i also checked all those patches and  are indeed merged, so probably the log from WallyCZ is more relevant at this point.

Unfortunately, for a proper bug report to python devs, an easy way to reproduce the issue must be provided which in our case it is not (install kodi, addons, etc).
Probably this bug will become more relevant if it's not fixed until Kodi 20 betas or RCs are released.

prahal, 2022-03-22:
@pretoriano80 yes but I do not have netflix. I reproduced with jellyfin your use case. I go in jellyfin addon, open a menu then go to another addon namely Dailymotion, then back in jellyfin: crash.

About WallyCZ log, I am trying to reproduce. Rebuilding LibreElec with DEBUG=Python3,kodi . His crash report gave me hope that a test case could be build.

pretoriano80, 2022-03-22:
@prahal  great, i also managed to reproduce it without Netflix, so definitely there's more about this bug, not limited to only a few addons.
Yes, if we manage to provide some accurate logs, then probably Kodi devs will take it more seriously,especially because we are talking about a bug triggered by 3rd party addons annd sometimes Kodi team ignore these reports.

prahal, 2022-03-23:
@pretoriano80 you should reproduce with a debug build : "DEBUG=Python3,kodi" added to "make release".
My attempt with debug on was of a proper backtrace.
python 3.9.11 https://paste.kodi.tv/zebiyopoge.kodi
python 3.10.3 https://paste.kodi.tv/zebiyopoge.kodi
So this issue is one that is not already fixed in python 3.10.3

pretoriano80, 2022-03-23:
Ok,so i managed to get more crash logs from LE11 debug build.

- YouTube crashing  --> [kodi_crashlog_youtube.log](https://github.com/LibreELEC/LibreELEC.tv/files/8335510/kodi_crashlog_youtube.log)
This is from a clean build,no other addons were installed and YouTube crashed on first try,no addon switching required.
- Netflix crashing --> [kodi_crashlog_netflix.log](https://github.com/LibreELEC/LibreELEC.tv/files/8335530/kodi_crashlog_netflix.log)
This was also on clean install with only Netflix installed.
To make sure that Netflix wasn't crashing without the addon switch combo,i used this steps: Open Netflix -> Exit Netflix -> Return to Netflix -> Exit Netflix -> Open Libreelec Configuration addon -> Exit -> Open Netflix (kodi crashed)

 **Note:** It seems that other addons i've tested (eg. Tubed) aren't crashing if YouTube or Netflix addons are not in the mix (aka disabled or not installed), but they do crash with the "addon switch combo" and YT or Netflix involved.
I still have to figure out how all these are related ,but i thought it's better to share my complete testing experience.

prahal, 2022-03-24:
we have the same traces (Thread 1, python 3.9 one of mine). My first attempt at a fix was to check python 3.10 changes not backported to 3.9 but I found out that 3.10 was also broken.
I believe it is time to ask python devs to ask them for help in building a smaller test case (from our traces maybe they can hint).
I have a few more tests to attempt to check the second playback crash in jellyfin is the same trace as the enter jellyfin addon, then another addon, then back to jellyfin.
Feel free to beat me at finding a test case :-)
I believe @WallyCZ issue is not the same as ours.

WallyCZ, 2022-03-24:
That can be the same issue as yours even if stack trace is different since I used the build option `--with-pydebug` and `CFLAGS='-DPy_DEBUG'`, what you probably haven't. This will enable some python asserts, which can detect the bugs much more sooner than by you. It can have primary and secondary cause. But that's the theory, of course it can be something different. I have noticed that you have always the sqlite module in your stack traces, so this could be even bug only in sqlite module. Unfortunately, I had to switch to python 3.8 so now I can't do more explorations. I am pretty sure it can be debugged via gdb and resolved even without the deep knowledge of the python internals.

prahal, 2022-03-24:
@WallyCZ thanks. Could you paste the full build command for --with-pydebug ?
I agree no need for deep python internals knowledges. I was more into deep knowledge of python current weaknesses :-) ie https://bugs.python.org/issue46070 showed me that the fact that there were bugs related to global state used by different interpreters was a known issue for python devs.
Mind I am also making in kodi addon internals. I cannot tell if switching from an addon to the other keep the previously loaded addon state. So kodi devs could help in narrowing a test case in my opinion (this time by knowing the code).

WallyCZ, 2022-03-24:
Here is Python's package.mk I used https://paste.kodi.tv/acawitaqiy

prahal, 2022-03-24:
running this on the kodi Odroid C2 with python 3.9.11 segfault:
python3.9 sqlite3_crash.py 
```
# When this program is run on windows using python 3.9.9 it crashes about 50%
# of the time.

import _testcapi
import threading

code = """
import sqlite3
"""

def doIt():
    _testcapi.run_in_subinterp(code)

tt=[]

for i in range(16):
    t = threading.Thread(target=doIt)
    t.start()
    tt.append(t)

for t in tt:
    t.join()
```

This thanks to https://forum.libreelec.tv/thread/25454-generic-nightly-kodi-restarts-after-loading-sqlite3/ which found that an addon with only import sqlite3 in crashes kodi.

I extracted most of the test case from the initial https://bugs.python.org/issue46070 test case for the windows issue [win_py399_crash_reproducer.py](https://bugs.python.org/file50494/win_py399_crash_reproducer.py).

That should be enough for a python upstream bug report. Still I will do more tests on my side before that so feel free to open it beforehand.
I noticed than on x86_64 python 3.9.2 (debian stable) the segfault does not happen. I want to bisect the python 3.9 code down to a commit.

pretoriano80, 2022-03-24:
For now i won't open a bug report with python devs for several reasons.
First, i don't have python skills so this means that it may become a communication "barrier"  between me and the devs if more technical discussions are required.

Another reason would be the testing environment, meaning that probably they won't like the idea of testing this on LibreELEC (even if it's clear that this is not caused by LibreELEC).

Probably it would be best if one of the developers behind the affected addons (Netflix,YouTube, etc) open a bug report with cpython devs and i'm saying this because they know the code used in their addons, hence the capacity to better understand the technical details. 
And lastly,, Kodi devs may also do this if they want or have enough free time to help on this matter.

prahal, 2022-03-24:
I can reproduce with a vanilla build of cpython github 3.9.11 tag on x86_64 (I checked a few runs and even on LibreElec the test case only crashes 90% of the time, so if it works the first time, retry). This vanilla build was without LE patches to python.
```
(gdb) py-list
  75    
  76            val = datetime.datetime(year, month, day, hours, minutes, seconds, microseconds)
  77            return val
  78    
  79    
 >80        register_adapter(datetime.date, adapt_date)
  81        register_adapter(datetime.datetime, adapt_datetime)
  82        register_converter("date", convert_date)
  83        register_converter("timestamp", convert_timestamp)
  84    
  85    register_adapters_and_converters()
```

python backtrace : 
```
(gdb) thread apply all py-bt

Thread 7 (Thread 0x7ffff4dfe700 (LWP 436592) "python"):
Traceback (most recent call first):
  <built-in method register_adapter of module object at remote 0x7fffdc2e5a40>
  File "/home/prahal/Projects/WIP/libreelec/cpython_vanilla/Lib/sqlite3/dbapi2.py", line 80, in register_adapters_and_converters
    register_adapter(datetime.date, adapt_date)
  File "/home/prahal/Projects/WIP/libreelec/cpython_vanilla/Lib/sqlite3/dbapi2.py", line 85, in <module>
    register_adapters_and_converters()
  <built-in method exec of module object at remote 0x7ffff45edb80>
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "/home/prahal/Projects/WIP/libreelec/cpython_vanilla/Lib/sqlite3/__init__.py", line 57, in <module>
    from sqlite3.dbapi2 import *
  <built-in method exec of module object at remote 0x7ffff45edb80>
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<string>", line 2, in <module>
  <built-in method run_in_subinterp of module object at remote 0x7ffff77bc180>
  File "/home/prahal/Projects/WIP/libreelec/cpython_vanilla/../python-crash-kodi/sqlite3_crash.py", line 11, in doIt
    _testcapi.run_in_subinterp(code)
  File "/home/prahal/Projects/WIP/libreelec/cpython_vanilla/Lib/threading.py", line 910, in run
    self._target(*self._args, **self._kwargs)
  File "/home/prahal/Projects/WIP/libreelec/cpython_vanilla/Lib/threading.py", line 973, in _bootstrap_inner
    self.run()
  File "/home/prahal/Projects/WIP/libreelec/cpython_vanilla/Lib/threading.py", line 930, in _bootstrap
    self._bootstrap_inner()

Thread 1 (Thread 0x7ffff7c45280 (LWP 436579) "python"):
Traceback (most recent call first):
  File "/home/prahal/Projects/WIP/libreelec/cpython_vanilla/Lib/threading.py", line 1073, in _wait_for_tstate_lock
    if lock.acquire(block, timeout):
  File "/home/prahal/Projects/WIP/libreelec/cpython_vanilla/Lib/threading.py", line 1053, in join
    self._wait_for_tstate_lock()
  File "/home/prahal/Projects/WIP/libreelec/cpython_vanilla/../python-crash-kodi/sqlite3_crash.py", line 21, in <module>
--Type <RET> for more, q to quit, c to continue without paging--
    t.join()

```

pretoriano80, 2022-03-24:
> 

> @pretoriano80 Oh sorry you already opened [xbmc/xbmc#21045](https://github.com/xbmc/xbmc/issues/21045) . Then I will do my side :-)

No problem! Yeah, the Kodi report was on Android, not so easy to debug as on LibreELEC.
I also pinged @CastagnaIT  on Kodi forums about it, he couldn't reproduce it, but that's because he only has Netflix addon installed and it won't crash (tested myself).
YouTube addon is more prone to crashing, even without any other addon being involved.

pretoriano80, 2022-03-25:
Here's the journalctl log from the Netflix addon on a debug build - >  http://ix.io/3TnX

It seems that it's indeed related to sqlite3.

prahal, 2022-03-26:
@WallyCZ are you sure you did a make clean before make release? When I add your changes to Python3 package.mk I get:
```
<<< Pillow:target seq 256 <<<
(...)
BUILD      Pillow (target)
    TOOLCHAIN      manual
running build
(...)
building 'PIL._imagingmath' extension
/home/prahal/Projects/WIP/libreelec/LibreELEC.tv/build.LibreELEC-AMLGX.arm-11.0-devel/toolchain/bin/armv8a-libreelec-linux-gnueabihf-gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -march=native -O2 -Wall -pipe -I/home/prahal/Projects/WIP/libreelec/LibreELEC.tv/build.LibreELEC-AMLGX.arm-11.0-devel/toolchain/include -Wno-format-security -march=native -O2 -Wall -pipe -I/home/prahal/Projects/WIP/libreelec/LibreELEC.tv/build.LibreELEC-AMLGX.arm-11.0-devel/toolchain/include -Wno-format-security -march=armv8-a+crc -mtune=cortex-a53 -mabi=aapcs-linux -Wno-psabi -Wa,-mno-warn-deprecated -mfloat-abi=hard -mfpu=neon-fp-armv8 -Wall -pipe -O2 -fomit-frame-pointer -DNDEBUG -fPIC -I/home/prahal/Projects/WIP/libreelec/LibreELEC.tv/build.LibreELEC-AMLGX.arm-11.0-devel/toolchain/armv8a-libreelec-linux-gnueabihf/sysroot/usr/include/freetype2 -I/home/prahal/Projects/WIP/libreelec/LibreELEC.tv/build.LibreELEC-AMLGX.arm-11.0-devel/toolchain/armv8a-libreelec-linux-gnueabihf/sysroot/usr/include -I/home/prahal/Projects/WIP/libreelec/LibreELEC.tv/build.LibreELEC-AMLGX.arm-11.0-devel/toolchain/armv8a-libreelec-linux-gnueabihf/sysroot/usr/include/fribidi -I/home/prahal/Projects/WIP/libreelec/LibreELEC.tv/build.LibreELEC-AMLGX.arm-11.0-devel/toolchain/include -I/home/prahal/Projects/WIP/libreelec/LibreELEC.tv/build.LibreELEC-AMLGX.arm-11.0-devel/toolchain/armv8a-libreelec-linux-gnueabihf/sysroot/usr/include/python3.9 -c src/_imagingmath.c -o build/temp.linux-x86_64-3.9/src/_imagingmath.o
src/_imaging.c:75:10: fatal error: Python.h: No such file or directory
   75 | #include "Python.h"
      |          ^~~~~~~~~~
compilation terminated.
src/_imagingft.c:22:10: fatal error: Python.h: No such file or directory
   22 | #include "Python.h"
      |          ^~~~~~~~~~
compilation terminated.
```

This because Pillow includes -I/home/prahal/Projects/WIP/libreelec/LibreELEC.tv/build.LibreELEC-AMLGX.arm-11.0-devel/toolchain/armv8a-libreelec-linux-gnueabihf/sysroot/usr/include/python3.9 instead of the now -I/home/prahal/Projects/WIP/libreelec/LibreELEC.tv/build.LibreELEC-AMLGX.arm-11.0-devel/toolchain/armv8a-libreelec-linux-gnueabihf/sysroot/usr/include/python3.9d (note the d added when I add py-debug and your CFLAGS.

I can work around this pillow issue if I symlink the includes from python3.9d to python3.9 pillow in the build and redo the make release.

But then kodi build fails with:
```
<<< kodi:target seq 266 <<<
BUILD      kodi (target) [DEBUG]
    TOOLCHAIN      cmake (auto-detect)
(...)
-- Configuring done
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
Python3_LIBRARY
    linked by target "libkodi" in directory /home/prahal/Projects/WIP/libreelec/LibreELEC.tv/build.LibreELEC-AMLGX.arm-11.0-devel/build/kodi-2f11e994f322f7376476b186293a3df34246b9a7
    linked by target "kodi" in directory /home/prahal/Projects/WIP/libreelec/LibreELEC.tv/build.LibreELEC-AMLGX.arm-11.0-devel/build/kodi-2f11e994f322f7376476b186293a3df34246b9a7

-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    ENABLE_LDGOLD

CMake Generate step failed.  Build files cannot be regenerated correctly.
FAILURE: scripts/build kodi:target during configure_target (default)
*********** FAILED COMMAND ***********
cmake ${CMAKE_GENERATOR_NINJA} ${TARGET_CMAKE_OPTS} ${PKG_CMAKE_OPTS_TARGET} ${PKG_CMAKE_SCRIPT%/*}
**************************************
*********** FAILED COMMAND ***********
${SCRIPTS}/build "${1}" "${PARENT_PKG}"
**************************************
FAILURE: scripts/install kodi:target has failed!
```

I have your --with-pydebug in PKG_CONFIGURE_OPTS_TARGET, CFLAGS='-DPy_DEBUG' in pre_configure_target and symlinking of python d suffixed libraries to non suffixed one in Python3 package.mk post_makeinstall_target

NB: I noticed when one upgrade the python to build if the target python library name change I usually get a missing library from dependent packages and kodi does not start. A make clean is then required.

pretoriano80, 2022-03-26:
@prahal  i used his package.mk that he provided here and no "make clean" was required here.

prahal, 2022-03-26:
@pretoriano80 sorry, I meant you have to make clean if you want a proper build when one change python suffix or version number (which is the case in this pythn debug build) else your build still contains the old python suffix and the build succeed but is not fine (at runtime it still attempt to load the python library with the old suffix).
Could you retry to make release but this time after a make clean?

WallyCZ, 2022-03-26:
@prahal I didn't make clean. If you do this, you will need couple of additional changes in build scripts because dependencies will try link with incorrect library (without "d"). Easiest way is to build it with original make files, then replace that package.mk and build it again without doing make clean. The scripts contains symlink trick so every dependency will use debug version even if it was linked with non-debug.

prahal, 2022-03-27:
Thanks for this last tip that help me reproduce your struct bug. It turns out that yours is very likely a different bug. The struct crash happens even if you only wait doing nothing. From my logs, I suspect script.tubed to trigger it.
My kodi logs end up with:
```
2022-03-28 00:10:00.432 T:674      info <general>: [script.tubecast] ssdp: Starting SSDP server
```
and more importantly, I get this debug trace with gdbserver and sourcing the LE build python-gdb.py into gdb-multiarch client:
```
Thread 4 "LanguageInvoker" received signal SIGABRT, Aborted.
[Switching to Thread 1086.1111]
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:49
49	  return ret;
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:49
#1  0xf5494798 in __GI_abort () at abort.c:79
#2  0xf54a19cc in __assert_fail_base (fmt=0xf55937d0 "%s%s%s:%u: %s%sAssertion `%s' failed.\n%n", assertion=0xe7123ba4 "PyStruct_Check(self)", 
	assertion@entry=0xe71236d0 "/home/prahal/Projects/WIP/libreelec/LibreELEC.tv/build.LibreELEC-AMLGX.arm-11.0-devel/build/Python3-3.9.12/Modules/_struct.c", 
	file=0xe71236d0 "/home/prahal/Projects/WIP/libreelec/LibreELEC.tv/build.LibreELEC-AMLGX.arm-11.0-devel/build/Python3-3.9.12/Modules/_struct.c", 
	file@entry=0xf55937d0 "%s%s%s:%u: %s%sAssertion `%s' failed.\n%n", line=line@entry=1870, function=0xe7124388 <PRETTY_FUNCTION.9> "s_pack", 
	function@entry=0xdd070280 "\001") at assert.c:92
#3  0xf54a1a7c in GI_assert_fail (
	assertion=0xe71236d0 "/home/prahal/Projects/WIP/libreelec/LibreELEC.tv/build.LibreELEC-AMLGX.arm-11.0-devel/build/Python3-3.9.12/Modules/_struct.c", 
	file=0xf55937d0 "%s%s%s:%u: %s%sAssertion `%s' failed.\n%n", line=line@entry=1870, function=0xdd070280 "\001") at assert.c:101
#4  0xe7121078 in s_pack (self=<unknown at remote 0xdd070280>, args=0xf55937d0, args@entry=0xce375228, nargs=-418236720, nargs@entry=1)
	at /home/prahal/Projects/WIP/libreelec/LibreELEC.tv/build.LibreELEC-AMLGX.arm-11.0-devel/build/Python3-3.9.12/Modules/_struct.c:1870
#5  0xe71211ac in pack (self=<optimized out>, args=0xce375224, nargs=2)
	at /home/prahal/Projects/WIP/libreelec/LibreELEC.tv/build.LibreELEC-AMLGX.arm-11.0-devel/build/Python3-3.9.12/Modules/_struct.c:2177
#6  0xf765dcbc in cfunction_vectorcall_FASTCALL (func=func@entry=<built-in method pack of module object at remote 0xcafce9f8>, args=args@entry=0xce375224, 
	nargsf=nargsf@entry=2147483650, kwnames=kwnames@entry=0x0)
	at /home/prahal/Projects/WIP/libreelec/LibreELEC.tv/build.LibreELEC-AMLGX.arm-11.0-devel/build/Python3-3.9.12/Objects/methodobject.c:430
#7  0xf773e180 in _PyObject_VectorcallTstate (tstate=0xf2d0a940, callable=<built-in method pack of module object at remote 0xcafce9f8>, args=0xce375224, nargsf=2147483650, 
	kwnames=0x0) at /home/prahal/Projects/WIP/libreelec/LibreELEC.tv/build.LibreELEC-AMLGX.arm-11.0-devel/build/Python3-3.9.12/Include/cpython/abstract.h:118
#8  0xf774247c in PyObject_Vectorcall (kwnames=0x0, nargsf=2147483650, args=0xce375224, callable=<built-in method pack of module object at remote 0xcafce9f8>)
	at /home/prahal/Projects/WIP/libreelec/LibreELEC.tv/build.LibreELEC-AMLGX.arm-11.0-devel/build/Python3-3.9.12/Include/cpython/abstract.h:127
#9  call_function (tstate=tstate@entry=0xf2d0a940, pp_stack=0xdd06eebc, pp_stack@entry=0xdd06eeb4, oparg=oparg@entry=2, kwnames=0x0)
	at /home/prahal/Projects/WIP/libreelec/LibreELEC.tv/build.LibreELEC-AMLGX.arm-11.0-devel/build/Python3-3.9.12/Python/ceval.c:5077
#10 0xf774e0c8 in _PyEval_EvalFrameDefault (tstate=0xf2d0a940, f=<unknown at remote 0xce1190f4>, throwflag=<optimized out>)
	at /home/prahal/Projects/WIP/libreelec/LibreELEC.tv/build.LibreELEC-AMLGX.arm-11.0-devel/build/Python3-3.9.12/Python/ceval.c:3489
#11 0xf75f9c8c in _PyEval_EvalFrame (throwflag=0, f=
	Frame 0xce3750e0, for file /storage/.kodi/addons/script.tubecast/resources/lib/tubecast/ssdp.py, line 63, in set_loopback_mode (self=<MulticastServer(server_address=('0.0.0.0', 1900), RequestHandlerClass=<type at remote 0xce36da40>, _BaseServeris_shut_down=<Event(_cond=<Condition(_lock=<_thread.lock at remote 0xce371608>, acquire=<built-in method acquire of _thread.lock object at remote 0xce371608>, release=<built-in method release of _thread.lock object at remote 0xce371608>, _waiters=<collections.deque at remote 0xf2de0a30>) at remote 0xce372ad0>, _flag=False) at remote 0xce371638>, _BaseServershutdown_request=False, socket=<socket at remote 0xce374d88>, _thread=None, poll_interval=<float at remote 0xce34c940>, _handler=<type at remote 0xce36da40>, chromecast_addr=('0.0.0.0', 39141), _multicast_address=('239.255.255.250', 1900), _listen_interfaces=None) at remote 0xce3716b0>, mode=1), tstate=0xf2d0a940)
	at /home/prahal/Projects/WIP/libreelec/LibreELEC.tv/build.LibreELEC-AMLGX.arm-11.0-devel/build/Python3-3.9.12/Include/internal/pycore_ceval.h:40
#12 function_code_fastcall (tstate=0xf2d0a940, co=<optimized out>, args=0xce374cf4, args@entry=0xce374cec, nargs=nargs@entry=2, 
	globals={'name': 'resources.lib.tubecast.ssdp', 'doc': None, 'package': 'resources.lib.tubecast', 'loader': <SourceFileLoader(name='resources.lib.tubecast.ssdp', path='/storage/.kodi/addons/script.tubecast/resources/lib/tubecast/ssdp.py') at remote 0xcaf21d40>, 'spec': <ModuleSpec(name='resources.lib.tubecast.ssdp', loader=<...>, origin='/storage/.kodi/addons/script.tubecast/resources/lib/tubecast/ssdp.py', loader_state=None, submodule_search_locations=None, _set_fileattr=True, _cached='/storage/.kodi/addons/sc [... cut ...]
Note the `Frame 0xce3750e0, for file /storage/.kodi/addons/script.tubecast/resources/lib/tubecast/ssdp.py, line 63, in set_loopback_mode (self=<MulticastServer(server_address=('0.0.0.0', 1900), RequestHandlerClass=<ty` in frame 11.

that is by:
```
<kodi box>
# gdbserver  localhost:1234 /usr/lib/kodi/kodi.bin

<build host>
$ gdb-multiarch build.LibreELEC-AMLGX.arm-11.0-devel/build/kodi-2f11e994f322f7376476b186293a3df34246b9a7/.armv8a-libreelec-linux-gnueabihf/kodi.bin

(gdb) source -v build.LibreELEC-AMLGX.arm-11.0-devel/build/Python3-3.9.12/.armv8a-libreelec-linux-gnueabihf/python-gdb.py 
(gdb) c
```

prahal, 2022-03-27:
@WallyCZ sorry I forgot to add that the current bug is fixed by reverting cpython commit 52937c26adc35350ca0402070160cf6dc838f359 (for python 3.9). But yours is still there even with that commit reverted.

pretoriano80, 2022-03-28:
@prahal  you meant script.tubecast?Tubed it's a different addon.

prahal, 2022-03-28:
@pretoriano80 thank you. @WallyCZ Indeed your issue is triggered by script.tubecast (not script.tubed)

pretoriano80, 2022-03-28:
> @WallyCZ sorry I forgot to add that the current bug is fixed by reverting cpython commit 52937c26adc35350ca0402070160cf6dc838f359 (for python 3.9). But yours is still there even with that commit reverted.

I will revert that commit and try a build without it.
However, if that's the case, then python team will have to find a way out of this. xD

prahal, 2022-03-28:
@pretoriano80 my pending push request is in http://sprunge.us/o9kNvR . I also told about this fix in https://bugs.python.org/issue46070 , no reply yet. Note that in 3.11 this commit is fine. It only breaks previous branches where it was backported.

heitbaum, 2022-03-28:
> @pretoriano80 my pending push request is in http://sprunge.us/o9kNvR . I also told about this fix in https://bugs.python.org/issue46070 , no reply yet. Note that in 3.11 this commit is fine. It only breaks previous branches where it was backported.

@prahal / @pretoriano80  - if this indeed fixes the issues you are experiencing in 3.9.y - raise a PR to include this patch in LE11?

pretoriano80, 2022-03-28:
> @prahal / @pretoriano80 - if this indeed fixes the issues you are experiencing in 3.9.y - raise a PR to include this patch in LE11?

I'm testing a LE11 build with patched python 3.9.12 right now, :)

WallyCZ, 2022-03-28:
> @pretoriano80 thank you. @WallyCZ Indeed your issue is triggered by script.tubecast (not script.tubed)

Thank you for your analysis and it looks clear now that it is another issue. And indeed, I am confirming issues with this plugin even with non-debug version (more rare crashes). It crashes Kodi in about 20% of cases I am using it with the release version of python. Although I am not sure if the problem is in the addon itself because in my opinion no addon should crash Kodi even if it would do whatever. Probably should open new ticket on this, but wonder we have enough of information so somebody would bother with this...

pretoriano80, 2022-03-28:
@prahal i can confirm that reverting that commit fixed the crashing ,tested with YouTube addon and Netflix .
I only tested with these ones because it was reproducible at 100%, so most likely will also fix other addons affected by this bug.

You said that you already have the PR on hold, so in this case feel free to raise the PR with LibreELEC.

pretoriano80, 2022-03-29:
If someone wants to test the patch before a PR is raised,here's my branch with the patch -> https://github.com/pretoriano80/LibreELEC.tv/tree/revert-bpo46070

prahal, 2022-03-30:
@WallyCZ you should definitely open a bug report, against the addon. Even if it is a python issue the addon manager might be able to help you narrow a test case. You have enough debug if this is an addon issue.
The fact that you cannot produce the LibreElec pydebug build after a make clean is a bad point. If you could get trace with a non pydebug build but with DEBUG="Python3,kodi" it will help check it is not related to this nonclean build.

I will not investigate further this issue for a long while as I have more pressing issues.
At least I had no crash with non pydebug build though I have the addon that crashes with it, ie script.tubecase.

closed 2022-03-31 by CvH