LibreELEC/LibreELEC.tv #6284: [BUG] Screen constantly blanks out with RetroPlayer on Tanix-TX6 devices

danboid, 2022-03-08:
### Describe the bug

The screen blacks out after a few seconds when running RetroPlayer add-ons, or at least it does with the MAME 2000 - 2003 add-ons, as if its gone into a power saving mode. You can turn the screen back on my moving the mouse / pointer but you have to dismiss the game menu and it will do it again after a few seconds so RetroPlayer is unusable.

This is the case with both a Tanix TX6-H an a TX6 clone H10-play and it happens on both displays I tried. I tried changing the RetroPlayer video mode / aspect but it didn't make a difference. Video playback eg from Vimeo is fine under LE on the same device.

### To Reproduce
Install one of the MAME (2000-2003) add-ons on an Allwinner H6 device and load a MAME rom by right-clicking its zip file in the file manager and load it with RetroPlayer.

### Informations
 - LE Version: 10.0.1
 - Hardware Platform: Tested on a Tanix TX6-H and and H10-Play TV box (a TX6 clone)

jernejsk, 2022-03-08:
Please provide output of pastekodi

danboid, 2022-03-08:
http://ix.io/3RHO

jernejsk, 2022-03-08:
I suppose 4k@60 resolution is used on LG TV? I added a fix for that to LE 10.0.2, which should be released soon.

danboid, 2022-03-08:
Yes I think thats its default res and refresh rate. Let me know when the new release is out please and I'll try again.

jernejsk, 2022-03-09:
It's released.

danboid, 2022-03-10:
This still happens under 10.0.2.

The second display that I tried RetroPlayer / MAME on is UHD, not full 4K, and it also has this issue so I was unsure about your 4K patch fixing it.

Deoptim, 2022-03-27:
The same issue on Orange PI Lite (H3 Allwinner) with 1080p@60 screen

danboid, 2022-04-10:
Today I tried the MAME 2000 plugin under the latest stable CoreELEC  (19.4 Matrix) on my X96 Air (Amlogic S905X3) TV box on the same display I was testing LE on and I had no issues with the screen blanking so I think we can rule out my display as the cause of the issue here.

jernejsk, 2022-05-07:
I just tested latest nightly image with MAME 2003 on Tanix TX6 with 4k 60 Hz monitor. ROM loads and runs just fine. I don't have any controller, so I can't play it, but graphics is fine. Can you re-test it?

jernejsk, 2022-05-07:
I see the issue on 10.0.x though, so something in nightly image fixes it. It will be hard to determine what... Suspects: kernel, mesa, kodi.

jernejsk, 2022-05-07:
@danboid Fix for this bug is https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/10033 but it's not possible to cleanly cherry-pick it. We'll discuss if it's possible to bump mesa or not.
@Deoptim I can't reproduce the issue on OrangePi Plus2E. Lite is not officially supported.

heitbaum, 2023-01-15:
Closing as is fixed in LE11.

closed 2023-01-15 by heitbaum