LibreELEC/LibreELEC.tv #6309: [BUG] LibreELEC patch for kodi prevents pulseaudio from working as expected

cr3bs, 2022-03-17:
### Describe the bug
Because of this [patch](https://github.com/LibreELEC/LibreELEC.tv/blame/master/packages/mediacenter/kodi/patches/kodi-100.15-rename-default-pulse-device.patch) pulseaudio does not work as expected, because only the "default" device is available. This breaks all other pulseaudio devices which are configured.

This patch also breaks the snapserver addon if you want to stream the kodi sound to snapserver.

Would it be possible to drop the `return` statement (line 27)  from the patch?

### Additional context
I am using LibreELEC on a Raspberry Pi 4 and I want to use snapserver/snapclient to provide multi room audio in parallel to kodi. As kodi uses ALSA this is not possible, because the sound card cannot use [dmix](https://patchwork.kernel.org/project/alsa-devel/patch/1488451773-22010-1-git-send-email-boris.brezillon@free-electrons.com/). So the only option left for the moment is to use pulseaudio.

### Informations
 - LE Version: 10.0.0
 - Hardware Platform: RPi4

CvH, 2022-03-17:
maybe @fritsch or @lrusak can remember the usecase for this patch ?

lrusak, 2022-03-17:
> maybe @fritsch or @lrusak can remember the usecase for this patch ?
> 
> 

The usecase is that we always run pulse audio and we only officially use it for Bluetooth audio. So we only have the default pulse device listed to avoid confusion.

You can still use this device with any other manual pulse configuration with `pactl set-default-sink NAME`, then verify the default device with `pactl info`.

closed 2022-03-17 by lrusak

cr3bs, 2022-03-17:
@lrusak Actually you can't. At least, that won't work (with passthrough). I have a workaround for this, by adding a null-sink so that I would not hit the default case.

lrusak, 2022-03-17:
The default sink is a null-sink until another sink is added. https://github.com/LibreELEC/LibreELEC.tv/blob/859d6eb63bd226fd01733c3b07b118b1d6473cda/packages/audio/pulseaudio/config/system.pa#L36

I'm not sure why you think you can't make another sink the default though.

I would make a forum thread about your usecase.

cr3bs, 2022-03-17:
I did not say that I can't make another sink a default sink. I did say that it does not work with passthrough.

If you look at [this](https://github.com/xbmc/xbmc/blob/af1846c37bc9997c65d97881237c13ab8af9736c/xbmc/cores/AudioEngine/Sinks/AESinkPULSE.cpp#L490) case, which is what the patch is modifying, you see that the return statement will exit immediately and everything else in that method will be skipped.

lrusak, 2022-03-17:
It's likely not going to work with passthrough regardless.

If you look at the output of `pactl list sinks` and `pactl list cards` you will see if a sink supports passthrough or not. You also have to manually activate passthrough for a given sink using `pactl set-sink-formats ...`.

So if your idea is to use some special sink (that doesn't actually support passthrough) you will be fine with this patch.

lrusak, 2022-03-17:
I'm not sure what you think passthrough does. Unless you have a device that actually accepts passthrough it's useless to try and use it.

What you have there doesn't mean passthrough is used. You still need to activate it.

See: https://kodi.wiki/view/PulseAudio#Passthrough_Mode

cr3bs, 2022-03-17:
I know what passthrough does, and I have a device that is happy to accept DTS etc. And with the above calls it does. Without the null sink it does not, because kodi does not recognize that it has passthrough capabilities. Do you have a device that allows passthrough via HDMI and a raspberry pi 4? If so, please try my autostart.sh with and without out the null sink and see for yourself.

lrusak, 2022-03-17:
So why aren't you using ALSA?

cr3bs, 2022-03-17:
Because of what I wrote above:

> I am using LibreELEC on a Raspberry Pi 4 and I want to use snapserver/snapclient to provide multi room audio in parallel to kodi. As kodi uses ALSA this is not possible, because the sound card [...]

lrusak, 2022-03-17:
Passthrough is exclusive though, so you can't use two sources when passthrough is active.

Is your idea to have one device so you don't have to switch audio output devices in Kodi? If so that's your choice but our typical usecase does not want to use pulseaudio for all Kodi output.

cr3bs, 2022-03-17:
I understand that this is not the default use case for LibreELEC. I also understand what passthrough does. I also understand that it is exclusively using the device. All I want is to be able to select a output device from pulseaudio that is different from the default device, because the patch I mentioned above skips about 70 lines of code, which will actually make passthrough work. I want to be able to decide on my own, which output device from pulseaudio I want to use. My use case is as stated above, I want to be able to use snapserver/snapclient addons to provide multi room audio. Of course I do not want to play audio via snapserver/snapclient while watching a movie that requested passthrough via pulseaudio.

fritsch, 2022-03-17:
The patch only had the default usecase in mind, it was not meant as a generic solution for anything. Therefore I don't see a solution without reworking exactly this by either removing it and building it yourself or contributing a solution that continues to work as is for all LE users minus you and you on top.

cr3bs, 2022-03-17:
Absolutely! I do understand the default use case and that you do want to avoid confusion. Just wanted to discuss if it would be possible to somehow allow both uses cases. Removing the `return`statement would allow to select any configured sink, which would by default not include any sound cards as moduel-udev-detect is not loaded and make passthrough work.

cr3bs, 2022-03-17:
This is a log of the relevant sections from the kodi.log with and without the `null-sink`. Notice that `alsa_output.0.hdmi-stereo` does not appear in the log without the sink and also that the "default" sink in the log without the `null-sink` does not have passthrough capabilities.

[kodi.log](https://github.com/LibreELEC/LibreELEC.tv/files/8292648/kodi.log)

fritsch, 2022-03-18:
The point with the udev module is good one. We can check if it is loaded and if yes continue!

There is another module check already there. Want to give it a try?

cr3bs, 2022-03-19:
Your suggestion would work but IMHO it would also make the patch more complex and harder to maintain.

I have another suggestion though. Maybe an environment variable could be used? So the default would be as is with the current patch and if the environment variable is set it would skip the return statement. This could be documented in the wiki and set via `systemctl edit kodi.service` and maybe even integrated into libreelec settings in the future?

fritsch, 2022-03-19:
Not on me to decide this - your approach is a question to LE devs if they want to carry / maintain such a patch, while I agree maintaining the "return" one or maintaining the env version is not much of a difference on kodi side, not so sure what pulse audio configuration questions will come up after that.

cr3bs, 2022-03-19:
So the question would be the same if I were to check for the loaded module?

fritsch, 2022-03-19:
Slight difference: in the udev module case the user has already properly configured Pulseaudio :-).

Still: not my take. I was asked to make this least intrusive hack around 10 years ago, roughly :-) and it worked until now.