LibreELEC/LibreELEC.tv #6332: How to add kernel modules to INITRAMFS correctly ?

150balbes, 2022-03-28:
How to correctly add the procedure for adding certain kernel modules to INITRAMFS to the build system ? Via the virtual\initramfs package or via the kernel package?

chewitt, 2022-03-28:
initramfs uses different userspace to normal boot, but the same kernel, so mod the kernel

closed 2022-03-28 by chewitt

150balbes, 2022-03-28:
If everything was so simple and the issue was solved by modifying the kernel, I wouldn't ask the question. And before closing the question, try to understand what is at stake.

chewitt, 2022-03-28:
Please ask the Q in Slack with more detail then. This is a bug tracker and this is not a bug report.

150balbes, 2022-03-28:
> This is a bug tracker and this is not a bug report

OK, I get you

> Please ask the Q in Slack with more detail then.

I can't log in to Slack.

vpeter4, 2022-03-28:
Start from here and then go backwards: https://github.com/LibreELEC/LibreELEC.tv/commit/9b10ad20d54d906f3d4143ad21bd89d801f82b8d

150balbes, 2022-03-28:
> Start from here and then go backwards: [9b10ad2](https://github.com/LibreELEC/LibreELEC.tv/commit/9b10ad20d54d906f3d4143ad21bd89d801f82b8d)

Tnx