LibreELEC/LibreELEC.tv #6342: [FEATURE]

racerlucas, 2022-03-29:
Please integrate the driver for Xbox One wireless dongle. It a real pain to use the Bluetooth for the connection on a PC.

SupervisedThinking, 2022-03-29:
you something like https://github.com/medusalix/xone ?

racerlucas, 2022-03-29:
Yes

SupervisedThinking, 2022-03-29:
I have a package in my repo https://github.com/SupervisedThinking/LibreELEC-RR/blob/master-rr/packages/supervisedthinking/linux-drivers/xone/package.mk guess we have to check if it works for everyone

racerlucas, 2022-03-29:
I know that your Libreelec-RR is great and have support for the dongle. But we don't have apt or git or curl in the vanilla Libreelec.tv to install the driver.

All I want is just an HTPC with kodi and Moonlight to stream some games from my gaming PC and watch some Blu-ray videos from my NAS. And I definitely  don't want to use mouse and keyboard to control the HTPC on the couch.

BTW, the only reason I haven't tried your repo is that I can't find a way to install it on a VM to test it.

racerlucas, 2022-03-30:
So the dongle is still unavailable in the LE-RR 10.x?

SupervisedThinking, 2022-03-30:
> Really? How to load it? Do I need anything like the xone?

https://wiki.libreelec.tv/how-to/add-firmware

> So the dongle is still unavailable in the LE-RR 10.x?

I've never added xone to 10.x since I'm not using at at any system anymore - I'm just rebasing the packages from time to time.
Anyway this kernel driver addon is probably something that can / should be added to upstream since the Xbox controller are pretty widespread

SupervisedThinking, 2022-03-30:
@chewitt @CvH @HiassofT what's the general approach if it comes to kernel drivers for peripherals?

racerlucas, 2022-03-30:
> https://wiki.libreelec.tv/how-to/add-firmware
>
>
> I've never added xone to 10.x since I'm not using at at any system anymore - I'm just rebasing the packages from time to time. Anyway this kernel [...]

So no I should download LE 11.x?

SupervisedThinking, 2022-03-30:
LE-RR ≠ LE

If you want to simply use LibreELEC + Moonlight & stuff it's probably conveniant to use my builds. I can backport xone to 10.x too but rebuilding all images is not something with high priority for me.

Since LE/Kodi can be controlled by Xbox controllers & offers Retroplayer it is most likely a kernel driver which should be considered for upstream too. But since it's adds up some space other team members should have a look at the problem first.

racerlucas, 2022-03-30:
> 

Maybe you can make an addon to do this?

racerlucas, 2022-03-30:
The only solution I can think about now is to build a HTPC running windows or linux and use AEL to launch Moonlight.

closed 2022-03-30 by lrusak