LibreELEC/LibreELEC.tv #6348: [BUG][LE11] Bluetooth remote stoped working after bluez 5.64 bump

pretoriano80, 2022-04-01:
### Describe the bug
My Amazon Fire TV bluetooth  remote (the one with volume buttons and voice search) is connected/paired correctly but no input is detected in Kodi.Reverting to bluez 5.63 fixes the issue

### To Reproduce
Steps to reproduce the behavior:
1. Pair the remote from the Bluetooth menu
2. Try to navigate thru Kodi UI
3. Kodi UI doesn't respond to any input

### Informations
 - LE Version: LE11
 - Hardware Platform: RPi4

### Log file
bluez 5.63 --> **remote working** -> [bt_563.log](https://github.com/LibreELEC/LibreELEC.tv/files/8396331/bt_563.log)

bluez 5.64 --> **remote not working** --> [bt_564.log](https://github.com/LibreELEC/LibreELEC.tv/files/8396346/bt_564.log)

### Additional context
Bluez DEBUG is enabled on my test build,so let me know if you need more logs.

heitbaum, 2022-04-01:
Looking at the git. Nothing obvious would cause it… https://git.kernel.org/pub/scm/bluetooth/bluez.git/log/

nothing obvious in your dmesg logs. Btmon / https://github.com/bluez/bluez/blob/master/monitor/btmon.rst and hci tools will be the next step to see what is actually “connected” and sending.

pretoriano80, 2022-04-01:
Ok ,here are somo more logs captured with btmon:
[btmon.log](https://github.com/LibreELEC/LibreELEC.tv/files/8396929/btmon.log)
[hcidump.log](https://github.com/LibreELEC/LibreELEC.tv/files/8396930/hcidump.log)

The logs were aquired while doing this:
1) Pairing the bluetooth remote from the LibreELEC Configuration -> Bluetooth 
2) Remote was successfully paired (with Trust and Connect)
3) Pressing random buttons on remote (no response from Kodi)

pretoriano80, 2022-04-01:
And here are the logs with the  same steps as above and only bluez reverted to 5.63 on master branch:
[btmon_5.63.log](https://github.com/LibreELEC/LibreELEC.tv/files/8396997/btmon_5.63.log)
[hcidump_5.63.log](https://github.com/LibreELEC/LibreELEC.tv/files/8397001/hcidump_5.63.log)

Pairing and input working OK on this version.

pretoriano80, 2022-04-01:
As a side note,with bluez 5.64 even the CEC functionality of Volume,Mute and Power buttons is lost .Reverting to 5.63 and those buttons are also working again.

pretoriano80, 2022-04-01:
Tested with this commit https://github.com/bluez/bluez/commit/8e1892cc9217afcb82fd5ab50cfcbbe7b69b35ac and the remote is working...so we are down to 4 commits to find culprit.

closed 2022-04-02 by CvH