LibreELEC/LibreELEC.tv #6379: [BUG] failed to allocate buffer

JoyHusky, 2022-04-13:
### Describe the bug
Kodi hungs on welcome tool

### Informations
 - LE Version: 11.0
 - Hardware Platform: Samsung Exynos

### Log file

kodi.log:
2022-04-13 18:56:06.074 T:982      INFO <general>: SETTINGS: set_keyboard_layout # us
2022-04-13 18:56:06.074 T:982      INFO <general>: SETTINGS: set_keyboard_layout # loadkmap < `ls -1 /usr/lib/keymaps/*/us.bmap`
2022-04-13 18:56:08.356 T:982      INFO <general>: ## LibreELEC Addon ## updates::updateThread ## Started
2022-04-13 18:56:08.358 T:982      INFO <general>: ## LibreELEC Addon ## updates::set_auto_update ## auto
2022-04-13 18:56:08.816 T:949      INFO <general>: Loading skin file: /usr/share/kodi/addons/service.libreelec.settings/resources/skins/Default/1080i/service-LibreELEC-Settings-wizard.xml, load type: LOAD_ON_GUI_INIT
2022-04-13 18:56:08.863 T:949     ERROR <general>: Control 1500 in window 13001 has been asked to focus, but it can't
2022-04-13 18:56:08.864 T:1008     INFO <general>: SETTINGS: run # Waiting
2022-04-13 18:56:12.459 T:949      INFO <general>: Loading skin file: DialogSelect.xml, load type: KEEP_IN_MEMORY
2022-04-13 18:56:28.669 T:949     ERROR <general>: eglSwapBuffers failed (EGL_BAD_ALLOC)

DMESG:
[   10.268164] mali_midgard: loading out-of-tree module taints kernel.
[   10.297259] mali 11800000.gpu: Continuing without Mali bus clock
[   10.303266] mali 11800000.gpu: GPU identified as 0x0620 r0p1 status 0
[   10.303786] mali 11800000.gpu: Protected mode not available
[   10.379953] mali 11800000.gpu: Probed as mali0
[   11.514612] EXT4-fs (mmcblk0p2): mounted filesystem with ordered data mode. Quota mode: disabled.
[   12.291287] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   12.296584] r8152 6-1:1.0 eth0: carrier on
[   46.816359] exynos-mixer 14450000.mixer: [drm:exynos_drm_gem_create] *ERROR* failed to allocate buffer.
[   46.816395] exynos-drm exynos-drm: FB allocation failed.
[   46.828623] exynos-mixer 14450000.mixer: [drm:exynos_drm_gem_create] *ERROR* failed to allocate buffer.
[   46.828653] exynos-drm exynos-drm: FB allocation failed.
[   55.290732] BUG: Bad rss-counter state mm:019dcbe3 type:MM_FILEPAGES val:26646

systemd kodi.service log:
Apr 13 19:11:37 kodi systemd[1]: Starting Kodi Media Center...
Apr 13 19:11:37 kodi systemd[1]: Started Kodi Media Center.
Apr 13 19:12:46 kodi kodi.sh[1097]: terminate called after throwing an instance of 'std::runtime_error'
Apr 13 19:12:46 kodi kodi.sh[1097]:   what():  eglSwapBuffers failed
Apr 13 19:12:54 kodi kodi.sh[1093]: Aborted (core dumped)

lrusak, 2022-04-13:
I'm pretty sure this device won't be supported anymore as the mainline kernel needs changes to work with mali or something. @chewitt will know more.

heitbaum, 2022-04-15:
See the RFE here - #6250

heitbaum, 2023-01-15:
LE11 has an XU3 and Xu4 build now

closed 2023-01-15 by heitbaum

dandygig, 2024-09-13:
Still facing this error in Log on a

Hardware: ODROID N2+  / 4G S922X (vendor: Hardkernel)

2024-09-13 08:28:59.345 T:3739     info <general>: -----------------------------------------------------------------------
2024-09-13 08:28:59.345 T:3739     info <general>: Starting Kodi (21.1 (21.1.0) Git:72f13fb388a529094b8d5b5274ed0e5f1a2af8da). Platform: Linux ARM 32-bit
2024-09-13 08:28:59.345 T:3739     info <general>: Using Release Kodi x32
2024-09-13 08:28:59.345 T:3739     info <general>: Kodi compiled 2024-08-26 by GCC 13.2.0 for Linux ARM 32-bit version 4.9.255 (264703)
2024-09-13 08:28:59.345 T:3739     info <general>: Running on Hardkernel ODROID-N2Plus with CoreELEC (official): 21.1.1-Omega, kernel: Linux ARM 64-bit version 4.9.269
2024-09-13 08:28:59.345 T:3739     info <general>: FFmpeg version/source: 6.0.1
2024-09-13 08:28:59.345 T:3739     info <general>: Host CPU: Amlogic S922X rev b, 6 cores available
2024-09-13 08:28:59.345 T:3739     info <general>: ARM Features: Neon enabled

2024-09-13 08:29:02.081 T:3824     info <general>: ## CoreELEC Addon ## system::set_hostname ## CoreELEC
2024-09-13 08:29:02.085 T:3824     info <general>: ## CoreELEC Addon ## system::set_keyboard_layout ## us
2024-09-13 08:29:02.362 T:3824     info <general>: ## CoreELEC Addon ## updates::updateThread ## Started
2024-09-13 08:29:02.362 T:3824     info <general>: ## CoreELEC Addon ## updates::set_auto_update ## auto
2024-09-13 08:29:04.665 T:3739     info <general>: Loading skin file: /usr/share/kodi/addons/service.coreelec.settings/resources/skins/skin.estuary/xml/service-CoreELEC-Settings-wizard.xml, load type: LOAD_ON_GUI_INIT
2024-09-13 08:29:04.683 T:3739    **error <general>: Control 1500 in window 13001 has been asked to focus, but it can't**
2024-09-13 08:29:04.683 T:4261     info <general>: ## CoreELEC Addon ## _service_::run ## WAITING:
2024-09-13 08:29:04.703 T:4260     info <general>: ## CoreELEC Addon ## xdbus Monitor started. ## 
2024-09-13 08:29:08.981 T:3816     info <general>: CAddonMgr::FindAddon: metadata.common.fanart.tv v3.6.4 installed
2024-09-13 08:29:09.060 T:3831    error <general>: CCurlFile::CReadState::FillBuffer - (0xdeb17178) Failed: URL using bad/illegal format or missing URL(3)