LibreELEC/LibreELEC.tv #6400: [BUG] cross-compile error with libretro-fsuae

heitbaum, 2022-04-16:
### Describe the bug

`libretro-fsuae` fails with the build due to the target binary `gen/build68k` being used during build.

### To Reproduce
Steps to reproduce the behavior:
1. `scripts/build libretro-fsuae`

### Informations
 - LE Version: LE11
 - Hardware Platform: Generic

### Log file

```
./gen/build68k < ./src/table68k > ./gen/cpudefs.c
./gen/build68k: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by ./gen/build68k)
make[1]: *** [Makefile:2011: gen/cpudefs.c] Error 1
make[1]: Leaving directory '/var/media/DATA/home-rudi/LibreELEC.tv/build.LibreELEC-Generic.x86_64-11.0-devel/build/libretro-fsuae-6b98f852e00a83ecdcf497c1032882ad7b6efc99/sources'
```

### Additional context

SupervisedThinking, 2022-04-16:
well `table68k` needs to be cross-compiled or a host build 🤔 
- https://github.com/FrodeSolheim/fs-uae/blob/master/Makefile.am#L1749-L1750
- https://github.com/FrodeSolheim/fs-uae/blob/master/src/ChangeLog#L2292-L2295

Anyway I've had the same problem with standalone FS-UAE & I guess it's a topic for the upstream dev.

heitbaum, 2022-05-22:
Fixed

closed 2022-05-22 by heitbaum