LibreELEC/LibreELEC.tv #6401: [BUG] Skip do not show a head-up if you are on pause

Mannshoch, 2022-04-16:
### Describe the bug
Skip do not show a head-up if you are on pause

### To Reproduce
Steps to reproduce the behavior:
1. Go to PVR Addon
2. Start a replay show
3. on Start of the Comercials press "Pause"
4. Press the arrow to skip Comercials
5. The headup information disappear.

### Informations
 - LE Version: 10.0.2
 - Hardware Platform: RPi4

### Log file
https://pastebin.com/6uNpvZzw

lrusak, 2022-04-16:
Please use the Kodi forums for Kodi support questions. https://forum.kodi.tv/

closed 2022-04-16 by lrusak

Mannshoch, 2022-04-18:
That's not a support request. It's a bug report. Please reopen.

[But for the record](https://forum.libreelec.tv/thread/25496-since-10-0-2-skip-do-not-show-a-head-up-if-you-are-on-pause/?postID=168342#post168342)

lrusak, 2022-04-18:
> That's not a support request. It's a bug report. Please reopen.
> 
> 
> 
> [Request on Forum](https://forum.libreelec.tv/thread/25496-since-10-0-2-skip-do-not-show-a-head-up-if-you-are-on-pause/?postID=168342#post168342)
> 
> 

It's a Kodi thing, that's why I linked the Kodi forum.