LibreELEC/LibreELEC.tv #6410: [BUG] Broken HDMI output on RPi4 with certain content starting with 10.0.2

haarp, 2022-04-20:
Hello!

### Describe the bug
After upgrading LibreELEC to 10.0.2, HDMI output disappears when certain content starts playing. As soon as playback starts, the TV claims that no signal is present. If playback is stopped, signal returns. This issue disappears after downgrading to 10.0.1 and is very reproducible. There is a regression.

### To Reproduce
Steps to reproduce the behavior:
1. Have the Kodi setting that adapts the resolution/framerate to the content enabled
2. Start playback of certain content (I haven't figured out which content triggers it yet. Might be related to HEVC?)

### Informations
 - LE Version: 10.0.2
 - Hardware Platform: RPi4, LG OLED55B19LA

### Log file
I'm attaching two log files. good.txt which is v10.0.1, and bad.txt which is v10.0.2. They can be diff'ed if the timestamps are stripped: `diff -u <(cut -b35- good.txt) <(cut -b35- bad.txt)`

In the diff, I found this line only present on the bad version, which looks quite suspicious:

`DEBUG <general>: CVideoLayerBridgeDRMPRIME::Configure - setting max bpc to 12 (true)`

Is it trying to set a color mode that the TV doesn't support?

[good.txt](https://github.com/LibreELEC/LibreELEC.tv/files/8517296/good.txt) [bad.txt](https://github.com/LibreELEC/LibreELEC.tv/files/8517299/bad.txt)

### Additional context
The following RPi4 config.txt entries are used in addition to the default ones:

```
# Video
hdmi_enable_4kp60=1
gpu_mem=192
# Argon One stuff
enable_uart=1
dtparam=i2c=on
```

I am no sure if this should go towards the LE or the Kodi team. Please let me know if this is the wrong place.

Thanks!

HiassofT, 2022-04-20:
This is a known issue and will be fixed in 10.0.3
see https://forum.libreelec.tv/thread/25427-le-10-0-2-on-rpi4-not-playing-files-that-10-0-1-had-no-problems-with/
and https://github.com/raspberrypi/linux/pull/4980

closed 2022-04-20 by HiassofT

haarp, 2022-04-20:
Thanks @HiassofT, that was exactly the issue. I have to say, it's kinda confusing to have a separate bug tracker in the forum tho.