LibreELEC/LibreELEC.tv #6424: [BUG] ScummVM does not start (LE 10, RPi3)

beta-tester, 2022-04-24:
### Describe the bug
i just switched from LibreELEC 9.2.8 for RPi3 to the new offical LibreELEC 10.0.2 for RPi3
and unfortunately i have to say, i have the issue again with the ScummMV add-on (v2.1.1.23.1)...
​the add-on does not start on LibreELEC 10.0.2

is it still the same issue as under LibreELEC 9.x?
https://forum.libreelec.tv/thread/14371-retroplayer-scummvm/
https://github.com/LibreELEC/LibreELEC.tv/pull/3479

### To Reproduce
Steps to reproduce the behavior:
1. install add-on: game.libretro.scummvm v2.1.1.23.1
2. select Games / ScummVM
3. try to start it
[image]
4. See error
[image]

### Informations
 - LE Version: 10.0.2 (LibreELEC release: RPi2.arm-10.0.2)
 - Hardware Platform: RPi3

### Log file
```
################################################################################
# ... output of cat /storage/.kodi/temp/kodi.log
# LibreELEC release: RPi2.arm-10.0.2
# 
################################################################################
﻿2022-04-24 16:22:45.379 T:673      INFO <general>: -----------------------------------------------------------------------
2022-04-24 16:22:45.380 T:673      INFO <general>: Starting Kodi (19.4 (19.4.0) Git:19.4-Matrix). Platform: Linux ARM 32-bit
2022-04-24 16:22:45.380 T:673      INFO <general>: Using Release Kodi x32
2022-04-24 16:22:45.380 T:673      INFO <general>: Kodi compiled 2022-03-05 by GCC 10.2.0 for Linux ARM 32-bit version 5.10.95 (330335)
2022-04-24 16:22:45.380 T:673      INFO <general>: Running on BCM2835 with LibreELEC (official): 10.0.2, kernel: Linux ARM 32-bit version 5.10.95
2022-04-24 16:22:45.380 T:673      INFO <general>: FFmpeg version/source: 4.3.2-Kodi
2022-04-24 16:22:45.380 T:673      INFO <general>: Host CPU: ARMv7 Processor rev 4 (v7l), 4 cores available
2022-04-24 16:22:45.380 T:673      INFO <general>: ARM Features: Neon enabled
2022-04-24 16:22:45.380 T:673      INFO <general>: special://xbmc/ is mapped to: /usr/share/kodi/
2022-04-24 16:22:45.380 T:673      INFO <general>: special://xbmcbin/ is mapped to: /usr/lib/kodi
2022-04-24 16:22:45.380 T:673      INFO <general>: special://xbmcbinaddons/ is mapped to: /usr/lib/kodi/addons
2022-04-24 16:22:45.380 T:673      INFO <general>: special://masterprofile/ is mapped to: /storage/.kodi/userdata
2022-04-24 16:22:45.380 T:673      INFO <general>: special://envhome/ is mapped to: /storage
2022-04-24 16:22:45.380 T:673      INFO <general>: special://home/ is mapped to: /storage/.kodi
2022-04-24 16:22:45.380 T:673      INFO <general>: special://temp/ is mapped to: /storage/.kodi/temp
2022-04-24 16:22:45.380 T:673      INFO <general>: special://logpath/ is mapped to: /storage/.kodi/temp
2022-04-24 16:22:45.380 T:673      INFO <general>: The executable running is: /usr/lib/kodi/kodi.bin
2022-04-24 16:22:45.380 T:673      INFO <general>: Local hostname: LibreELEC
2022-04-24 16:22:45.380 T:673      INFO <general>: Log File is located: /storage/.kodi/temp/kodi.log
2022-04-24 16:22:45.380 T:673      INFO <general>: -----------------------------------------------------------------------

2022-04-24 16:22:46.071 T:673      INFO <general>: CAddonMgr::FindAddons: game.libretro.scummvm v2.1.1.23.1 installed
2022-04-24 16:23:14.506 T:673      INFO <general>: RetroPlayer[PROCESS]: Created process info for GBM
2022-04-24 16:23:14.542 T:673     ERROR <general>: ADDON: Dll ScummVM - Client returned bad status (5) from Create and is not usable
2022-04-24 16:23:14.543 T:673      INFO <general>: Loading skin file: DialogConfirm.xml, load type: KEEP_IN_MEMORY
2022-04-24 16:23:16.379 T:673     ERROR <general>: RetroPlayer[PLAYER]: Failed to initialize game.libretro.scummvm
2022-04-24 16:24:05.205 T:673      INFO <general>: RetroPlayer[PROCESS]: Created process info for GBM
2022-04-24 16:24:05.211 T:673     ERROR <general>: ADDON: Dll ScummVM - Client returned bad status (5) from Create and is not usable
2022-04-24 16:24:10.296 T:673     ERROR <general>: RetroPlayer[PLAYER]: Failed to initialize game.libretro.scummvm

```

### Additional context

heitbaum, 2022-10-15:
New report.
https://forum.libreelec.tv/thread/26164-scummvm-add-on-could-not-be-loaded-an-unknown-error-has-occurred/

HiassofT, 2022-11-19:
Fixed builds are now available in the repo

closed 2022-11-19 by HiassofT

beta-tester, 2022-11-28:
thank you for the fix.

- Day of the Tentacle - working.
- Discworld 1 - **ScummVM/Kodi/LibreELEC crashing immediately.**
- Discworls 2 - working
- Monkey Island 1 - working
- Monkey Island 2 - working
- Monkey Island 3 - **crashing ScummVM/Kodi/LibreELEC after intro "the monkey are listening"**
- Simon the sorcerer 1 - working
- Simon the sorcerer 2 - working

all games are german PC versions from CD.