LibreELEC/LibreELEC.tv #6429: [BUG] Playing some PVR streams causes reboot (RPi4)

lytenyn, 2022-04-26:
### Describe the bug
Since today, kodi crashes and reboots when playing certain PVR channels.
This seems affect only SD channels. Neither the PVR backend (tvheadend) nor the libreelec installation was modified in any way.

### To Reproduce
Steps to reproduce the behavior:
1. Play some PVR SD streams (like Eurosport on Astra)
2. Immediate reboot

### Informations
 - LE Version: 10.0.2
 - Hardware Platform: RPi4

### Log file
http://ix.io/3Wk4

### Additional context

HiassofT, 2022-04-26:
Please provide a short sample that triggers the issue and a full log ("pastecrash" via command line or crash log upload from LE settings addon)

lytenyn, 2022-04-27:
I have a 4 MB example video. Not sure how to send it publicly, maybe try this: https://file.io/3cn72tlLX6Bh ?
Please tell me if you have a better recommendation how to send it. I can also provide a link to a server of mine, but I do not want to post it publicly.

Note that I have uploaded `kodi_crash.log` already. Since it was more than 1MB in size, I had to cut it in the middle to make it fit into ix.io. If I have cut too much, I can also try to cut it differently.

HiassofT, 2022-04-27:
That link just shows me "file has been deleted". A 4MB file is no problem, just send it to me by email: hias@horus.com

The reason why we want "pastecrash" log is that it'll also contain journal, your config.txt and other important info - kodi_crash.log only contains the kodi part which is often only half part of the story.

If you run "pastecrash -c > crash.log" it'll save it to a file (run that immediately after kodi restarted, don't reboot the rpi in between) - just send it to me by email as well

therealcarnivor, 2022-04-28:
I don't have any logs to show but I have also noticed this issue,

LE Version: 10.0.2
Hardware Platform: RPi4

I've had it happen about a week ago, when the wife kept crashing it until LE came up in safe mode, when i did a controlled restart after that it worked fine again, then it did it again this morning, which is how i've ended up here

lytenyn, 2022-04-28:
Sent an email with the test file to HiassofT. Unfortunately, I could not reproduce the crash today after a clean reboot, but I attached the original, full crash log and the output of pastecrash (but without an actual crash).

I'll send an updated pastecrash output once it happens again.

HiassofT, 2022-04-29:
Thanks a lot, I got your email.

Can you test with this build? https://www.horus.com/~hias/tmp/libreelec/LibreELEC-RPi4.arm-10.0-devel-20220328220759-4d9d4e2.tar

It includes the latest LE10 ffmpeg updates and from the changelog it looks like it might already contain a fix for your crash

lytenyn, 2022-04-30:
Thank you! Installed the test build, seems to work fine so far. As it was not 100% reproducible, will keep it running for a few days and will report back.

lytenyn, 2022-05-03:
It still seems to work fine. I would propose to close the issue, unless you want to track this until the next version is released.
Thanks again for your help!

HiassofT, 2022-05-03:
Thanks for reporting back!

I'll close this issue as the fix is already included in the LE10 branch https://github.com/LibreELEC/LibreELEC.tv/pull/6336

closed 2022-05-03 by HiassofT

lytenyn, 2022-06-06:
Actually, the problem has come back. I'm still running the test build that you provided to me, now multiple channels crash the RPi and cause a reboot. If I remember correctly, the error is the same: `ERROR <general>: ffmpeg[0x228bf10X]: [deinterlace_v4l2m2m] VIDIOC_REQBUFS failed: Invalid argument`.

Maybe this is a duplicate of #6508 ?

I can provide a crash log and test file via email again, if you want.

HiassofT, 2022-06-06:
Can you test with this build instead?
https://www.horus.com/~hias/tmp/libreelec/LibreELEC-RPi4.arm-10.0-devel-20220603151054-eb85a9e.tar

I've added a couple more updates, maybe they help with your issue, too.

If you still see the issue please mail me the crashlog and test file

lytenyn, 2022-06-08:
Thank you! Strangely, like last time, this build seems to fix the (current) problems. Let's see whether it sticks this time..