LibreELEC/LibreELEC.tv #6432: [BUG] CECActivateSource() turns on tv repeatedly until reboot

kcd83, 2022-04-29:
### Describe the bug
CECActivateSource() turns on tv until reboot. Upon turning off the TV it will immediately turn back on. If power cycle LibreELEC/Kodi the TV can turn off.

### To Reproduce
Steps to reproduce the behavior:
1. Do one of:
- Use the Android's remote app Yatse, under power choose "CEC Activate"
- Use UDP based API directly sending CECActivateSource()
- Initiate playback on any video (I think this is dependant on various settings so I cannot specify exactly)
2. TV turns on automatically (must be set to respond to CEC Power calls)
3. Attempt to turn off TV... go back to 2
4. Power cycle Kodi (I used the JSONRPC API for System.Reboot but anything works)
5. Turn off the TV

### Informations
 - LE Version: 9.2.8
 - Hardware Platform: RPi3
 - TV: lg 55la6620 and older Netcast version

### Log file
http://ix.io/3Wxs

### Additional context
The reason I use the [UDP API](https://gitlab.com/kcd/kodi-send-udp-action-tool) is because I use Kodi to turn on my TV from HomeAssistant. (Unrelated but for some reason the JSONRPC API does not support this command but the UDP one does 😃 )

This topic has come up before and most people just turn off the HDMI CEC power on function on their TV. I am hopeful this issue will be more useful because it is very repeatable and specific to this CECActivateSource() function without abiguity around other configuration.

Also people blame the TV brand, however since power cycling Kodi fixes the issue, to me, that implies that Kodi is sending _some_ CEC command to the TV (my suspicion being Activate Source ever second or so). If it was just the TV then power cycling Kodi wouldn't stop it would it?

Let me know if I can provide more help

Joeviocoe, 2022-05-18:
I tried to reproduce, but mine is working.

Using steps:
- Yatse "CEC Activate" 
or 
- the [JSONCEC](https://github.com/joshjowen/script.json-cec) script addon (modified for Python3 and urllib.parse)

LE version is the latest 10.0.2 on RPi2B (installed new yesterday)
TV is a Samsung UN65NU6950

Comparing the log file you uploaded:
Yours - Register - new cec device registered on cec->RPI: CEC Adapter (2708:1001)
Mine - Register - new cec device registered on cec->Linux: CEC Adapter (0000:0000)
but I think this is just a minor difference in the way the RPi's register.

Have you sniffed and/or blocked UDP 9777 to eliminate the possibility of multiple packets?

kcd83, 2022-05-19:
Thank you for trying to reproduce this. Perhaps there is something the LG TV does to trigger the repeated wake behaviour (eg. Doesn't ack so constantly retries). I am still convinced there is an issue otherwise restarting Kodi/libreELEC wouldn't stop the behaviour.

I think I did a tcpdump when I first wrote my script to port 9777 but may need to repeat it. Given both my simple script and the Yatse app do this every time I don't think either is sending the CECActivateSource command multiple times, was that your thought?

Joeviocoe, 2022-05-19:
When bug is occurring, try unplugging and replugging the HDMI cable and see if that fixes it the same way as a RPi reboot.

kcd83, 2022-06-05:
Unplugging the HDMI allowed me to turn off the TV normally

I wait 15 minutes, plugged the HDMI back in. The TV turned on, and would repeated turn back on after putting off

Rebooting Kodi, as expected, resolved the issue

Sorry it took so long for me to confirm this

HiassofT, 2022-06-05:
Please test with the build and cmdline.txt option from here:
https://forum.libreelec.tv/thread/24783-tv-avr-turns-back-on-right-after-turning-them-off/?postID=170126#post170126

kcd83, 2022-06-05:
Looks very promising, absolutely will try that and will report back. Thanks

prettyv, 2022-06-06:
I have the same issue with the Raspberry Pi 4 connected to a receiver, which is woken up again pretty much immediately after turning off the TV.

Tried the new test build with the specified cmdline.txt option and so far this seems to do the trick for me.

cr3bs, 2022-06-08:
I too have CEC issues on my raspberry pi 4, albeit somewhat different. I have my raspberry pi 4 connected via HDMI to a soundbar, which is in turn connected via HDMI to a projector. As soon as I turn off the soundbar, the raspberry pi will loose it's physical address on the CEC bus and report that the HDMI port is disconnected.

I can confirm that this patch: https://github.com/raspberrypi/linux/pull/5052 fixes the issue.

HiassofT, 2022-06-08:
FYI: the kernel update PR which also adds https://github.com/raspberrypi/linux/pull/5052 has been merged so the vc4.force_hotplug cmdline.txt option will be available in the next nightly le11 build

cr3bs, 2022-06-08:
Thanks! @HiassofT Are these changes by any chance backported to le 10?

HiassofT, 2022-06-08:
@cr3bs yes, I plan to include the change in LE10, too. The LE10 build in the linked forum post already contain the change, but I'd like to get some more feedback before adding them

cr3bs, 2022-06-08:
@HiassofT Thanks, awesome! (That build is the one I tested.)

Joeviocoe, 2022-06-25:
I occasionally have this problem too.  Very intermittent and hard to test consistently.

MusterGit, 2022-06-28:
I had problems with CEC for weeks now and ever since wondered why. Both cec-ctl (linux kernel) and cec-client (libcec) stopped working some minutes after boot on my RPi4. The adapter lost its logical address and could not be reset by any means. This patch fixes the problem, so the adapter stays in the game. Thank you very much, @HiassofT and posse 👍

Joeviocoe, 2022-07-11:
Is there a way to use this [patch](https://github.com/raspberrypi/linux/pull/5052) without switching completely to LE11 nightly?
I don't want to leave LE 10.0.2 stable branch but I want to fix this.

HiassofT, 2022-07-12:
@Joeviocoe the patch is already in LE10 branch https://github.com/raspberrypi/linux/pull/5052
just use the latest LE10 nightly from https://test.libreelec.tv/10.0/RPi/

Joeviocoe, 2022-07-12:
> @Joeviocoe the patch is already in LE10 branch [raspberrypi/linux#5052](https://github.com/raspberrypi/linux/pull/5052) just use the latest LE10 nightly from https://test.libreelec.tv/10.0/RPi/

Thanks.  So reinstall the nightly build, which would reset the kodi database, right?

HiassofT, 2022-07-12:
No, LE 10 nightlies still use kodi 19 so won't change anything to your database - just copy the img.gz to the update share / folder and reboot

Joeviocoe, 2022-07-12:
> No, LE 10 nightlies still use kodi 19 so won't change anything to your database - just copy the img.gz to the update share / folder and reboot

Thank you.  That was painless. Copied the img.gz to /storage/.update directory and rebooted.

prettyv, 2022-11-06:
@HiassofT Since the update of the RPi kernel to 6.0 in #6960 this issue is back in the LE11 nightly builds because the [fix](https://github.com/raspberrypi/linux/commit/6efdadb1106f8fa06330ba3a095649d2c6417d3f) has been applied to the `5.15.y` branch but not to `6.0.y` (nor `6.1.y`). Is there a chance this patch can also be adopted for newer kernel versions?

HiassofT, 2022-11-06:
@prettyv thanks for the report! I've pinged RPi devs that they should add it to 6.0 and 6.1 kernels

prettyv, 2022-11-09:
@HiassofT I see the patch has already been applied to both branches, would be great if this can be incorporated into the nightly releases. Thanks for pinging them!

HiassofT, 2022-11-12:
@prettyv PR with kernel update is here: #7120

prettyv, 2022-11-13:
@HiassofT Great, just installed the latest nightly build and tested, works as expected, thanks a lot!

heitbaum, 2023-09-04:
Closing as fixed,

closed 2023-09-04 by heitbaum