LibreELEC/LibreELEC.tv #6475: Bluetooth keyboard connectivity issue | Logitech K850

anshulosahub, 2022-05-09:
**Platform**: Raspberry pi3, ARMV7 with LibreELEC Build(19.4.0), Git (19.4-Matrix), Compiled: 2022-03-05
**Summary**: Was able to connect the wireless mouse (M720) , which comes with Logitech K850 keyboard but the keyboard setup was not completed due to a bug.
**Expected behaviour**: How keyboard connects generally: there is a pop-up window displaying a 6-digit code (on Android/ iOS/ Windows), on the keyboard you have to enter that, to connect. 
**Bug**: On LibreELEC I am getting the pop-up but that's blank, no number/ pin to enter, and hence unable to connect
[image]

antonlacon, 2022-10-24:
Should be resolved on 10.0.3. Please re-test.

chewitt, 2022-11-29:
Closing as the issue was resolved in 10.0.3 (and master). If not, please request reopen of file a new bug report. Thanks!

closed 2022-11-29 by chewitt