LibreELEC/LibreELEC.tv #6488: [FEATURE] Add patch RPi for license codec MPEG-2 VC-1

Ilia1, 2022-05-14:
[How to patch](https://www.reddit.com/r/raspberry_pi/comments/8sg0a6/patch_for_mpeg2_vc1_license_again_for_osmc/)
[MPEG-2 Patents Have Expired](https://yro.slashdot.org/story/18/02/14/1621259/mpeg-2-patents-have-expired)

closed 2022-05-14 by CvH

HiassofT, 2022-05-14:
This is something you'd need to discuss with RPi folks, we ship unmodified binary firmware and have no intentions to patch it.

Ilia1, 2022-05-14:
https://github.com/raspberrypi/firmware/issues/939
The patent remained only in Malaysia. Maybe make image for use in other countries?

HiassofT, 2022-05-14:
There's nothing we can do until RPi changes the firmware

Also keep in mind that redistributing modified firmwares is not allowed https://github.com/raspberrypi/firmware/blob/master/boot/LICENCE.broadcom