LibreELEC/LibreELEC.tv #6549: [BUG] automake inconsistently fails to build (race condition)

heitbaum, 2022-05-31:
This occurs on all targets (but randomly)
```
<<< automake:host seq 31 <<<
UNPACK      automake
    APPLY PATCH (common)      packages/devel/automake/patches/automake-01-fix-help2man.patch
patching file Makefile.in
Hunk #1 succeeded at 703 (offset 2 lines).
    APPLY PATCH (common)      packages/devel/automake/patches/automake-02-build-fix-race-in-parallel-builds.patch
patching file Makefile.in
[030/277] [FAIL] build   automake:host

Hunk #1 succeeded at 688 (offset -3 lines).
The following log for this failure is available:
  /build/build.LibreELEC-H5.arm-11.0-devel/.threads/logs/31.log

patching file bin/local.mk
    FIXCONFIG      /build/build.LibreELEC-H5.arm-11.0-devel/build/automake-1.16.5/lib/
BUILD      automake (host)
    TOOLCHAIN      configure (auto-detect)
Executing (host): /build/build.LibreELEC-H5.arm-11.0-devel/build/automake-1.16.5/configure --host=x86_64-linux-gnu --build=x86_64-linux-gnu --prefix=/build/build.LibreELEC-H5.arm-11.0-devel/toolchain --bindir=/build/build.LibreELEC-H5.arm-11.0-devel/toolchain/bin --sbindir=/build/build.LibreELEC-H5.arm-11.0-devel/toolchain/sbin --sysconfdir=/build/build.LibreELEC-H5.arm-11.0-devel/toolchain/etc --libexecdir=/build/build.LibreELEC-H5.arm-11.0-devel/toolchain/lib --localstatedir=/build/build.LibreELEC-H5.arm-11.0-devel/toolchain/var --disable-static --enable-shared --target=armv8a-libreelec-linux-gnueabihf --disable-silent-rules
configure: WARNING: unrecognized options: --disable-static, --enable-shared
checking whether make supports nested variables... yes
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether ln -s works... yes
checking for perl... /usr/bin/perl
checking for tex... no
checking for yacc... yacc
checking for lex... no
checking for flex... no
checking whether autoconf is installed... yes
checking whether autoconf works... yes
checking whether autoconf is recent enough... yes
checking whether ln works... yes
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
configure: will now look for a sturdy POSIX shell, for our testsuite
checking for sh... /usr/bin/sh
checking for sh5... no
checking for dash... /usr/bin/dash
checking for ash... no
checking for bash... /usr/bin/bash
checking for zsh... no
checking for ksh... no
checking for pdksh... no
checking whether /bin/bash supports $(cmd)... yes
checking whether /bin/bash supports $((expr))... yes
checking whether /bin/bash supports ${#var}... yes
checking whether /bin/bash supports ${var#glob} and ${var%glob}... yes
checking whether /bin/bash preserves exit traps with "set -e"... yes
checking whether /bin/bash can define exit traps in a shell function... yes
checking whether /bin/bash corrupts stderr with "set -x"... no
checking whether /bin/bash can return early from "dot-sourced" files... yes
checking whether /bin/bash supports alias named like shell builtins... yes
checking whether /bin/bash supports "test -e"... yes
configure: shell /bin/bash is good enough, stop looking
configure: will use /bin/bash as the testsuite shell
configure: will now look for generic compilers
checking for x86_64-linux-gnu-cc... /build/build.LibreELEC-H5.arm-11.0-devel/toolchain/bin/host-gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
[... 93 lines cut ...]
checking for x86_64-linux-gnu-gcj... no
checking for gcj... no
checking that generated files are newer than configure... done
configure: creating ./config.status
config.status: creating Makefile
config.status: creating pre-inst-env
config.status: linking /build/build.LibreELEC-H5.arm-11.0-devel/build/automake-1.16.5/GNUmakefile to GNUmakefile
configure: WARNING: unrecognized options: --disable-static, --enable-shared
Executing (host): make 
make[1]: Entering directory '/build/build.LibreELEC-H5.arm-11.0-devel/build/automake-1.16.5/.x86_64-linux-gnu'
 cd /build/build.LibreELEC-H5.arm-11.0-devel/build/automake-1.16.5 && "/build/build.LibreELEC-H5.arm-11.0-devel/build/automake-1.16.5/.x86_64-linux-gnu/pre-inst-env" automake-1.16 --gnu --ignore-deps Makefile
/build/build.LibreELEC-H5.arm-11.0-devel/build/automake-1.16.5/.x86_64-linux-gnu/pre-inst-env: 41: exec: automake-1.16: not found
make[1]: *** [Makefile:[248](https://github.com/LibreELEC/actions/runs/6656715205?check_suite_focus=true#step:11:249)5: /build/build.LibreELEC-H5.arm-11.0-devel/build/automake-1.16.5/Makefile.in] Error 127
make[1]: Leaving directory '/build/build.LibreELEC-H5.arm-11.0-devel/build/automake-1.16.5/.x86_64-linux-gnu'
FAILURE: scripts/build automake:host during make_host (default)
*********** FAILED COMMAND ***********
make ${PKG_MAKE_OPTS_HOST}
**************************************
FAILURE: scripts/build automake:host has failed!

The following log for this failure is available:
  /build/build.LibreELEC-H5.arm-11.0-devel/.threads/logs/31.log

>>> automake:host seq 31 >>>
```

the-dreamer, 2022-06-01:
for me it is happend consequent always ;-)

CvH, 2022-06-01:
on what distro + cpu are you building ?

the-dreamer, 2022-06-02:
Fedora 34 on a AMD Kabini A-5000. Its slow, but is/was working stable to build LE.

The last days i have tried to build LE for RPi and x86-64 it was failing always. After 4 or 5 times i was given up.

heitbaum, 2022-06-02:
@the-dreamer Im not seeing repeat errors. So with a scripts/build automake:host or a make image a second time it works. Can you please share your log / errors? Also the release that build*/.threads/log/##.log file.

the-dreamer, 2022-06-02:
@heitbaum, yes i can. 

https://paste.centos.org/view/7ac30666
https://paste.centos.org/view/2700fea9

Both, x86 and rpi are failing. 

on 27th april a build was successful :-)

So the the issue is new. 

i try to revert and report.

the-dreamer, 2022-06-02:
@heitbaum 

reverting the two commits is solving the issue: 

` chmod +x '/home/vncuser/libreelec/libreelec.tv/build.LibreELEC-RPi4.arm-11.0-devel/.sysroot/automake.host/usr/share/automake-1.16/ylwrap'
 chmod +x '/home/vncuser/libreelec/libreelec.tv/build.LibreELEC-RPi4.arm-11.0-devel/.sysroot/automake.host/usr/share/automake-1.16/depcomp'
 chmod +x '/home/vncuser/libreelec/libreelec.tv/build.LibreELEC-RPi4.arm-11.0-devel/.sysroot/automake.host/usr/share/automake-1.16/compile'
 chmod +x '/home/vncuser/libreelec/libreelec.tv/build.LibreELEC-RPi4.arm-11.0-devel/.sysroot/automake.host/usr/share/automake-1.16/py-compile'
 chmod +x '/home/vncuser/libreelec/libreelec.tv/build.LibreELEC-RPi4.arm-11.0-devel/.sysroot/automake.host/usr/share/automake-1.16/ar-lib'
 chmod +x '/home/vncuser/libreelec/libreelec.tv/build.LibreELEC-RPi4.arm-11.0-devel/.sysroot/automake.host/usr/share/automake-1.16/test-driver'
 chmod +x '/home/vncuser/libreelec/libreelec.tv/build.LibreELEC-RPi4.arm-11.0-devel/.sysroot/automake.host/usr/share/automake-1.16/tap-driver.sh'
make[3]: Leaving directory '/home/vncuser/libreelec/libreelec.tv/build.LibreELEC-RPi4.arm-11.0-devel/build/automake-1.16.5/.x86_64-redhat-linux'
make[2]: Leaving directory '/home/vncuser/libreelec/libreelec.tv/build.LibreELEC-RPi4.arm-11.0-devel/build/automake-1.16.5/.x86_64-redhat-linux'
make[1]: Leaving directory '/home/vncuser/libreelec/libreelec.tv/build.LibreELEC-RPi4.arm-11.0-devel/build/automake-1.16.5/.x86_64-redhat-linux'
>>> automake:host seq 18 >>>
[028/278] [DONE] build   automake:host

`

heitbaum, 2022-06-08:
Further research / follow up may be required. Some learnings maybe be at 
https://gitweb.gentoo.org/repo/gentoo.git/tree/sys-devel/automake/automake-1.16.5.ebuild

vpeter4, 2022-06-08:
What exactly is the reason for automake to fail? Is it build in parallel twice maybe?

heitbaum, 2022-06-08:
> What exactly is the reason for automake to fail? Is it build in parallel twice maybe?

It first raised as an issue on heavily loaded systems - so likely parallel out of order build. There are a number of patches in other distributions for “the error” - the patch I first tested worked fine on some buildhost. And not on others. This is the patch that has been reverted. The -parallel option has been added since the first attempt to fix automake. 🤞it is the only required workaround/fix.

vpeter4, 2022-06-09:
To me this looks like the build order problem and automake is build twice at the same time?