LibreELEC/LibreELEC.tv #6643: wlan0: disassociated from xx:xx:xx:xx:xx:xx (Reason: 252=<unknown>)

heitbaum, 2022-06-25:
### Describe the bug

wlan0: disassociated from xx:xx:xx:xx:xx:xx (Reason: 252=<unknown>)

A) need to improve the error messages.
B) need to let connman reconnect of 252 - 23

### To Reproduce
Wait till session timeout of 86400 on Cisco 9130 access point

### Informations
 - LE Version: 11.x
 - Hardware Platform: Tanix TX6 RTW88

### Log file

https://www.cisco.com/c/en/us/support/docs/wireless/catalyst-9800-series-wireless-controllers/213970-catalyst-9800-wireless-controllers-commo.html

### Additional context

jernejsk, 2022-06-25:
Did you mixed up log link?

heitbaum, 2022-06-25:
@jernejsk - no it’s the documentation of the logs from the Cisco WLC (or in my case the Embedded WIreless) - the logs from the TX6 are http://ix.io/40S7 . But will be the same in other devices.

Based on reading the blogs, ieee site …. It looks like an “undocumented” “unratified” Cisco standard. I have added the following kernel code. I’ll see if I get the right feedback. But I will want to / need to make an update to connman to try to reconnect On 252 - 23

```
commit 9298ce0daa0aaaaa1fe309c808fae78fc6f97641
Author: Rudi Heitbaum <rudi@heitbaum.com>
Date:   Sat Jun 25 12:06:59 2022 +0000

    add cisco deauth

diff --git a/include/linux/ieee80211.h b/include/linux/ieee80211.h
index 559b6c644938..18a9680da579 100644
--- a/include/linux/ieee80211.h
+++ b/include/linux/ieee80211.h
@@ -2819,6 +2819,8 @@ enum ieee80211_reasoncode {
        WLAN_REASON_MAC_EXISTS_IN_MBSS = 64,
        WLAN_REASON_MESH_CHAN_REGULATORY = 65,
        WLAN_REASON_MESH_CHAN = 66,
+       /* Cisco */
+       WLAN_REASON_ADMIN_RESET = 252, /* CLIENT_DEAUTH_REASON_ADMIN_RESET */
 };
 
 
diff --git a/net/mac80211/mlme.c b/net/mac80211/mlme.c
index 1eeabdf10052..fbec3f9abe9d 100644
--- a/net/mac80211/mlme.c
+++ b/net/mac80211/mlme.c
@@ -3107,6 +3107,7 @@ const char *ieee80211_get_reason_code_string(u16 reason_code)
        case_WLAN(MAC_EXISTS_IN_MBSS);
        case_WLAN(MESH_CHAN_REGULATORY);
        case_WLAN(MESH_CHAN);
+       case_WLAN(ADMIN_RESET);
        default: return "<unknown>";
        }
 }
```

jernejsk, 2022-06-26:
Interesting. Do you plan to work with kernel/connman community to solve this?

heitbaum, 2022-06-26:
Yes - plan on improving / understanding / fixing the interaction between connman/iwd/cisco/intel… The rtw88 (and the conman/iwd direction) gives some opportunity in this area. 

I have a pair of 9130 APs at home in Embedded Wireless Controller mode (and at work where we have 1000+ APs) WiFi 6, along with EAP authentication (and Intel drivers) have been a challenging issue. Drivers have been getting better and firmware on both the WLC and EWC. We are still experiencing issues with MIMO power save mode / spatial multiplexing power save (SMPS) mode. Stability of RF/IoT Devices (mostly Android now) on Fast Transition Enabled/Adaptive networks also a black art.