LibreELEC/LibreELEC.tv #6677: [BUG]

doucettom, 2022-07-03:
### Describe the bug
No hidden SSID Network found in the UI

### To Reproduce
Go to LibreELEC and connections and there is a list of detected SSIDs but no hidden SSIDs are detected.
### Informations
LE11 nightly from July 3rd
Generic-Legacy

### Additional context
Works fine with LE10, even LE10 June 28th Nightly

chewitt, 2022-07-03:
iwctl --passphrase <passphrase> station <device> connect-hidden <ssid>

^ You can try this. It's probably a difference between wpa_supplicant and the switch to iwd. If yes, there isn't much we can do about it other than point out that hidden SSIDs are a bit pointless (which is probably why the iwd devs haven't wasted much effort on support).

doucettom, 2022-07-08:
Might be related to Intel drivers in the end.

https://www.reddit.com/r/archlinux/comments/kcm1yd/cannot_connect_to_hidden_24ghz_network_with_iwctl/

bananenfisch, 2023-02-10:
After upgrading from 10.0.3 to 11beta1 i also ran in this bug.

> ```
> iwctl --passphrase <passphrase> station <device> connect-hidden <ssid>
> ```

Thanks @chewitt - this is the trick. For a dirty workaround, i just put this in my autostart.sh
(Hidden SSIDs are pointless in matter of security but there are still a few use-cases for it ;-))

chewitt, 2023-08-10:
I'm closing this ticket as ConnMan intentionally has no support for hidden networks (and if they are hidden, how is it supposed to show them in a list?). Not-a-bug.

closed 2023-08-10 by chewitt