LibreELEC/LibreELEC.tv #6688: [BUG][Allwinner] Green screen when playing dvds (pine64-lts)

enen92, 2022-07-11:
### Describe the bug
Not sure if this is a known issue, feel free to close if duplicated (I didn't find any related).
While looking into https://github.com/xbmc/xbmc/issues/21670 in my pine64-lts (latest master) I've found out the dvd under test resulted in a green screen on the kodi side. Log below:

https://paste.kodi.tv/ebeqigubar.kodi

Same file plays fine with `Allow hardware acceleration with DRM PRIME` disabled.
Since the reason was to debug the linked issue, I updated the kodi revision to latest master and dropped some patches which I would have otherwise to rebase. Surprised to check the file played just fine by dropping https://github.com/LibreELEC/LibreELEC.tv/blob/master/projects/Allwinner/patches/kodi/0002-WIP-DRMPRIME-deinterlace-filter.patch while still keeping the aforementioned setting enabled.

### To Reproduce
Steps to reproduce the behavior:
1. Just play any dvd (or iso) in pine64-lts or similar devices.

### Informations
 - LE Version: devel-20220711221910-f682eb5 11.0
 - Hardware Platform: pine64-lts

### Log file
https://paste.kodi.tv/ebeqigubar.kodi

jernejsk, 2022-07-12:
I tested two different DVDs with pine64-plus, which should be similar to your board. It worked with both. Did you build your own image or did you take nightly image? Can you try today nightly?

enen92, 2022-07-13:
Yeah it happens with any build (also affects latest nightly). Maybe this small .vob will help reproduce?

https://drive.google.com/file/d/1Hzv_5_PRVzrvciS-piTZHdZry-5ToOZq/view?usp=sharing

jernejsk, 2022-07-13:
It still works with nightly-20220713-5ce4da1: http://jernej.libreelec.tv/tmp/VID_20220713_205927.mp4 Not sure what to check next.

enen92, 2022-07-15:
I guess I might try to debug it further later on and check if I can adapt the patch to check exactly what breaks it - thanks for the effort

heitbaum, 2023-10-16:
@enen92 does this still occur with the latest LE12 nightlies?

enen92, 2023-10-16:
@heitbaum yes still happens with LE12 nightlies. Even worse, now I cannot play any video at all (black screen)

enen92, 2023-10-16:
log file: https://paste.kodi.tv/bedesaleza

enen92, 2023-10-16:
Actually if I use EGL instead of direct to plane I now get video, but the green screen is still present when playing the DVD or my tv channels: https://paste.kodi.tv/ugoxupeluq

jernejsk, 2023-10-16:
I guess introduced DRM plane rendering issue when I worked on 10-bit output support for H6. I'll fix that. However, as mentioned at the beginning of the thread, this is most likely issue with deinterlace driver. You disabled it and then it worked fine, according to the report. Note that I also noticed something similar on other SoCs, but it went away with new builds, so I can't really test now. 

@enen92  Can you confirm that disabling deinterlacing still helps? If so, I'll go through vendor driver again to see if I missed anything.

jernejsk, 2023-10-16:
DRM plane rendering is fixed with #8221

jernejsk, 2023-10-16:
It's not completed, original issue probably still persist

enen92, 2023-10-16:
> I guess introduced DRM plane rendering issue when I worked on 10-bit output support for H6. I'll fix that. However, as mentioned at the beginning of the thread, this is most likely issue with [...]

Interesting stuff, after toggling deinterlace off it started to play fine (well slow as hell but I guess that's the hw limitation)...

Then, I changed again to direct to plane and now all the files that exhibited this behaviour (green screen) started to play just fine (even with direct to plane that before just caused a black screen). Log: 
https://paste.kodi.tv/usujubodar

Guess the toggle persisted some state or register...I think I'll give a go to tomorrow nightly (fresh install) and redo the same experiments since now I cannot get a green screen anymore (even after a reboot).

jernejsk, 2023-10-16:
Can you put [this patch](http://ix.io/4Jav) into projects/Allwinner/patches/linux and see if it works? I know driver code is enabling deinterlace core in wrong order, so this might influence behaviour later.

enen92, 2023-10-17:
@jernejsk I tested today's nightly and I and I can confirm the "direct to render" option is now working fine. Green screen still occurs unless the interlacer is disabled.
Regarding your patch test request I currently don't have the environment to build libreelec set up so it will take me a while...
Do you guys have the necessary infra to generate testbuilds on demand? Can you generate some testbuild to the pine-lts including the patch?

jernejsk, 2023-10-17:
I found another discrepancy with vendor driver. Coefficients should be set always, after module is enabled. So this would be logical explanation why it doesn't work first time. Please test http://ix.io/4Jfy

enen92, 2023-10-17:
@jernejsk that's it :)
All working great now with the linked patch. Thanks a lot for looking into it

PR #8227 (2023-10-19): [LE12] Allwinner: linux: fix deinterlacer driver
This fixes #6688
closed 2023-10-19 by HiassofT