LibreELEC/LibreELEC.tv #6692: [BUG] Dvd playback over nfs failed

A1RF0RCE1, 2022-07-12:
### Describe the bug

Playback of DVD video folder over nfs fails. 
Error: One or more items failed to play. Check the log for more information.

### To Reproduce
Steps to reproduce the behavior:
1. Go to '...'
2. Play '....'
3. See error

### Informations
 - LE Version: 11
 - Hardware Plaform: RPi4

### Log file

https://pastebin.com/04qgXYkK

### Additional context

Extra info: https://github.com/xbmc/xbmc/issues/21670

enen92, 2022-08-05:
While this is purely a libreelec issue since it seems the library is not being properly wrapped I am trying to finally get rid of this on libdvdread upstream: https://code.videolan.org/videolan/libdvdread/-/merge_requests/40

This will solve this and similar issues, let's see if in time for v20 final.

lomion0815, 2023-01-25:
Is there any workaround known (like using SMB or such)?

lomion0815, 2023-01-25:
This problem still or again occurs in the latest nightly builds which use the latest xmbc library (libdvd*) versions.

lomion0815, 2023-01-31:
> While this is purely a libreelec issue since it seems the library is not being properly wrapped I am trying to finally get rid of this on libdvdread upstream: [...]

Its not in time for v20! What next?

vpeter4, 2023-02-02:
From what I see playing DVD folder works on LE Generic but not on arm build? Is this again some arm issue like Kodi sorting?

lomion0815, 2023-03-06:
No problems while testing on RPi3 and RPi4 for 2 weeks now. Very grateful for this fix!

heitbaum, 2023-09-04:
Closing as fixed

closed 2023-09-04 by heitbaum